openbmc/openbmc #1849: Help/Assistance with configuration of VLAN in openbmc /etc/systemd/network/ files, supported?

whowutwut, 2017-06-26:
@ratagupt  I'm trying to understand how the VLAN configuration works with systemd,  and just validating whether the openbmc OS can support this...   

I'm not an expert on this but after searching a little, created the following config files... in `/etc/systemd/network` 

### Without VLAN
```
root@witherspoon:/etc/systemd/network# cat 00-bmc-eth0.network
[Match]
Name=eth0

[Network]
Address=172.21.140.2/16
Gateway=172.21.254.1

[DHCP]
ClientIdentifier=mac
```

Restarting worked and a Static IP address is assigned. 
```
root@witherspoon:/etc/systemd/network# systemctl restart systemd-networkd
root@witherspoon:/etc/systemd/network# tail -n 10 /var/log/messages
Jun 26 19:26:39 witherspoon daemon.info systemd[1]: Starting Network Service...
Jun 26 19:26:39 witherspoon daemon.warn systemd-networkd[1102]: Could not create runtime directory: File exists
Jun 26 19:26:39 witherspoon daemon.warn systemd-networkd[1102]: Could not create runtime directory 'links': File exists
Jun 26 19:26:39 witherspoon daemon.warn systemd-networkd[1102]: Could not create runtime directory 'leases': File exists
Jun 26 19:26:39 witherspoon daemon.info systemd-networkd[1102]: eth0: Gained IPv6LL
Jun 26 19:26:39 witherspoon daemon.info systemd-networkd[1102]: Enumeration completed
Jun 26 19:26:39 witherspoon daemon.info systemd[1]: Started Network Service.
Jun 26 19:26:39 witherspoon daemon.info avahi-daemon[782]: Joining mDNS multicast group on interface eth0.IPv4 with address 172.21.140.2.
Jun 26 19:26:39 witherspoon daemon.info avahi-daemon[782]: New relevant interface eth0.IPv4 for mDNS.
Jun 26 19:26:39 witherspoon daemon.info avahi-daemon[782]: Registering new address record for 172.21.140.2 on eth0.IPv4.
```

### With VLAN 
```
root@witherspoon:/etc/systemd/network# cat 00-bmc-eth0.network
[Match]
Name=eth0

[Network]
Address=172.21.140.2/16
Gateway=172.21.254.1
VLAN=eth0.553

[DHCP]
ClientIdentifier=mac
root@witherspoon:/etc/systemd/network# cat eth0.553.netdev
[NetDev]
Name=eth0.553
Kind=vlan

[VLAN]
Id=553
```

After trying to start the networkd... got the following messages: 

```

Jun 26 19:28:19 witherspoon daemon.info systemd[1]: Started Network Service.
Jun 26 19:28:19 witherspoon daemon.info avahi-daemon[782]: Withdrawing address record for 172.21.140.2 on eth0.
Jun 26 19:28:19 witherspoon daemon.info avahi-daemon[782]: Leaving mDNS multicast group on interface eth0.IPv4 with address 172.21.140.2.
Jun 26 19:28:19 witherspoon daemon.info avahi-daemon[782]: Interface eth0.IPv4 no longer relevant for mDNS.
Jun 26 19:28:19 witherspoon daemon.err systemd-networkd[1113]: eth0: Could not join netdev: Operation not supported
Jun 26 19:28:19 witherspoon daemon.warn systemd-networkd[1113]: eth0: Failed
```

Any ideas or can you help point me in the correct direction?

mdmillerii, 2017-06-26:
Just a bystander, but a strace of systemd-networkd should show what request trigggerd the error, I would guess some kernel configs missing.

whowutwut, 2017-06-26:
@mdmillerii Thanks! Any feedback is appreciated!  Looks like it's missing that command: 
```
root@witherspoon:/etc/systemd/network# ls /usr/bin/strace
ls: /usr/bin/strace: No such file or directory
root@witherspoon:/etc/systemd/network# ls /sbin/strace
ls: /sbin/strace: No such file or directory
root@witherspoon:/etc/systemd/network# which strace
root@witherspoon:/etc/systemd/network#
```

mdmillerii, 2017-06-26:
strace is in the debug tarball image, you can unpack it to /usr/local.

bradbishop, 2017-06-27:
I'm not sure if we ship the debug tarball.  @geissonator do we put the debug tarballs anywhere?

geissonator, 2017-06-27:
debug tarball is built out on openpower.xyz with each build, can go to https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/lastSuccessfulBuild/artifact/deploy/images/witherspoon/ and grab the obmc-phosphor-debug-tarball-witherspoon.tar.xz file.

whowutwut, 2017-07-05:
After looking at this some more, I believe manually setting VLAN in the BMC is not a viable solution for xCAT.   @bradbishop  I believe after we talked, you were just going to enable the 802.1Q support into OpenBMC (1 line change) ...  regardless of whether this would be used by us or not.   Could we use this issue to track it?

ratagupt, 2017-07-07:
Once,Brad changes comes in the build,I will try and update the issue.

williamspatrick, 2017-07-07:
@bradbishop - I think you were going to drive getting the kernel config change in to support VLANs?

closed 2017-07-07 by williamspatrick

whowutwut, 2017-07-10:
@williamspatrick  Was this closed by accident?

williamspatrick, 2017-07-10:
No, it was done on purpose. You can see the linker commit that fixed the issue.

We put in the kernel support so you should no longer get the networkd error message when manually configuring a vlan.

whowutwut, 2017-07-10:
Nice! Thanks  I see that now.  Not used to the commits automatically closing the issue, sweet.