openbmc/openbmc #1850: Memory leak in phosphor-log-manager

bradbishop, 2017-06-27:
Opening based on this comment in #1844:

```
The heap size for the phosphor-log-manager process is 178671616. This is probably not ideal.
```

geissonator, 2017-10-17:
Investigate a bit, is it memory leak or just excessive memory use?  Obvious fix or should we defer?

devenrao, 2017-10-19:
Did not see any memory consumption of the phosphor-log-manager process, stack size is only increased when new errors are created. Tried various tests and captured below are the memory consumption details.

Suspecting the higher consumption of memory could be because of the number of errors created due to a failure, but with the error cap logic in place this is unlikely to happen.

Top command output at various stages
before poweron
 root     S     5984   1%   0% phosphor-log-manager

after poweron 
 root     S     6064   1%   0% phosphor-log-manager

after creating over 100 errors
 root     S     6132   1%   0% phosphor-log-manager

after deleting 100 errors
root     S     6132   1%   0% phosphor-log-manager

after stopping and restarting the service
root     S     5984   1%   6% phosphor-log-manager

closed 2019-04-26 by stale[bot]