openbmc/openbmc #1851: Ramulus system going into quiesce state after powering on.

rahulmah, 2017-06-27:
Romulus's host state is going into Quiesced State when powered on. Tested with latest BMC image (#851/v1.99.7-56-ga884877) and latest PNOR (#393/v1.17-85-g3f75086). 

BMC fw: 
```
root@romulus:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-56"
VERSION_ID="v1.99.7-56-ga884877"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-56"
BUILD_ID="v1.99.7"
```
PNOR version:
```
open-power-romulus-v1.17-85-g3f75086
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-123-g4df53124cb34
        hostboot-30db384
        linux-4.11.6-openpower1-p477f53e
        petitboot-v1.4.3-p7a792c7
        machine-xml-301279c
        occ-1dc97a6
        hostboot-binaries-24995fc
        capp-ucode-9c73e9f
        sbe-d770027
```

FFDC: 
[20170627061339318005_PowerOnTest.zip](https://github.com/openbmc/openbmc/files/1104361/20170627061339318005_PowerOnTest.zip)
[20170627061339318005_BMC_journalctl.txt](https://github.com/openbmc/openbmc/files/1104362/20170627061339318005_BMC_journalctl.txt)

mine260309, 2017-06-27:
From the log, it shows host is getting Checkstop error.
It's better to check host console log to see what's wrong, and it should be a PNOR issue.

rahulmah, 2017-06-27:
Here are the console logs from Romulus system after power on.

[Romulus_quiesce_state_console_logs.txt](https://github.com/openbmc/openbmc/files/1104437/Romulus_quiesce_state_console_logs.txt)

mine260309, 2017-06-27:
So hostboot gets error:
```
  9.96223|================================================
  9.96226|Error reported by initservice (0x0500) PLID 0x9000001B
  9.96227|  Initialization Service launched a function and the task returned an error.
  9.96228|  ModuleId   0x01 BASE_INITSVC_MOD_ID
  9.96229|  ReasonCode 0x0506 WAIT_FN_FAILED
  9.96230|  UserData1  task id or task return code : 0x0000000000000084
  9.96231|  UserData2  returned status from task : 0x0000000000000001
  9.96232|------------------------------------------------
  9.96232|  Callout type             : Procedure Callout
  9.96233|  Procedure                : EPUB_PRC_HB_CODE
  9.96234|  Priority                 : SRCI_PRIORITY_HIGH
  9.96235|------------------------------------------------
  9.96236|  host_discover_targets
  9.96236|------------------------------------------------
  9.96237|  Hostboot Build ID:
  9.96238|================================================
```
I believe this issue is not related to OpenBMC.

rahulmah, 2017-06-28:
@mine260309 : Okay. So is it a PNOR issue?

williamspatrick, 2017-06-28:
Yes, a hostboot task is crashing.  They need to debug it.

closed 2017-06-28 by williamspatrick