openbmc/openbmc #1852: P8 OCC sensor error

mine260309, 2017-06-27:
OCC sensor access is via i2c on P8 systems.
E.g. on Palmetto OpenBMC uses `/sys/class/i2c-adapter/i2c-3/3-0050/online` detect and control OCC sensor.
```
ls -l /sys/class/i2c-adapter/i2c-3/3-0050
total 0
lrwxrwxrwx    1 root     root             0 Apr 29 05:18 driver -> ../../../../../../../../bus/i2c/drivers/occ-i2c
drwxr-xr-x    3 root     root             0 Apr 29 05:17 hwmon
-r--r--r--    1 root     root          4096 Apr 29 05:18 modalias
-r--r--r--    1 root     root          4096 Apr 29 05:18 name
lrwxrwxrwx    1 root     root             0 Apr 29 05:18 of_node -> ../../../../../../../../firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@100/occ@50
-rw-r--r--    1 root     root          4096 Apr 29 05:17 online
lrwxrwxrwx    1 root     root             0 Apr 20 17:29 subsystem -> ../../../../../../../../bus/i2c
-rw-r--r--    1 root     root          4096 Apr 20 17:29 uevent
```
But this path is missing in kernel 4.10:
```
ls -l /sys/class/i2c-adapter/i2c-3/3-0050/
total 0
-r--r--r--    1 root     root          4096 Jun 27 08:06 modalias
-r--r--r--    1 root     root          4096 Jun 27 08:06 name
lrwxrwxrwx    1 root     root             0 Jun 27 08:06 of_node -> ../../../../../../../../firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@100/occ@50
drwxr-xr-x    2 root     root             0 Jun 27 08:06 power
lrwxrwxrwx    1 root     root             0 Jun 27 08:00 subsystem -> ../../../../../../../../bus/i2c
-rw-r--r--    1 root     root          4096 Jun 27 08:00 uevent
```

This causes the OCC sensor not working:
```
Jun 27 08:05:35 palmetto sensor_manager2.py[789]: Deleting OCC device
Jun 27 08:05:35 palmetto sensor_manager2.py[789]: sh: /sys/bus/i2c/devices/3-0050/online: Permission denied
Jun 27 08:05:35 palmetto sensor_manager2.py[789]: Installing OCC device
Jun 27 08:05:35 palmetto sensor_manager2.py[789]: sh: /sys/bus/i2c/devices/3-0050/online: Permission denied
```

mine260309, 2017-06-27:
I discuss this issue with @shenki and he suggests @eddiejames may have a look of this issue.

eddiejames, 2017-06-27:
There is no online attribute anymore. The driver will have to be bound by a service or manually once the system is powered on. Vishwa is handling this for P9.

mine260309, 2017-06-30:
I got @eddiejames 's confirmation that we need some user space change to use the new OCC driver for P8 as well.
@williamspatrick @geissonator @vishwabmc Could you help to plan a story (or use this issue) for implementing this?

mine260309, 2017-07-06:
For P8 systems, user space needs manually bind the new occ-hwmon driver, e.g. for Palmetto:
```
echo 3-0050 > /sys/bus/i2c/drivers/occ-hwmon/bind
```
After that, hwmon`x` will be created in `/sys/class/hwmon/`.
But the driver may have a bug that cause the readings of the hwmon items return empty.
@eddiejames is looking into the issue.

mine260309, 2017-07-07:
I will work with MSI on a solution to bind/unbind the driver for P8 systems, probably via udev rules.

williamspatrick, 2017-07-07:
Is there any reason you wouldn't be able to use the same occ-control program we are using for P9?  We should be able to enable that for P8 systems as well.

eddiejames, 2017-07-07:
On P8 its a different path for binding the driver

mine260309, 2017-07-10:
@williamspatrick As Eddie says, P8 uses different path on binding/unbinding the driver.
This makes the current https://github.com/openbmc/openpower-occ-control does not work on P8.

Specifically, on P9, the code uses below device name and driver path to bind/unbind:
```
echo occ1-dev0 > /sys/bus/platform/drivers/occ-hwmon/bind
echo occ2-dev0 > /sys/bus/platform/drivers/occ-hwmon/bind
```

While on P8, the device name should be the i2c device address, and driver path is different, e.g. on Palmetto:
```
echo 3-0050 > /sys/bus/i2c/drivers/occ-hwmon/bind
```
The i2c device address is different for different systems.

This is why we can not directly re-use the code.

williamspatrick, 2017-07-10:
Those should be constants in the code. Just add a configure.ac option and #ifdefs in the right spot. That's pretty minor compared to writing all new code.

mine260309, 2017-07-10:
Yup, we should re-use the code for P8, the only problem is how do we write correct device address (i2c bus-address) to the driver path for different systems.
One way to go is to iterate the i2c devices in sysfs and find all the occ-hwmon devices.
I'm checking if we have better options.

williamspatrick, 2017-07-10:
Prefer #ifdef over runtime checks for things that are system specific.  Ideally we can just update a couple configure.ac variables.

mine260309, 2017-07-12:
@williamspatrick I have a question.
During building OpenBMC, there shall be a way to tell if it's building for P8 or P9, so that `configure.ac` could generate different configs.

I was thinking to use a envrironment variable, but I do not see such a env that could tell if it's P8 or P9. The most close one is `MACHINE`, but it does not resolve the problem.

So probably we may need to add some config in some `.bb` (or `.bbappend`) to tell the config?
E.g. for all P8 systems, add `occ/openpower-occ-control.bbappend` with:
```
EXTRA_OECONF_append = " --enable-p8occ"
```

Do you have other options?

williamspatrick, 2017-07-12:
EXTRA_OECONF_append is the right way to do it.

I think we could create a machine feature for openpower-power8 vs openpower-power9 and have the machine feature enabled for the whole machine.  Based on the machine feature you could enable the EXTRA_OECONF directly in the openpower-occ-control.bb file instead of doing a .bbappend.  That seems better than having to do a .bbappend in each P8 machine.

@bradbishop - did I get this all right?

mine260309, 2017-07-13:
I am working with MSI on `openpower-occ-control` changes to enable i2c occ-hwmon on P8 systems.
The changes related to bitbake files will be done later.

mine260309, 2017-07-14:
Submitted `openpower-occ-control` changes to https://gerrit.openbmc-project.xyz/#/c/5303

mine260309, 2017-07-20:
* Submitted machine feature related changes to
https://gerrit.openbmc-project.xyz/#/c/5401
* Submitted Palmetto hwmon rule update changes to
https://gerrit.openbmc-project.xyz/#/c/5402/

With all above patches, P8 occ hwmon sensor will work fine, except that it requires manually setting `OccActive` properties. But this is the same situation on P9 as well.

So I think this issue can be resolved by the above changes

mine260309, 2017-10-30:
* The change to update occ sensor configs:
https://gerrit.openbmc-project.xyz/#/c/7623/
* The change to update the occ-hwmon driver for P8 systems:
http://patchwork.ozlabs.org/patch/831086/

closed 2018-01-22 by mine260309