openbmc/openbmc #1853: phosphor-fru-fault-monitor[973]: Error in mapper call [Witherspoon]

gkeishin, 2017-06-27:
Driver info:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-56"
VERSION_ID="v1.99.7-56-ga884877"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-56"
BUILD_ID="v1.99.7"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-9e95993e2d6edc68af5ca3f2345789e0f450d0db #1 Mon Jun 26 23:21:00 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

```

```

Restarted logging service 
root@witherspoon:~# systemctl restart xyz.openbmc_project.Logging.service 

And on journald

Apr 20 17:36:07 witherspoon systemd[1]: Stopping Phosphor Log Manager...
Apr 20 17:36:07 witherspoon systemd[1]: Stopped Phosphor Log Manager.
Apr 20 17:36:07 witherspoon systemd[1]: Starting Phosphor Log Manager...
Apr 20 17:36:09 witherspoon systemd[1]: Started Phosphor Log Manager.
Apr 20 17:36:10 witherspoon phosphor-fru-fault-monitor[973]: Failed to invoke ObjectMapper method
Apr 20 17:36:12 witherspoon phosphor-fru-fault-monitor[973]: Error in mapper call
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Failed to invoke ObjectMapper method
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Error in mapper call
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Failed to invoke ObjectMapper method
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Error in mapper call
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Failed to invoke ObjectMapper method
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Error in mapper call
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Failed to invoke ObjectMapper method
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Error in mapper call
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Failed to invoke ObjectMapper method
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Error in mapper call
Apr 20 17:36:13 witherspoon phosphor-fru-fault-monitor[973]: Failed to invoke ObjectMapper method

Looks like the errors are not in logging, rather the LED FRU fault monitor. This seems an incorrect reaction from FRU monitor.
```

- [x] Journald  
[journald_v1.99.7-56-ga884877_logging_service.txt](https://github.com/openbmc/openbmc/files/1105083/journald_v1.99.7-56-ga884877_logging_service.txt)

dhruvibm, 2017-06-27:
This is a race condition after error log started persisting. FRU monitor is getting the signal but object is not ready in object mapper. will make a change in FRU monitor not to request for associations from error entry when the interface added signal is received, but extract the information from the message itself.

rfrandse, 2017-07-16:
<https://gerrit.openbmc-project.xyz/5206> Fix for mapper errors during logging service restart
Resolves  #1853 phosphor-fru-fault-monitor[973]: Error in mapper call [Witherspoon]

dhruvibm, 2017-07-21:
Waiting for review

gkeishin, 2017-07-21:
Super cool.. Lets get this in..

dhruvibm, 2017-07-24:
# **STATUS**
07/24 - need review

closed 2017-07-26 by williamspatrick