openbmc/openbmc #1855: DCMI : Identification function : Implement Management Controller ID

dkodihal, 2017-06-27:
Implement the ability to set and get the Management Controller ID (or hostname).

tomjoseph83, 2017-07-07:
DCMI Specification Link
https://www.intel.com/content/dam/www/public/us/en/documents/technical-specifications/dcmi-v1-5-rev-spec.pdf

This involves the implementation of the Get/Set Management Controller identifier String command.

The management controller identifier string needs to be updated to the HostName property implemented by the SystemConfiguration.interface.yaml

https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Network/SystemConfiguration.interface.yaml

stamerlan, 2017-08-16:
Hello,

Could you please review https://gerrit.openbmc-project.xyz/#/c/6044/

rfrandse, 2017-08-25:
<https://gerrit.openbmc-project.xyz/6044> Add DCMI Get/Set Management Controller Id String
Resolves:  #1855 DCMI : Identification function : Implement Management Controller ID

closed 2017-10-16 by geissonator