openbmc/openbmc #1858: watchdog timer is set when 'false' input is provided using CURL [witherspoon]

SrideviRamesh, 2017-06-27:
Issue:

watchdog timer is set when 'false' input is provided using CURL.

Expectation:

watchdog timer should be disabled.

System Info:

cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-43"
VERSION_ID="v1.99.7-43-ga399c98"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-43"
BUILD_ID="v1.99.7"

Test Procedure:

1. Disable watchdog timer using CURL command.
2. Verify the watchdog timer settings using CURL command.
 
Log Info:

```
 curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"false\"}" -X PUT  https://XX.XX.XX.XX/xyz/openbmc_project/watchdog/host0/attr/Enabled
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}

 curl -c cjar -b cjar -k -H "Content-Typeapplication/json" -X GET https://XX.XX.XX.XX//xyz/openbmc_project/watchdog/host0 d
{
  "data": {
    "Enabled": 1,
    "Interval": 120000,
    "TimeRemaining": 112279
  },
  "message": "200 OK",
  "status": "ok"
}

curl -c cjar -b cjar -k -H "Content-Type:application/json" -X GET https://XX.XX.XX.XX//xyz/openbmc_project/watchdog/host0
{
  "data": {
    "Enabled": 1,
    "Interval": 120000,
    "TimeRemaining": 99509
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$
```
FFDC,  journal, dmesg logs are attached.
[dmesg.txt](https://github.com/openbmc/openbmc/files/1106201/dmesg.txt)
[journal.txt](https://github.com/openbmc/openbmc/files/1106202/journal.txt)
[20170627120727746501_MyFFDCLogs.zip](https://github.com/openbmc/openbmc/files/1106203/20170627120727746501_MyFFDCLogs.zip)

vishwabmc, 2017-06-28:
I had worked with @SrideviRamesh on this before and everything was right when busctl / dbus-send was used. I had indicated that it must be something with the user data. I recommend that the data be validated again.

williamspatrick, 2017-06-28:
I believe the issue is with the data being sent:
```
 curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\": \"false\"}" -X PUT  https://XX.XX.XX.XX/xyz/openbmc_project/watchdog/host0/attr/Enabled
```

The data payload here would be, in JSON, `{ "data" : "false" }`.  JSON is very funny about parsing rules since it is based on Javascript.  In this case, a string containing 'false' is like any other string and if converted to a boolean evaluates to true.

You can test this by opening the javascript console in your browser and evaluating this:
```
if ("false") { alert("This is actually true.")}
```
You will see the alert fire.

You need to either pass the literal `false` without string escapes or the literal 0.

closed 2017-06-28 by williamspatrick

gkeishin, 2017-06-29:
@SrideviRamesh could u write the piece of code in the test and run it.

vishwabmc, 2017-06-29:
Please refer the rest command used to test LED since they use true / false as indicated before.