openbmc/openbmc #1859: [Palmetto] Segmentation Fault(core dumped)

swe12345, 2017-06-27:
Core dumped on checking version while updating to the latest PNOR Build
```
root@palmetto:~# pflash -r /tmp/out.txt -P VERSION
FFS: Flash header not found. Code: 100
Segmentation fault (core dumped)
```
```
root@palmetto:~# pflash -E -f -p /tmp/palmetto.pnor
About to erase chip !
Erasing... (may take a while !) done !
About to program "/tmp/palmetto.pnor" at 0x00000000..0x02000000 !
Programming & Verifying...
[==================================================] 100% ETA:0s
root@palmetto:~#
```
```
root@palmetto:~# hexdump -C -n 5 /dev/mtd6
00000000  00 00 00 01 00                                    |.....|
00000005
root@palmetto:~#
```
```
root@palmetto:~# hexdump -C -n 50 /dev/mtd6
00000000  00 00 00 01 00 00 00 01  00 00 00 80 00 00 00 19  |................|
00000010  00 00 10 00 00 00 20 00  00 00 00 00 00 00 00 00  |...... .........|
00000020  00 00 00 00 00 00 00 00  50 41 62 cd 70 61 72 74  |........PAb.part|
*
00000030
```
```
root@palmetto:~# flash_eraseall /dev/mtd6
Erasing 64 Kibyte @ 4000000 - 100% complete.

root@palmetto:~# hexdump -C -n 50 /dev/mtd6
00000000  00 00 00 01 00 00 00 01  00 00 00 80 00 00 00 19  |................|
00000010  00 00 10 00 00 00 20 00  00 00 00 00 00 00 00 00  |...... .........|
00000020  00 00 00 00 00 00 00 00  50 41 62 cd 70 61 72 74  |........PAb.part|
```

mdmillerii, 2017-06-27:
@swe12345 was the program and erase after or before the version check core dumped?

swe12345, 2017-06-28:
@mdmillerii Erased After the version check core dump

legoater, 2017-06-28:
@cyrilbur-ibm, I think they were some recent fixes in pflash. Does the above crash rings a bell ? Thanks,

legoater, 2017-06-28:
pflash does not handle correctly malformed partition headers. Mainlline has a fix for that.

legoater, 2017-06-28:
After a reboot, the pnor can be accessed again.  this is the same issue as reported in  https://github.com/openbmc/openbmc/issues/1638

closed 2017-06-28 by williamspatrick