openbmc/openbmc #1860: add error log support to PDM

bradbishop, 2017-06-27:
Add support for creating an error log in phosphor-dbus-monitor.

msbarth, 2017-06-27:
Along with add support for creating an error log, phosphor-dbus-monitor should also be enhanced to allow a configurable option to mark error logs as "resolved" when a defined error log event that occurred is corrected. i.e.) faulted fan -> error log -> fan replaced -> mark "resolved"

gtmills, 2017-06-29:
- [ ] Create an error log 
- [ ] Make error log as resolved using https://github.com/openbmc/phosphor-logging/blob/707a3e1b4d224ca273c67fcc1dba2a21d6d8e2c5/elog_entry.hpp#L102

bradbishop, 2017-06-29:
I don't think the function @msbarth is looking for is containable in a single sprint, so I'd vote for another story for that.

Its also delving into custom logic which PDM frankly isn't super well equipped to express.  I'm starting to question having PDM make the error logs at all.

rfrandse, 2017-08-15:
<https://gerrit.openbmc-project.xyz/6015> Add Witherspoon Power Supply policy
Resolves  #1860 add error log support to PDM

closed 2017-08-30 by williamspatrick