openbmc/openbmc #1863: User power cap boundary validation

geissonator, 2017-06-27:
The user power cap has a lot of nuances.  From the interface specification:
```
•	Prior to any communication with the OCC, the BMC will have a default min/max power limit from the configuration file that must cover all power configuration settings.
•	On the first poll with the OCC the BMC must update the min/max power limit that the master OCC provides in the “CAPS” sensor section of poll response.  In addition, in the case that the current power limit set is now out of bounds from the new min/max power limit being reported from the OCC the BMC must clip the current power limit to be min or max.
```
#1771 is the story for the new settingsd code to verify limits.  The ideal solution here is for us to write a scrip that parses the MRW for these 2 values and puts them in as the defaults for the user power cap.
```
<attribute>
	<id>OPEN_POWER_N_PLUS_ONE_BULK_POWER_LIMIT_WATTS</id>
	<default>2375</default>
</attribute>
<attribute>
	<id>OPEN_POWER_SOFT_MIN_PCAP_WATTS</id>
	<default>1945</default>
</attribute>
```
It's not clear to me if these are good enough or if we still need to look at the OCC poll response and dynamically adjust these based on that.  That will need to be determined in this story.

williamspatrick, 2017-06-28:
Are you suggesting we would need to dynamically alter the limits we allow a user to put in place?  That sounds excessive.

geissonator, 2017-07-05:
Yes, in the past, code would do an inventory of the system (number of dimms, cores, io) and dynamically calculate what the power min/max boundaries should be.  Not clear to me if that's happening on our OP systems though.

gtmills, 2017-09-19:
- [ ] enforce boundaries here https://github.com/openbmc/openpower-occ-control/blob/6ac874e1a1752e29ef4756d9c7e7d33baee9f9d5/powercap.hpp

rfrandse, 2017-10-05:
<https://gerrit.openbmc-project.xyz/7079> Add power cap boundary validation
Resolves  #1863 User power cap boundary validation

closed 2017-10-24 by geissonator