openbmc/openbmc #1865: Romulus: host console hangs with VUART mirroring

mine260309, 2017-06-28:
On Romulus, the host console is mirrored to physical UART (`/dev/ttyS0`) by default, which is configured in `/etc/obmc-console.conf`
```
local-tty = ttyS0
local-tty-baud = 115200
```
It was not working because `obmc-console-server` gets erros:
```
Jun 25 21:41:57 romulus obmc-console-server[848]: obmc-console-server: Write error: Resource temporarily unavailable
Jun 25 21:41:57 romulus obmc-console-server[848]: obmc-console-server: Write error: Resource temporarily unavailable
```

The patch series of https://gerrit.openbmc-project.xyz/#/c/2316 ~ 2322 should fix the issue. And I tested the patches before and it worked fine.

After version bump of `obmc-console.bb` (29a4337), the host console mirroring is expected to work.
However, the mirroring does not work, and the host console hangs after producing a certain amount of log.

mine260309, 2017-06-28:
If I remove the mirroring config:
```
#local-tty = ttyS0
#local-tty-baud = 115200
```
Then host console works fine.

jk-ozlabs, 2017-06-28:
Hi @mine260309 , you mentioned that:

> And I tested the patches before and it worked fine.

What is different between that test scenario and this situation?

mine260309, 2017-06-28:
When it worked fine, it's months ago, so the whole build is different.
It worked fine, that the host console (`obmc-console-client`) works fine, and if I connect to the physical UART, it shows the host console as well and it's synced with the VUART.
It's working on 9600 baut rate so the physical UART was slow, but it's working well.

jk-ozlabs, 2017-07-03:
I've been doing some debug on https://github.com/openbmc/openbmc/issues/1882 , and it looks like this is related to the physical UART being held in reset. @shenki is working on a fix for that, which I believe will be applicable to romulus too.

mine260309, 2017-07-05:
Confirmed that https://github.com/openbmc/linux/commit/990f06c35b7e119f00967e90f0514dbbd2e1b51b fixed the issue.

closed 2017-07-05 by mine260309