openbmc/openbmc #1866: [Barreleye] Physical LEDs are not available with recent openbmc build

swe12345, 2017-06-28:
Physical LEDs are not available with recent openbmc build. 
Tested with latest openbmc build v1.99.7-61-gb42df4d.

Driver Details:
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-61"
VERSION_ID="v1.99.7-61-gb42df4d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-61"
BUILD_ID="v1.99.7"
```
```
root@barreleye:~# cat /etc/timestamp
20170628031923
```
Nothing is seen on /sys/class/leds on BMC :
```
root@barreleye:~# cd /sys/class/leds/
root@barreleye:/sys/class/leds# ls
root@barreleye:/sys/class/leds#
```
```
curl -c cjar -b cjar -k -H "Content-Type:application/json" -X GET https://9.3.23.32/xyz/openbmc_project/led/physical/
{
  "data": {
    "description": "org.freedesktop.DBus.Error.FileNotFound: path or object not found: /xyz/openbmc_project/led/physical"
  },
  "message": "404 Not Found",
  "status": "error"
```
[20170628020900451719_TestsTestLed.zip](https://github.com/openbmc/openbmc/files/1108530/20170628020900451719_TestsTestLed.zip)

vishwabmc, 2017-06-28:
I looked at the system and no LEDs were populated in `/sys/class/leds`

Was there any device tree change for Barreleye ?

williamspatrick, 2017-06-28:
Device tree entries seem to still be there for Barreleye.

```
	leds {
		compatible = "gpio-leds";

		heartbeat {
			gpios = <&gpio ASPEED_GPIO(R, 17) GPIO_ACTIVE_HIGH>;
		};

		identify {
			gpios = <&gpio ASPEED_GPIO(R, 2) GPIO_ACTIVE_LOW>;
		};

		beep {
			gpios = <&gpio ASPEED_GPIO(N, 7) GPIO_ACTIVE_HIGH>;
		};
	};
```

williamspatrick, 2017-06-28:
@shenki - Can you help debug what is going on?

shenki, 2017-06-28:
I booted the latest Barreleye master build (9910bce302e464) in qemu:
```
qemu-system-arm -m 256 -M palmetto-bmc -nodefaults -nographic -serial stdio -drive file=~/Downloads/image-bmc,format=raw,if=mtd
```
in dmesg, we see
```
[    1.540000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin B12 already requested by 1e660000.e
thernet; cannot claim for 1e780000.gpio:445
[    1.540000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-153 (1e780000.gpio:445) status -22
[    1.540000] leds-gpio: probe of leds failed with error -22
```
The Ethernet driver has requested one of the pins that we are trying to use as a LED. That's strange. I will continue investigating.

shenki, 2017-06-28:
B12 is `GPIOT1` and `RGMII 1 transmit control`.

shenki, 2017-06-29:
With debugging turned on:

```
[    1.100000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 153 (B12) for 1e780000.gpio:445
[    1.100000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin B12 already requested by 1e660000.ethernet; cannot claim for 1e780000.gpio:445
[    1.100000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: pin-153 (1e780000.gpio:445) status -22
[    1.100000] leds-gpio: probe of leds failed with error -22
```
The device tree dumped:
```
    leds {
        compatible = "gpio-leds";
        heartbeat {
            gpios = <0x0000001f 0x00000099 0x00000000>;
        };
        identify {
            gpios = <0x0000001f 0x0000008a 0x00000001>;
        };
        beep {
            gpios = <0x0000001f 0x0000006f 0x00000000>;
        };
    };
```

| label | GPIO number | GPIO |
| --- | --- | --- |
| heartbeat | 153 | GPIO T1 |
| identify | 138 | GPIO R2 |
| beep |111 | GPIO N7 |

The device tree entry is wrong. Instead of GPIOR17, we are after GPIO number 140. Should be GPIO R4.

Looking back through the git history, these are the numbers before we started using the ASPEED_GPIO macro:

| label | GPIO number | GPIO |
| --- | --- | --- |
| heartbeat | 140 | GPIO R4 |
| identify | 58 | GPIO H2 |
| beep | 111 | GPIO N7 |

shenki, 2017-06-29:
I created a patch to fix a few different issues with Barreleye

 http://patchwork.ozlabs.org/patch/782034/

With this the driver correctly probes:
```
# ls /sys/class/leds/
beep       heartbeat  identify   power
```
It would be good if someone could look at a Barreleye and check that the correct LEDs turn on/off as they should.

closed 2017-07-05 by williamspatrick