openbmc/openbmc #1869: Need a way for parsing ESEL messages (/tmp/esel) after hardware or hostbooting failed

yadron, 2017-06-28:
For sure, one of the main BMC functions is analyses after hardware failed. For example, if once server failed to boot, then information stored at BMC and can be analyzed later.

I've tried finding and discussing if similar exists in OpenBMC, bot looks like currently doesn't. Probably the binary (blob) of ESEL message from PNOR (hostboot) or hardware fail will be stored, but no way to parse it.

I suggest that apart from regular OpenBMC error SEL messages like " Fan #0x37 | Upper Critical going high" (that is going to be generated by OpenBMC itself) similar messages should be recorded for each ESEL received by IPMID daemon.

User case. If failed to boot host, then user uses BMC for diagnostics what failed. Both with saved blob of original /tmp/esel some IDs should be saved.
For example, yesterday I've seen such a message during booting host failed (that is output to console/VGA/UART or somewhere during booting host):
"32.77123|  ModuleId   0x61 VPD::VPD_SPD_GET_KEYWORD_VALUE 32.77123|  ReasonCode 0x1d07 VPD::VPD_INVALID_BASIC_MEMORY_TYPE".
The user possibly cannot read this output, so the only place where it is stored is BMC.
If no words "VPD_INVALID_BASIC_MEMORY_TYPE" in events log of BMC, then user cannot understand himself that problem related to memory and calls support for any question.
If many servers sold, do you expect that user cannot prior-diagnose the problem himself or via phone with support and each time has to send /tmp/esel blob to support even for simple cases?

So from my point of view /tmp/esel must be parsed and if not text message, then IDs (like "INVALID_BASIC_MEMORY_TYPE") should be extracted and added to the list of OpenBMC/SEL messages.

I've started analyses how to implement this.
First, it could be developed separate tool for parsing /tmp/esel messages at OpenBMC.
The tool probably could be developed without problems, but the problem for me where to obtain IDs to messages (*.h) info?
I've tried fast search of "INVALID_BASIC_MEMORY_TYPE" message everywhere (like "op-build$ find ./ -type f -name "*" -exec grep INVALID_BASIC_MEMORY_TYPE \{\} \;" ), but have not found it yet.

The info how to parse /tmp/ESEL binary format itself (without knowing IDs of messages):

https://github.com/open-power/hostboot/blob/master/src/build/debug/eSEL.pl
- the problem with eSEL.pl is that it uses some binary programs for actual decoding and this binary not available there.

Ok, I found the answer to my question "how to parse binary ESEL generated by hostboot/skiboot/sbe/etc boot-loaders of PNOR":

```
https://github.com/open-power/skiboot/blob/master/hw/ipmi/ipmi-sel.c

struct sel_record and record_type == df for AMI
(means "#define SEL_REC_TYPE_AMI_ESEL 0xDF") and I see that my /tmp/esel has df and
generator_id == SEL_GENERATOR_ID_AMI 0x2000.

Also it looks like after sel_record it is going to be ELOG_SID_PRIVATE_HEADER
(value 4850 that is 5048 of https://github.com/open-power/skiboot/blob/master/include/pel.h ,
ELOG_SID_PRIVATE_HEADER = 0x5048 , and the code
that fills this part is https://github.com/open-power/skiboot/blob/master/core/pel.c
```

-- so it looks quite simple parsing ESEL binary knowing how it was compiled. The only problem for me that no info regarding all of possible ID values, so where to get this info?

I've heard from OpenBMC team that "not sure if there even is a tool to decode the eSEL.  I think if there were one it would be in the hostboot repository." and "ESEL is an opaque binary blob currently in OpenBMC".

Other simple case: many times I've seen different messages like "MAEB" during hostboot, host boot failed and no way to find info what is "ERROR_IN_MAEB" (ex. https://github.com/openbmc/openbmc/issues/1518 ). So if no parse-esel-tool, then still need at least the complete list of possible messages that non-existent now too.

Can I ask for such a parse-ESEL tool being included to OpenBMC 2.0 and list of "ERROR_IN_MAEB|INVALID_BASIC_MEMORY_TYPE|..." that is usual troubleshooting list error-messages?

I've found VPD_INVALID_BASIC_MEMORY_TYPE code here, but still searching for MAEB (cannot find MAEB anywhere):

Ok for VPD_INVALID_BASIC_MEMORY_TYPE:
```
grep VPD_INVALID_BASIC_MEMORY_TYPE ./op-build/output/build/hostboot-695bd891343faf1f0ef85fe53148590e58239efd/src/include/usr/vpd/vpdreasoncodes.H
    VPD_INVALID_BASIC_MEMORY_TYPE       = VPD_COMP_ID | 0x07,
```

, but no info for MAEB:

`op-build]$ find ./ -type f -name "*" -exec grep MAEB \{\} \;
`
- no results.

Where to get info about error of host-booting
```
"  1.75122|  ModuleId   0x0a FSI::MOD_FSIDD_CHECKFORERRORS
  1.75123|  ReasonCode 0x0a11 FSI::RC_ERROR_IN_MAEB"
```
?

yadron, 2017-06-28:
Continuing question. During discussing with OpenBMC team, it was noted that

"how parsing ESEL can be one of the main functions of the OpenBMC.  ESELs are host firmware proprietary.  They contain lots and lots of information that only firmware producer really knows how to read and understand."
- however many ESEL messages generated by software (including PNOR boot-loaders sbe/skiboot/etc), so I don't agree that it always proprietary. And from what I see no other messages (SEL?) being generated when failed at PNOR boot-loaders.

Other note:
"The main item we are missing is that we can use the corresponding SEL that comes with the ESEL to identify the part-at-fault."

For two cases that I've described above, looks like no other SEL messages and only ESEL existed.

BTW, if not only ESEL, but both SEL message exists after hardware failed, then where to get it from?

Simple example how ESEL/SEL could be displated in BMC:
https://docs.oracle.com/cd/E19121-01/sf.x4440/820-3067-14/dimms.html

dkodihal, 2018-04-10:
@mine260309 pointed me to this issue. I had a proposal about this here - https://lists.ozlabs.org/pipermail/openbmc/2018-March/011084.html. I have someone on my team that implemented this according to the proposal in the email list. We'll put that out on Gerrit in a day or two and will add you to the review.

For now the (not reviewed but working) code is in a forked github :

https://github.com/sampmisr/openpower-dbus-interfaces/commit/17510616645df4fcfbbab40701bb1239bb917293

https://github.com/sampmisr/phosphor-logging/commit/6ddb10c9cf47a9b98641fc3181a590e9a7cd498a

dkodihal, 2018-04-18:
The proposed plan is that the existing REST API to retrieve OpenBMC error logs, will also give you back a parsed eSEL (if there was one).

closed 2019-04-27 by stale[bot]