openbmc/openbmc #1871: Witherspoon Poweron fails after starting fsi-disable.service

yadlapati, 2017-06-28:
Using master Jenkins build to test openfsi changes on w42
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-9e95993e2d6edc68af5ca3f2345789e0f450d0db #1 Wed Jun 28 03:20:58 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-61"
VERSION_ID="v1.99.7-61-gb42df4d"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-61"
BUILD_ID="v1.99.7"

Following sequence is giving error:

root@witherspoon:~#  systemctl start obmc-chassis-poweron@0.target
root@witherspoon:~#   systemctl start fsi-disable.service  
root@witherspoon:~# obmcutil poweroff
root@witherspoon:~#  systemctl start obmc-chassis-poweron@0.target
A dependency job for obmc-chassis-poweron@0.target failed. See 'journalctl -xe' for details.

Apr 20 17:35:08 witherspoon systemd[1]: Started Start Power0.
Apr 20 17:35:08 witherspoon systemd[1]: Reached target Power0 On.
Apr 20 17:35:08 witherspoon systemd[1]: Reached target Power0 (On).
Apr 20 17:35:08 witherspoon systemd[1]: Started Phosphor Fan Presence Tach Daemon.
Apr 20 17:35:08 witherspoon systemd[1]: Starting Scan FSI devices...
Apr 20 17:35:08 witherspoon systemd[1]: Started Phosphor Fan Control Initialization.
Apr 20 17:35:08 witherspoon systemd[1]: Started Scan FSI devices.
Apr 20 17:35:08 witherspoon systemd[1]: Starting Run VCS workaround on host0...
Apr 20 17:35:08 witherspoon openpower-proc-control[1551]: Failed to open the device.
Apr 20 17:35:09 witherspoon systemd[1]: vcs_workaround@0.service: Main process exited, code=exited, status=255/n/a
Apr 20 17:35:09 witherspoon systemd[1]: Failed to start Run VCS workaround on host0.
Apr 20 17:35:09 witherspoon systemd[1]: Dependency failed for Chassis0 (Power On).
Apr 20 17:35:09 witherspoon systemd[1]: obmc-chassis-poweron@0.target: Job obmc-chassis-poweron@0.target/start failed with result 'dependency'.
Apr 20 17:35:09 witherspoon systemd[1]: obmc-chassis-poweron@0.target: Triggering OnFailure= dependencies.
Apr 20 17:35:09 witherspoon systemd[1]: vcs_workaround@0.service: Unit entered failed state.
Apr 20 17:35:09 witherspoon systemd[1]: vcs_workaround@0.service: Failed with result 'exit-code'.
Apr 20 17:35:09 witherspoon systemd[1]: Stopping Phosphor Fan Control Initialization...
Apr 20 17:35:09 witherspoon systemd[1]: Stopping Phosphor Fan Presence Tach Daemon...

geissonator, 2017-06-28:
Something not quite right Chris with the new fsi-enable/fsi-disable service we put in to dynamically switch the gpio's.

cbostic, 2017-06-28:
Which change is thought to have introduced this?

geissonator, 2017-06-28:
The new stuff from @shenki I assume?  Where we moved the gpio switching to the fsi enable/disable services.  Uma was testing that out for us per last night's interlock.

cbostic, 2017-10-16:
@yadlapati  Is this still an issue?  I believe this was taken care of some time ago.

cbostic, 2017-10-16:
Marking as 'Phase 7' for now.

closed 2019-04-27 by stale[bot]