openbmc/openbmc #1872: Systemd security vulnerability

williamspatrick, 2017-06-28:
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2017-9445

amboar, 2017-06-30:
Can we update the title to include `systemd-resolved` and the CVE number? It's pretty vague at the moment.

joseph-reynolds, 2018-05-24:
From the CVE: "In systemd through 233, certain sizes passed to dns_packet_new in systemd-resolved can cause it to allocate a buffer that's too small, etc."
From the openbmc repo, I found: import-layers/yocto-poky/meta/recipes-core/systemd/systemd_234.bb which seems to pull in resolved (but I don't yet fully understand bitbake).  Does this mean we are already at the 234 level of systemd_resolved where this problem is fixed?

amboar, 2018-05-30:
@joseph-reynolds yes, we're running systemd 234 on master.

closed 2018-06-01 by williamspatrick