openbmc/openbmc #1873: Romulus: Add IR35221 hwmon config

mine260309, 2017-06-29:
Similar to Witherspoon, Romulus uses IR35221 for VCS/VDD and VDN on i2c bus 4/5, address 0x70, 0x73.
This task is to add IR35221 related hwmon configs.

mine260309, 2017-06-29:
On Romulus, when host is not powered on, IR35221 chips can not be accessed.
I tried to add IR35221 in Romulus dts but it fails to probe and init:
```
Jun 27 21:14:42 romulus kernel: ir35221 4-0070: Failed to read PMBUS_MFR_ID
Jun 27 21:14:42 romulus kernel: ir35221: probe of 4-0070 failed with error -110
Jun 27 21:14:42 romulus kernel: ir35221 4-0073: Failed to read PMBUS_MFR_ID
Jun 27 21:14:42 romulus kernel: ir35221: probe of 4-0073 failed with error -110
Jun 27 21:14:42 romulus kernel: ir35221 5-0070: Failed to read PMBUS_MFR_ID
Jun 27 21:14:42 romulus kernel: ir35221: probe of 5-0070 failed with error -110
Jun 27 21:14:42 romulus kernel: ir35221 5-0073: Failed to read PMBUS_MFR_ID
Jun 27 21:14:42 romulus kernel: ir35221: probe of 5-0073 failed with error -110
```
And the driver enters in a strange state that it can not unbind/bind the devices.

So we need to find a proper way to use this driver, e.g. to instantiate the devices when host powers on, and delete the devices before host powers off.
The scripts that set VCS/VDD/VDN need to be considered as well.

mine260309, 2017-06-29:
One possible solution:
1. Do not add IR35221 to dts;
2. When host powers on, after `avsbus-enable@.service` which is the last script to use IR35221 chips, instantiate the devices;
3. When host powers off, before `op-wait-power-off@.service`, delete the devices.

mine260309, 2017-06-29:
The above solution itself works fine, that the hwmon devices can be created and deleted on system power on/off.
But there is a problem, that the devices lack `OF_FULLNAME`, which causes the udev does not trigger the wanted service.

A little details: we have udev rule for hwmon:
```
cat /lib/udev/rules.d/70-hwmon.rules
SUBSYSTEM=="hwmon", ACTION=="add", TAG+="systemd", ENV{SYSTEMD_WANTS}+="xyz.openbmc_project.Hwmon@%E{OF_FULLNAME}.service"
```
When a hwmon device is added, it will start `xyz.openbmc_project.Hwmon@%E{OF_FULLNAME}.service`. If the device is not in device tree or in device tree but disabled, it does not have `OF_FULLNAME`, and so udev will not start the corresponding service.

mdmillerii, 2017-06-29:
@shenki this sounds like we need a solution for "switch power" devices .

I can see a regulator-supply modeling power to these devices, but we would still need a more generic solution to adding i2c devices becoming available. Maybe a i2c mux concept that uses the power supply as an enable?

mine260309, 2017-06-30:
I tried to workaround the above issue by:
1. Modify the udev rule, adding `IMPORT{parent}` so that the device gets property of `DEVPATH` pointing to `/devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a180.i2c-bus/i2c-4/4-0070`
2. Modify the hwmon config file to the new path

Then the `xyz.openbmc_project.Hwmon@%E{DEVPATH}.service` will start.
Unfortunately, the service will call `phosphor-hwmon-readd`, which requires `of_node` to exist in `/sys/class/hwmon/hwmon0`. But in such case there is no `of_node` dir so it fails with exception:
```
Jun 30 00:25:58 romulus phosphor-hwmon-readd[1167]: terminate called after throwing an instance of 'std::experimental::filesystem::v1::__cxx11::filesystem_error'
Jun 30 00:25:58 romulus phosphor-hwmon-readd[1167]:   what():  filesystem error: cannot canonicalize: No such file or directory [/sys/class/hwmon/hwmon0/of_node] [/]
```

So without `of_node` (a.k.a without the enabled device in dts), it won't work anyway.

gtmills, 2018-10-12:
@mine260309 Still a problem?

mine260309, 2018-10-15:
Yup, still an issue, but no one is working on this.
Close it and mark this not implemented.

closed 2018-10-15 by mine260309