openbmc/openbmc #1880: Milestone: XCAT 3

bradbishop, 2017-06-29:
End of July.

sivassrr, 2017-06-30:
@bradbishop What are all changes can we expect here.  Let know if there is any recommended Test Cases as well. That helps and thank you @bradbishop

sivassrr, 2017-06-30:
Are we going to consider rvitals as well.

whowutwut, 2017-07-05:
@sivassrr  xCAT team has implemented support for `rvitals` as a generic pass through, so anything new added to the REST response should be displayed by the xCAT `rvitals ` command.   I think that can be pushed to lower priority. Is there something specific you were asking about when you say `rvitals`?

whowutwut, 2017-07-05:
For this milestone, I'd like to push the following minimum set of functions to be delivered to the top: 

- [ ] VLAN Tagging support inband
- [ ] Ability to flash firmware (OpenBMC & PNOR) using REST commands

If time persists:  
- [ ] Exposing the [Network](https://github.com/openbmc/phosphor-dbus-interfaces/tree/master/xyz/openbmc_project/Network) API into the build

closed 2017-11-08 by bradbishop