openbmc/openbmc #1881: On very first REST call, command take a long time

whowutwut, 2017-06-29:
xCAT is using the REST interfaces to control the P9 boxes and we have been seeing intermittent behavior where our commands take a long time and then running it again, it's very quick.   I think we started seeing this when we began updating the firmware but I could not pinpoint it.  

With the new ITC machines coming in, right from MFG, when power is applied to the PSU, and the OpenBMC OS loads, I re-created this issue.  

Node is connected to power and gets a DHCP IP Address:
```
Jun 29 15:53:06 fs3 dhcpd: DHCPDISCOVER from 70:e2:84:14:29:1b via enP3p3s0
Jun 29 15:53:07 fs3 dhcpd: DHCPOFFER on 172.20.253.107 to 70:e2:84:14:29:1b (witherspoon) via enP3p3s0
Jun 29 15:53:08 fs3 dhcpd: DHCPREQUEST for 172.20.253.107 (172.20.253.27) from 70:e2:84:14:29:1b (witherspoon) via enP3p3s0
Jun 29 15:53:08 fs3 dhcpd: DHCPACK on 172.20.253.107 to 70:e2:84:14:29:1b (witherspoon) via enP3p3s0
```

I wait 2 minutes after getting the DHCP IP address to make sure that things are booting...  
```
[root@fs3 UT]# sleep 120
date
[root@fs3 UT]# date
Thu Jun 29 15:55:47 EDT 2017
```

Now run a rest call to login... 1st takes 11.6 seconds, 2nd call takes under .3 seconds. 

```
[root@fs3 UT]# time curl -c cjar -k -X POST -H "Content-Type: application/json" -d '{"data": [ "root", "XXXXXXX" ] }' https://172.20.253.107/login;
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}
real    0m11.634s
user    0m0.120s
sys     0m0.020s
[root@fs3 UT]# time curl -c cjar -k -X POST -H "Content-Type: application/json" -d '{"data": [ "root", "XXXXXXX" ] }' https://172.20.253.107/login;
{
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}
real    0m0.298s
user    0m0.109s
sys     0m0.030s
```

FYI We find we can also re-create this when we flash the firmware.  

 Is there something that isn't pre-loaded or loaded on the first run that should be preloaded to help performance? After the first command is run, everything returns quickly.

williamspatrick, 2017-06-29:
This is somewhat by design.  We use 'socket activation' to start network based processes like the REST server.  This makes it so that someone doesn't pay a penalty for the boot time for an interface they are not using (ex. starting the IPMI server in an environment that doesn't use IPMI).

whowutwut, 2017-06-30:
@williamspatrick  but unless full IPMI 2.0 backwards compatibility is in plan for GA, there's been strong push to always use the REST interfaces...  so why wouldn't REST be started by default and like you said...  leave IPMI server off? 

My concern here is we don't want to prerun a command to prime the servers.  When we discover thousands of nodes, we scan the network for the DHCP IP addresses and then attempt to do a REST call to get info back from the servers.  Scanning the network already takes it's share of time, and now it's compounded by the need for each BMC to start the REST server.    In a perfect world, it would only add constant 11 seconds and no one would notice anything off...  but things never go as plan. .  

Also, issue comes back when the firmware is flashed, maybe on power outages too?

whowutwut, 2017-06-30:
Thinking some more...  What is the penalty on boot time to start the REST service and what's the use case where we need the boot of the BMC to be so quick..     Say power is applied to the data center... The BMC will just be waiting there for the command to power on the server.  Maybe the admins need to do some diagnostics checking before powering on the whole data center,....  why wait until the command actually comes to start the REST server?

williamspatrick, 2017-06-30:
@whowutwut - In the use-case of Witherspoon at CORAL, it might be reasonable for us to forgo the normal socket activation and always start the REST server.  We should probably also look into why it takes 11 seconds to startup.

It is just a general practice with systemd to defer everything until the last moment.  They have 'dbus activation', 'socket activation', etc.  As a general practice items like that allow a Linux system with systemd to boot up much faster than a traditional sysvinit system.  We currently have all of our external interfaces triggered via socket-activation to defer the cost (and memory impact) of starting them unless they are actually used.

mdmillerii, 2017-06-30:
The mention of "fresh from mfg" and "after a flash" makes me think we are paying an additional penalty to compile python code when we clear the read-write file system (or when the code is changed).

@williamspatrick I remember talk of exploring shipping pyc with our remaining python code.

whowutwut, 2017-07-19:
Making a cross reference to the xcat issue that shows the effect of this...  https://github.com/xcat2/xcat-core/issues/3336 Then from the xcat project we can see the status of this issue if anyone searches it.

whowutwut, 2017-08-15:
I still think this is a particularly bad design ...  If I set the BMC to DHCP, and reboot it.   The BMC gets a DHCP IP address and then the first call that we make to discover this BMC hits this 30+ seconds required kick off the rest API service...

```
====================================================
[Date]       2017-08-15 14:06:40
[ClientType] cli
[Request]    bmcdiscover --range 172.12.254.102 -z
[Response]
node-70e28414291b:
        objtype=node
        groups=all
        bmc=172.12.254.102
        cons=openbmc
        mgt=openbmc
        mtm=01DH118

[ElapsedTime] 49 s
====================================================
[Date]       2017-08-15 14:07:44
[ClientType] cli
[Request]    bmcdiscover --range 172.12.254.102 -z
[Response]
node-70e28414291b:
        objtype=node
        groups=all
        bmc=172.12.254.102
        cons=openbmc
        mgt=openbmc
        mtm=01DH118

[ElapsedTime] 2 s
```

So the subsequent calls completes really quicky, 2 seconds. 

So here again, if the REST method is the primarily support way to interface with OpenBMC, why do we not just start the service so if the BMC is idling.. for 1 minute before start to discover it, we don't pay this penalty.

geissonator, 2017-08-19:
Noticed this myself today with the latest code, we do need to dig into this 10s+ startup time (or just start by default on certain systems as Victor has requested).

Basically my first call started at 58:58 and returned 14s later at 59:12.  The same command the next time took less then a second.
```
Aug 03 15:58:58 witherspoon systemd[1]: Started Phosphor Webserver.
Aug 03 15:59:12 witherspoon phosphor-gevent[2989]: ::8019:6276:8e91:f402%1995250460 - - [2017-08-03 15:59:12] "POST /login HTTP/1.1" 200 320 0.096605
Aug 03 15:59:12 witherspoon phosphor-gevent[2989]: ::8019:6276:8e91:f402%1995250460 - - [2017-08-03 15:59:12] "GET /xyz/openbmc_project/logging/entry/ HTTP/1.1" 200 484 0.055308
Aug 03 15:59:24 witherspoon phosphor-gevent[2989]: ::8019:6276:8e91:f402%1995250460 - - [2017-08-03 15:59:24] "POST /login HTTP/1.1" 200 320 0.032550
Aug 03 15:59:24 witherspoon phosphor-gevent[2989]: ::8019:6276:8e91:f402%1995250460 - - [2017-08-03 15:59:24] "GET /xyz/openbmc_project/logging/entry/ HTTP/1.1" 200 484 0.046795

```

bradbishop, 2017-08-23:
I think the problem is the two minute delay after ping response.  Can we just get rid of that?

geissonator, 2017-08-23:
@bradbishop - even the 14s delay I see above seems a bit high right?  That was just a system at bmc ready state and me issuing a REST command.  Can we optimize that?

whowutwut, 2017-08-23:
@bradbishop the 2 minute delay was just what I had run to give the BMC some time to boot, maybe too short, but the point was that even waiting 2 minutes or more.  Still had some delay.  

Brad is going to update this after we talked on the phone so will just wait for that.

closed 2019-04-27 by stale[bot]