openbmc/openbmc #1882: Host SOL console stopped working on OS reboot [Witherspoon]

gkeishin, 2017-06-30:
Driver Info:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-70"
VERSION_ID="v1.99.7-70-gce19185"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-70"
BUILD_ID="v1.99.7"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-9e95993e2d6edc68af5ca3f2345789e0f450d0db #1 Thu Jun 29 22:26:25 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

```

Test:
```
1. Boot to OS
2. Trigger reboot from OS
3. OS shutdown 
4. Host auto powered on
5. Boot back to OS (but SOL console stopped logging)
```
The Host SOL via port 2200 and IPMI both are frozen

- [x] Port 2200 SOL console log  
[SOL_Reboot_freeze.txt](https://github.com/openbmc/openbmc/files/1114183/SOL_Reboot_freeze.txt)

- [x] IPMI SOL also stopped working
-bash-4.1$ ipmitool -I lanplus -C 3 -P 0penBmc -H xx.xx.xx.xx sol activate
[SOL Session operational.  Use ~? for help]

- [x] OS is booted and pinging and can be logged in
```
-bash-4.1$ ping -c 5 xx.xx.xx.xx
PING xx.xx.xx.xx (xx.xx.xx.xx) 56(84) bytes of data.
64 bytes from xx.xx.xx.xx: icmp_seq=1 ttl=63 time=0.184 ms
64 bytes from xx.xx.xx.xx: icmp_seq=2 ttl=63 time=0.159 ms
64 bytes from xx.xx.xx.xx: icmp_seq=3 ttl=63 time=0.151 ms
64 bytes from xx.xx.xx.xx: icmp_seq=4 ttl=63 time=0.153 ms
64 bytes from xx.xx.xx.xx: icmp_seq=5 ttl=63 time=0.151 ms

--- xx.xx.xx.xx ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 3999ms
rtt min/avg/max/mdev = 0.151/0.159/0.184/0.018 ms
-bash-4.1$ ssh -k root@xx.xx.xx.xx
root@xx.xx.xx.xx's password:
Welcome to Ubuntu Zesty Zapus (development branch) (GNU/Linux 4.10.0-gd46959c ppc64le)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage
Last login: Fri Jun 30 00:18:40 2017 from 9.41.165.94
root@05425:~# uname -a
Linux 05425 4.10.0-gd46959c #1 SMP Thu Mar 9 20:47:59 CST 2017 ppc64le ppc64le ppc64le GNU/Linux
root@05425:~# uptime
 00:18:55 up 0 min,  1 user,  load average: 0.78, 0.20, 0.07
root@05425:~#

```

- [x] Journald 
[journald_v1.99.7-70-gce19185_SOL_hang.txt](https://github.com/openbmc/openbmc/files/1114194/journald_v1.99.7-70-gce19185_SOL_hang.txt)

jk-ozlabs, 2017-07-03:
Looks like our serial port is somehow wedged:

```
root@witherspoon:~# /tmp/strace obmc-console-server ttyVUART0
[...]
open("/dev/ttyS0", O_RDWR|O_NONBLOCK)   = 6
fcntl64(6, F_GETFL)                     = 0x802 (flags O_RDWR|O_NONBLOCK)
ioctl(6, TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(6, SNDCTL_TMR_CONTINUE or TCSETSF, {B115200 -opost -isig -icanon -echo ...}
```

.. and it remains blocked on that `ioctl()`..

Investigating further.

jk-ozlabs, 2017-07-03:
So, it looks like the local tty mirror (`ttyS0`) is blocking indefinitely, causing the server to (eventually) wait for that tty to drain once kernel & console server buffers are full.

We see the open and init:

```
open("/dev/ttyS0", O_RDWR|O_NONBLOCK)   = 6
fcntl64(6, F_GETFL)                     = 0x802 (flags O_RDWR|O_NONBLOCK)
ioctl(6, TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(6, SNDCTL_TMR_CONTINUE or TCSETSF, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(6, TCGETS, {B115200 -opost -isig -icanon -echo ...}) = 0
ioctl(6, SNDCTL_TMR_CONTINUE or TCSETSF, {B115200 -opost -isig -icanon -echo ...}) = 0
```

And we can write console data to fd 6 for a little while (likely filling up kernel buffers), until we hit a limit:

```
write(6, ".19107|  <none>\r\n 81.19274|  Mod"..., 599) = -1 EAGAIN (Resource temporarily unavailable)
```

(that write doesn't block, as we're using `O_NONBLOCK`). This then starts filling the console server's ring buffer. Once that fills, the server forces writes in blocking mode (so that we don't lose characters):

```
fcntl64(6, F_SETFL, O_RDWR)             = 0
write(6, ".19107|  <none>\r\n 81.19274|  Mod"..., 3533
```
(indefinite block).

So I assume we're hitting issues with hardware flow control here... Fix coming soon.

jk-ozlabs, 2017-07-03:
Curiously, hardware flow control is disabled. Could be a HW issue...

jk-ozlabs, 2017-07-03:
So it sounds like this is simply an artefact of the UARTs being non-functional until they're brought out of reset. @shenki  should be providing a fix for that soon.

williamspatrick, 2017-07-03:
I thought we put a fix in for that about a month ago. How did it regress again?

jk-ozlabs, 2017-07-04:
AFAIK, that fix was only for Romulus?

williamspatrick, 2017-07-04:
No, we had a defect at IBM explicitly for Witherspoon.

closed 2017-07-05 by williamspatrick

gkeishin, 2017-07-05:
Looking good with https://gerrit.openbmc-project.xyz/#/c/5090/  build which is now merged.. able to see the Host SOL for few iteration of Poweron.. but eventually, hangs at boot. The OS didn't come up this time, so that could another issue not related to this.

Looking forward to the master build.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-75"
VERSION_ID="v1.99.7-75-gbe34c6e"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-75"
BUILD_ID="v1.99.7"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-990f06c35b7e119f00967e90f0514dbbd2e1b51b #1 Tue Jul 4 06:24:11 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```

````
[    6.979246] i2c /dev entries driver
[    6.979782] at24 0-0050: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    6.980273] at24 1-0054: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    6.980790] at24 2-0050: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    6.981286] at24 3-0054: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    6.981785] at24 4-0050: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    7.134182] at24 7-0050: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    7.134678] at24 8-0054: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    7.135192] at24 9-0050: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    7.135689] at24 10-0054: 16384 byte 24c128 EEPROM, writable, 1 bytes/write
[    7.170312] ata3: SATA link down (SStatus 0 SControl 300)
[    7.177858] ata4: SATA link down (SStatus 0 SControl 300)
[    7.309717] powernv-cpufreq: cpufreq pstate min 60 nominal 57 max 0
[    7.309820] powernv-cpufreq: Workload Optimized Frequency is enabled in the platform
[    7.312157] usbcore: registered new interface driver usbhid
[    7.312225] usbhid: USB HID core driver
```