openbmc/openbmc #1883: hwmon driver for PMBUS Power Supply

bradbishop, 2017-06-30:
Add powerN_input sensors for PMBUS power supply.

geissonator, 2017-07-13:
Needs task list when ready to work on

zahrens, 2017-07-14:
@eddiejames Do you have the task list?

eddiejames, 2017-07-14:
- [ ] Investigate pmbus interface
- [ ] Create power supply (p17531) driver shell using pmbus and set up to export input power
- [ ] Test it.

bjwyman, 2017-08-07:
P17531 refers to a power supply specification document revision. I believe driver was renamed to ibmps.

geissonator, 2017-08-15:
Merged into openbmc kernel

closed 2017-08-15 by geissonator

bjwyman, 2017-08-15:
https://github.com/openbmc/openbmc/commit/83a4307ecef9a5451af4f46ca3f84c1b45d9cd1b

mdmillerii, 2017-08-15:
>  I believe driver was renamed to ibmps.

And later renamed again to "IBM Common Form Factor power supply" (compatible `ibm,cffps1`).