openbmc/openbmc #1885: [Barreleye] phosphor-gpio-monitor@checkstop.service: Failed with result 'core-dump

swe12345, 2017-06-30:
Driver Details
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-67"
VERSION_ID="v1.99.7-67-gdd05b96"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-67"
BUILD_ID="v1.99.7"
root@barreleye:~#
```
```
root@barreleye:~# obmcutil poweron
root@barreleye:~# journalctl --no-pager | egrep 'SEGV|core-dump'
Jun 29 20:33:49 barreleye systemd[1]: phosphor-gpio-monitor@checkstop.service: Failed with result 'core-dump'.
```
[journal_ctl_30jun.txt](https://github.com/openbmc/openbmc/files/1115772/journal_ctl_30jun.txt)

mine260309, 2017-11-27:
Should be duplicated with #2349

gtmills, 2018-06-29:
#2349 was closed. Closing this issue.

closed 2018-06-29 by gtmills