openbmc/openbmc #1886: Is there any documents about which methods is supported by openbmc

sdemirhan, 2017-06-30:
Hi,
I want to discover methods which support by openbmc. For that, I use restful api and i list all interfaces. Like "/org/openbmc/control/fans".  I use "curl -b cjar -k https://10.42.0.160/org/openbmc/control/" command to list control interface. 
"/org/openbmc/control/fans"  has not an attribute. So i wrote "curl -b cjar -k https://10.42.0.160/org/openbmc/control/fans/schema" command to see methods.
```
curl -b cjar -k https://10.42.0.160/org/openbmc/control/fans/schema
{
  "data": {
    "org.freedesktop.DBus.Introspectable": {
      "method": {
        "Introspect": [
          {
            "direction": "out", 
            "name": "data", 
            "type": "s"
          }
        ]
      }, 
      "signal": {}
    }, 
    "org.freedesktop.DBus.Peer": {
      "method": {
        "GetMachineId": [
          {
            "direction": "out", 
            "name": "machine_uuid", 
            "type": "s"
          }
        ], 
        "Ping": []
      }, 
      "signal": {}
    }, 
    "org.freedesktop.DBus.Properties": {
      "method": {
        "Get": [
          {
            "direction": "in", 
            "name": "interface", 
            "type": "s"
          }, 
          {
            "direction": "in", 
            "name": "property", 
            "type": "s"
          }, 
          {
            "direction": "out", 
            "name": "value", 
            "type": "v"
          }
        ], 
        "GetAll": [
          {
            "direction": "in", 
            "name": "interface", 
            "type": "s"
          }, 
          {
            "direction": "out", 
            "name": "properties", 
            "type": "a{sv}"
          }
        ], 
        "Set": [
          {
            "direction": "in", 
            "name": "interface", 
            "type": "s"
          }, 
          {
            "direction": "in", 
            "name": "property", 
            "type": "s"
          }, 
          {
            "direction": "in", 
            "name": "value", 
            "type": "v"
          }
        ]
      }, 
      "signal": {
        "PropertiesChanged": [
          {
            "name": "interface", 
            "type": "s"
          }, 
          {
            "name": "changed_properties", 
            "type": "a{sv}"
          }, 
          {
            "name": "invalidated_properties", 
            "type": "as"
          }
        ]
      }
    }, 
    "org.openbmc.control.Fans": {
      "method": {
        "setMax": [
          {
            "direction": "out", 
            "type": "i"
          }
        ], 
        "updatePresent": [
          {
            "direction": "out", 
            "type": "i"
          }
        ]
      }, 
      "signal": {}
    }
  }, 
  "message": "200 OK", 
  "status": "ok"
}

```
I want to use "setMax" methods.  But when i wrote command, i get the below error.

```
curl c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://10.42.0.160/org/openbmc/control/power0/action/setPowerState -d '{"state": 1}'  
{
  "data": {
    "description": "Expecting request format { 'data': <type 'list'> }, got '{u'state': 1}'"
  }, 
  "message": "400 Bad Request", 
  "status": "error"
}
```
Do you have any suggestion to see methods and how can i use these methods? 

Do you have any idea why i get this error?

Thank you for your time.

Seda

williamspatrick, 2017-06-30:
* Your question talked about setMax but then your example interacted with setPowerState on an entirely different object.
* The response error seems obvious to me.  `"Expecting request format { 'data': <type 'list'> }, got '{u'state': 1}'"`  The JSON is expected to be a map of 'data' and I think you're trying to pass the data itself.
* It appears that setMax takes no data.
* Unless you are running v1.0 code, the latest code will not use /org/openbmc objects but /xyz/openbmc_project.  If you're using any /org/openbmc objects on recent code you're probably making a mistake.
* The schema for /xyz/openbmc_project objects are all documented in the phosphor-dbus-interfaces repository.  The old /org/openbmc objects had no good documentation.
* This document is suppose to give a tutorial on REST->DBus.  This also gives an example for passing data over REST.  https://github.com/openbmc/docs/blob/master/rest-api.md#http-post-operations

closed 2017-06-30 by williamspatrick

sdemirhan, 2017-07-03:
Hi Patrick,

I know this issue is closed but i realize that i post wrong Restful command for the setMax. The Restful command for setMax is given below.
```
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST https://10.42.0.160/org/openbmc/control/fans/action/setMax -d "{\"data\": 1 }"
{
  "data": {
    "description": "Expecting request format { 'data': <type 'list'> }, got '{u'state': 1}'"
  }, 
  "message": "400 Bad Request", 
  "status": "error"
}
```

I use the last version of openbmc. I use the org/openbmc repository because some of the repostory(in org/openbmc) is not found in xyz/openbmc_project. For example, /org/openbmc/sensors repository does not match /xyz/openbmc_project/sensors. Becuse there is not /xy/openbmc_project/sensonr repository in xyz/openbmc_project.
```
curl -b cjar -k https://10.42.0.160/xyz/openbmc_project/
{
  "data": [
    "/xyz/openbmc_project/dump", 
    "/xyz/openbmc_project/control", 
    "/xyz/openbmc_project/object_mapper", 
    "/xyz/openbmc_project/inventory", 
    "/xyz/openbmc_project/logging", 
    "/xyz/openbmc_project/software", 
    "/xyz/openbmc_project/led", 
    "/xyz/openbmc_project/state"
  ], 
  "message": "200 OK", 
  "status": "ok"
}
```
I want to use sensors repository in xyz/openbmc_project.
Which steps should i follow? Do you have any recommend about this? Where do i start to add sensors repository in xyz/openbmc_project?

Thank you for your time.
Seda