openbmc/openbmc #1887: Out of memory: python killed

anoo1, 2017-06-30:
On witherspoon, with today's master build, attempted power on, and failed with the following msgs:

```
Jun 27 14:05:22 witherspoon systemd[1]: Starting Start Power0...
Jun 27 14:05:26 witherspoon mapper[1474]: Error starting event loop: No buffer space available
Jun 27 14:05:26 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-led-groups-power_on.service: Main process exited, code=exited, status=1/FAILURE
Jun 27 14:05:26 witherspoon systemd[1]: Failed to start Wait for /xyz/openbmc_project/led/groups/power_on.
...
Jun 27 14:05:42 witherspoon systemd[1]: op-wait-power-on@0.service: Start operation timed out. Terminating.
Jun 27 14:05:42 witherspoon systemd[1]: op-wait-power-on@0.service: Main process exited, code=killed, status=15/TERM
Jun 27 14:05:42 witherspoon systemd[1]: Failed to start Wait for Power0 to turn on.
```

Tried to run the mapper calls manually but would get the no buffer space error
```
root@witherspoon:~# mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition
Error starting event loop: No buffer space available
```

Then the bmc became very slow to respond, but a ps command eventually returned in addition to journal msgs
```
[ 4018.450000] rngd invoked oom-killer: gfp_mask=0x14201ca(GFP_HIGHUSER_MOVABLE|__GFP_COLD), nodemask=0, order=0, oom_score_adj=0
[ 4018.450000] CPU: 0 PID: 941 Comm: rngd Not tainted 4.10.17-9e95993e2d6edc68af5ca3f2345789e0f450d0db #2
[ 4018.450000] Hardware name: ASpeed SoC
[ 4018.450000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[ 4018.450000] [<8000d934>] (show_stack) from [<801dbecc>] (dump_stack+0x20/0x28)
[ 4018.450000] [<801dbecc>] (dump_stack) from [<800deebc>] (dump_header.constprop.5+0x80/0x1b0)
[ 4018.450000] [<800deebc>] (dump_header.constprop.5) from [<800aa900>] (oom_kill_process+0x2a0/0x46c)
[ 4018.450000] [<800aa900>] (oom_kill_process) from [<800aaed0>] (out_of_memory+0xe8/0x418)
[ 4018.450000] [<800aaed0>] (out_of_memory) from [<800aef48>] (__alloc_pages_nodemask+0x960/0x9b8)
[ 4018.450000] [<800aef48>] (__alloc_pages_nodemask) from [<800a75e0>] (filemap_fault+0x3bc/0x58c)
[ 4018.450000] [<800a75e0>] (filemap_fault) from [<800c9930>] (__do_fault+0x30/0x94)
[ 4018.450000] [<800c9930>] (__do_fault) from [<800cca5c>] (handle_mm_fault+0x72c/0xe04)
[ 4018.450000] [<800cca5c>] (handle_mm_fault) from [<80011d30>] (do_page_fault+0x19c/0x2c8)
[ 4018.450000] [<80011d30>] (do_page_fault) from [<8000930c>] (do_PrefetchAbort+0x48/0xac)
[ 4018.450000] [<8000930c>] (do_PrefetchAbort) from [<8000e920>] (ret_from_exception+0x0/0x20)
[ 4018.450000] Exception stack(0x9cb03fb0 to 0x9cb03ff8)
[ 4018.450000] 3fa0:                                     00000001 00000001 ffffffff 00000004
[ 4018.450000] 3fc0: 00022d24 00000040 00022d1c 000000a8 7ec38c64 7ec38be4 7ec38ce8 00000000
[ 4018.450000] 3fe0: 00000000 7ec3830c 00011c30 4b8793ac 60000010 ffffffff
[ 4018.450000] Mem-Info:
[ 4018.450000] active_anon:96692 inactive_anon:3243 isolated_anon:0
[ 4018.450000]  active_file:27 inactive_file:27 isolated_file:0
[ 4018.450000]  unevictable:1595 dirty:0 writeback:0 unstable:0
[ 4018.450000]  slab_reclaimable:772 slab_unreclaimable:1916
[ 4018.450000]  mapped:370 shmem:8318 pagetables:807 bounce:0
[ 4018.450000]  free:644 free_pcp:204 free_cma:0
[ 4018.450000] Node 0 active_anon:386768kB inactive_anon:12972kB active_file:108kB inactive_file:108kB unevictable:6380kB isolated(anon):0kB isolated(file):0kB mapped:1480kB dirty:0kB writeback:0kB shmem:33272kB writeback_tmp:0kB unstable:0kB pages_scanned:0 all_unreclaimable? no
[ 4018.450000] Normal free:2576kB min:2624kB low:3280kB high:3936kB active_anon:386768kB inactive_anon:12972kB active_file:108kB inactive_file:108kB unevictable:6380kB writepending:0kB present:442368kB managed:432324kB mlocked:0kB slab_reclaimable:3088kB slab_unreclaimable:7664kB kernel_stack:1104kB pagetables:3228kB bounce:0kB free_pcp:816kB local_pcp:816kB free_cma:0kB
[ 4018.450000] lowmem_reserve[]: 0 0
[ 4018.450000] Normal: 304*4kB (UE) 170*8kB (UE) 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 2576kB
[ 4018.450000] 9967 total pagecache pages
[ 4018.450000] 110592 pages RAM
[ 4018.450000] 0 pages HighMem/MovableOnly
[ 4018.450000] 2511 pages reserved
[ 4018.450000] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
[ 4018.450000] [  538]     0   538     3378      450       8       0        0             0 systemd-journal
[ 4018.450000] [  876]     0   876     1054      210       5       0        0         -1000 systemd-udevd
[ 4018.450000] [  886]   997   886     3734       65       8       0        0             0 systemd-timesyn
[ 4018.450000] [  927]     0   927     1474       50       8       0        0             0 phosphor-invent
[ 4018.450000] [  928]     0   928     1465       46       8       0        0             0 phosphor-image-
[ 4018.450000] [  929]     0   929     1517       57       8       0        0             0 phosphor-hwmon-
[ 4018.450000] [  930]   998   930    34460    33234      72       0        0          -900 dbus-daemon
[ 4018.450000] [  931]   996   931     1427       65       6       0        0             0 systemd-network
[ 4018.450000] [  932]     0   932     1484       51       8       0        0             0 phosphor-log-ma
[ 4018.450000] [  933]     0   933     1434       46       8       0        0             0 phosphor-ledcon
[ 4018.450000] [  934]     0   934     3915     1010      14       0        0             0 python
[ 4018.450000] [  935]     0   935    28414    25526      60       0        0             0 python
[ 4018.450000] [  936]     0   936     1434       46       9       0        0             0 phosphor-ledcon
[ 4018.450000] [  937]     0   937      751       12       7       0        0             0 syslogd
[... 33 lines cut ...]
[ 4018.450000] [ 1023]     0  1023     1325       40       8       0        0             0 phosphor-fru-fa
[ 4018.450000] [ 1040]     0  1040     1436       48       8       0        0             0 phosphor-host-s
[ 4018.450000] [ 1041]     0  1041     8158       84      13       0        0             0 power_control.e
[ 4018.450000] [ 1042]     0  1042     1434       46       8       0        0             0 phosphor-bmc-st
[ 4018.450000] [ 1043]     0  1043     8155       81      12       0        0             0 button_power.ex
[ 4018.450000] [ 1044]     0  1044     8154       80      12       0        0             0 button_reset.ex
[ 4018.450000] [ 1045]     0  1045     1434       46       8       0        0             0 phosphor-chassi
[ 4018.450000] [ 1080]     0  1080     1309       41       8       0        0             0 timemanager
[ 4018.450000] [ 1081]     0  1081     1738      124       8       0        0             0 ipmid
[ 4018.450000] [ 1091]     0  1091      645       42       7       0        0             0 dropbear
[ 4018.450000] [ 1092]     0  1092      763       51       7       0        0             0 sh
[ 4018.450000] [ 1099]     0  1099      645       42       7       0        0             0 dropbear
[ 4018.450000] [ 1100]     0  1100      395       10       5       0        0             0 obmc-console-cl
[ 4018.450000] [ 1101]     0  1101    14292     1141      19       0        0             0 python
[ 4018.450000] [ 1233]     0  1233     1457       45       8       0        0             0 phosphor-watchd
[ 4018.450000] [ 1235]     0  1235     1331       36       8       0        0             0 phosphor-gpio-m
[ 4018.450000] [ 1514]     0  1514     1517       59       8       0        0             0 phosphor-hwmon-
[ 4018.450000] [ 1658]     0  1658     1517       58       8       0        0             0 phosphor-hwmon-
[ 4018.450000] [ 1661]     0  1661     1517       58       9       0        0             0 phosphor-hwmon-
[ 4018.450000] [ 1668]     0  1668     1517       58       8       0        0             0 phosphor-hwmon-
[ 4018.450000] [ 1669]     0  1669     1517       58       8       0        0             0 phosphor-hwmon-
[ 4018.450000] Out of memory: Kill process 935 (python) score 229 or sacrifice child
[ 4018.450000] Killed process 935 (python) total-vm:113656kB, anon-rss:102104kB, file-rss:0kB, shmem-rss:0kB
[ 4022.980000] oom_reaper: reaped process 935 (python), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
```

geissonator, 2017-07-14:
@anoo1 did this one work itself out or still an issue?

anoo1, 2017-09-19:
Have not seen this since then, closing.

closed 2017-09-19 by anoo1