openbmc/openbmc #1888: host watchdog not being enabled all the time on boot

geissonator, 2017-06-30:
I've seen a few situations where the watchdog is not getting armed and the boot is just hanging.  It seems like the watchdog, in certain boots (maybe reboots) is not getting armed until after the host sends down it's first watchdog commands.  Here's a journal of it:
```
Apr 20 18:25:27 witherspoon systemd[1]: Reached target Chassis0 (Power On).
Apr 20 18:25:27 witherspoon phosphor-chassis-state-manager[1043]: Received signal that power ON is complete
Apr 20 18:25:27 witherspoon phosphor-chassis-state-manager[1043]: Change to Chassis Power State
Apr 20 18:25:27 witherspoon openpower-proc-control[3042]: Running P9 procedure startHost
Apr 20 18:25:28 witherspoon openpower-proc-control[3042]: Setting SBE seeprom side to 1
Apr 20 18:25:28 witherspoon systemd[1]: Started Start host0 SBE.
Apr 20 18:25:28 witherspoon systemd[1]: Reached target Start Host0.
Apr 20 18:25:28 witherspoon systemd[1]: Reached target Host0 (Starting).
Apr 20 18:25:28 witherspoon phosphor-host-state-manager[1046]: Received signal that host is running
Apr 20 18:25:28 witherspoon phosphor-host-state-manager[1046]: Change to Host State
Apr 20 18:25:44 witherspoon systemd[1]: Starting Phosphor Fan Monitor Daemon...
Apr 20 18:25:45 witherspoon systemd[1]: Started Phosphor Fan Monitor Daemon.
Apr 20 18:25:45 witherspoon systemd[1]: Started Phosphor Fan Control Daemon.
Apr 20 18:25:45 witherspoon systemd[1]: Reached target Fan Control Ready.
Apr 20 18:25:50 witherspoon phosphor-watchdog[2760]: watchdog: enabled and started
Apr 20 18:26:00 witherspoon inventory_items.py[943]: Caching: /org/openbmc/inventory/system/chassis/motherboard
Apr 20 18:26:00 witherspoon inventory_items.py[943]: Caching: /org/openbmc/inventory/system/chassis/motherboard/refclock
Apr 20 18:26:00 witherspoon inventory_items.py[943]: Caching: /org/openbmc/inventory/system/chassis/motherboard/pcieclock
Apr 20 18:26:00 witherspoon inventory_items.py[943]: Caching: /org/openbmc/inventory/system/chassis/motherboard/todclock
Apr 20 18:26:00 witherspoon inventory_items.py[943]: Caching: /org/openbmc/inventory/system/chassis/motherboard/apss
Apr 20 18:26:02 witherspoon ipmid[1054]: WATCHDOG RESET
```

You can see the watchdog enable is not happening until 22 seconds after we've started the SBE.

I saw some other cases where we never enabled and we just hung (when there was a bad host image in PNOR).

geissonator, 2017-06-30:
```
[witherspoon]# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-71"
VERSION_ID="v1.99.7-71-g89a1a35"
```

vishwabmc, 2017-07-03:
yeah. this is a known issue that I mentioned in one of my commit messages that it was an inadvertent change that put this.

rfrandse, 2017-07-04:
<https://gerrit.openbmc-project.xyz/5079> Watchdog: Enable Host Watchdog during early poweron
Fixes  #1888 host watchdog not being enabled all the time on boot

closed 2017-07-05 by williamspatrick