openbmc/openbmc #1889: REST Post method unable to handle BYTE data

prkatti1, 2017-07-03:
Need support for POST method to handle BYTE data

sivassrr, 2017-07-19:
Any update here.

sivassrr, 2017-07-19:
Currently we are testing it using 3000 port - This may not be available for GA, I guess. 
So we need to get this ASAP.

prkatti1, 2017-07-27:
@sivassrr Since IP address configuration, prefix length and network gateway scenarios coupled with this, All the scenarios involving configuring IP address, prefix length and configuring gateway IP address are blocked. This count will be around 35 tests.

sivassrr, 2017-07-27:
Thank you @prkatti1 for updating it. Working with development team and getting right focus here.

ratagupt, 2017-08-02:
*# STATUS

rfrandse, 2017-08-07:
<https://gerrit.openbmc-project.xyz/5813> Convert value to expected type during POST method
Resolves  #1889 REST Post method unable to handle BYTE data

closed 2017-08-11 by williamspatrick