openbmc/openbmc #1890: Error reported by istep (0x1700) PLID 0x9000003B

PavamanSubramaniyam, 2017-07-03:
Every time I reboot the HOST OS, the P9 Zaius hardware shows the following messages while doing ISTEP operations.

```
--== Welcome to Hostboot  ==--

  3.09607|secure|SecureROM invalid - skipping functionality
  4.74141|ISTEP  6. 5 - host_init_fsi
  4.82881|ISTEP  6. 6 - host_set_ipl_parms
  5.27032|ISTEP  6. 7 - host_discover_targets
  9.08259|HWAS|PRESENT> DIMM[03]=CFA5FFA500000000
  9.08261|HWAS|PRESENT> Proc[05]=C000000000000000
  9.11257|ISTEP  6. 8 - host_update_master_tpm
  9.11406|SECURE|Security Access Bit> 0x0000000000000000
  9.11407|SECURE|Secure Mode Disable (via Jumper)> 0xC000000000000000
  9.11423|ISTEP  6. 9 - host_gard
  9.16719|ISTEP  6.10 - host_revert_sbe_mcs_setup
  9.17705|ISTEP  6.11 - host_start_occ_xstop_handler
  9.17804|ISTEP  6.12 - host_voltage_config
  9.20791|ISTEP  7. 2 - mss_volt
  9.37774|ISTEP  7. 3 - mss_freq
  9.52934|ISTEP  7. 4 - mss_eff_config
 12.11401|ISTEP  7. 5 - mss_attr_update
 12.12268|ISTEP  8. 1 - host_slave_sbe_config
 12.15209|ISTEP  8. 2 - host_setup_sbe
 12.15660|ISTEP  8. 3 - host_cbs_start
 12.16484|ISTEP  8. 4 - proc_check_slave_sbe_seeprom_complete
 22.38213|ISTEP  8. 5 - host_attnlisten_proc
 22.38475|ISTEP  8. 6 - host_p9_fbc_eff_config
 22.38769|ISTEP  8. 7 - host_p9_eff_config_links
 22.39605|ISTEP  8. 8 - proc_attr_update
 22.39798|ISTEP  8. 9 - proc_chiplet_fabric_scominit
 22.44941|ISTEP  8.10 - proc_xbus_scominit
 28.20438|ISTEP  8.11 - proc_xbus_enable_ridi
 28.21026|ISTEP  9. 1 - fabric_erepair
 28.21849|ISTEP  9. 2 - fabric_io_dccal

 32.68823|------------------------------------------------
 32.68824|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 32.68824|  FFDC                       : TIMEOUT
 32.68825|    00
 32.68826|------------------------------------------------
 32.68826|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 32.68827|  FFDC                       : M_WDERF_START
 32.68828|    00000100 5AFAF700
 32.68829|------------------------------------------------
 32.68830|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 32.68830|  FFDC                       : M_WDERF_DONE
 32.68831|    00000000 000742F8
 32.68832|------------------------------------------------
 32.68833|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 32.68833|  FFDC                       : M_WDERF_FAILED
 32.68834|    00000000 00168B08
 32.68835|------------------------------------------------
 32.68836|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 32.68837|  FFDC                       : M_LANE_BAD_0_15
 32.68838|    00000000 00000000
 32.68838|------------------------------------------------
 32.68839|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 32.68840|  FFDC                       : M_LANE_BAD_16_23
 32.68841|    00000000 0002336C
 32.68842|------------------------------------------------
 32.68842|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 32.68843|  FFDC                       : M_LANE_DISABLED_VEC_0_15
 32.68844|    00000000 0004FD98
 32.68845|------------------------------------------------
 32.68846|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
 32.68847|  FFDC                       : M_LANE_DISABLED_VEC_16_23
 32.68848|    00000100 5AFAF700
 32.68849|------------------------------------------------
 32.68849|  HwpReturnCode              : RC_IO_XBUS_LINKTRAIN_ERROR
[... 165 lines cut ...]
 32.69008|  Target                   : Physical:/Sys0/Node0/Proc1/XBUS2
 32.69009|  Deconfig State           : DELAYED_DECONFIG
 32.69009|  GARD Error Type          : GARD_Unrecoverable
 32.69010|  Priority                 : SRCI_PRIORITY_NONE
 32.69011|------------------------------------------------
 32.69012|  Hostboot Build ID:
 32.69013|================================================
 33.17670|System shutting down with error status 0x9000003B
 33.17674|================================================
 33.17675|Error reported by istep (0x1700) PLID 0x9000003B
 33.17676|  IStep failed, see other log(s) with the same PLID for reason.
 33.17677|  ModuleId   0x01 MOD_REPORTING_ERROR
 33.17678|  ReasonCode 0x1703 RC_FAILURE
 33.17679|  UserData1  eid of first error : 0x9000003b0000090f
 33.17679|  UserData2  Reason code of first error : 0x0000000400000904
 33.17680|------------------------------------------------
 33.17681|  fabric_io_run_training
 33.17682|------------------------------------------------
 33.17682|  Callout type             : Procedure Callout
 33.17683|  Procedure                : EPUB_PRC_HB_CODE
 33.17684|  Priority                 : SRCI_PRIORITY_LOW
 33.17684|------------------------------------------------
 33.17685|  Hostboot Build ID:
 33.17686|================================================
```

PavamanSubramaniyam, 2017-07-03:
The machine is installed with the latest Open BMC and PNOR Firmware images.

```
root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-74"
VERSION_ID="v1.99.7-74-gec3c51e"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-74"
BUILD_ID="v1.99.7"

root@zaius:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x02820000..0x02821000 !
[                                                  ] 0%open-power-zaius-v1.17-111-gcc5a4f6-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-158-ga1e0a047b2a0
        hostboot-f576402
        linux-4.11.6-openpower1-p487dbeb
        petitboot-v1.4.3-p2462045
        machine-xml-5006472
        occ-7adb49a
        hostboot-binaries-0e2ad15
        capp-ucode-9c73e9f
        sbe-02021c6
[==================================================] 100%

```

PavamanSubramaniyam, 2017-07-03:
[dmesg.txt](https://github.com/openbmc/openbmc/files/1119460/dmesg.txt)
[journalctl.txt](https://github.com/openbmc/openbmc/files/1119464/journalctl.txt)

williamspatrick, 2017-07-03:
How is this a BMC issue?

closed 2017-07-03 by williamspatrick