openbmc/openbmc #1892: Performance Optimization

rednewguy, 2017-07-04:
We see CPU load average of 3.0 and the REST is several orders of magnitude slower than IPMI on AST2400. The faster CPU clock in AST2500 makes it a tad better but generally speaking it is still slow to respond. 

Are there any performance optimization in the backlog? Do you have a list of items that can help some refactoring or shredding to improve performance. 
DBus and REST seem to be major contributors. If you have anything in the pipeline or ideas to improve, we can explore them.

williamspatrick, 2017-07-04:
Which machine is this?  Generally, I haven't observed anything more than 2% CPU usage at idle. What kind of activity do you have going on?

If you are performing a lot of REST operations, be sure you are reusing the connection as much as possible. Creating a new SSL session is quite compute intensive.  Right now, all of the SSL processing is done in the python web server. It is possible that using something like nginx may reduce the cost to establish the connection, so that would be an area of performance improvement someone could experiment with.

rednewguy, 2017-07-04:
This machine is a flavor of OCP Mt Olympus. Sensor monitor + DBus averages around 40% CPU. We are also not using cookies for security implications. REST SSL sessions are authenticated and established everytime. nginx is a good suggestion, we will look into it. From what we have seen, a REST call spends more than half of its time in the DBus than in the web server and network. Are there any best practices to make DBus lighter and faster?

williamspatrick, 2017-07-05:
We would have to do some deeper performance analysis based on the dbus calls you are making. Dbus itself isn't slow; it is sub 1ms from what I've seen.

I suspect one of a few things:

The rest server itself, with the python dbus library is slow at packing and unpacking a dbus message. 

The rest server is doing an introspection operation somewhere as part of its call, which is computationally intensive. 

The dbus end point is being slow to respond because it is single threaded and doing blocking io to the kernel. I've seen this problem with the hwmon/ sensors application and we need to investigate how to make it do non-blocking / async io.  Right now the kernel blocks it while reading the end hwmon-based device, which could be over a slow i2c bus.

williamspatrick, 2017-07-05:
You said "not using cookies" and "connections are established every time". Those two are not necessarily connected.  If you are using something like the python requests library in your external application you can keep the connection alive for long periods of time. This is especially a good idea if you have an external app doing polling on the system to maintain status.

williamspatrick, 2017-07-05:
It would be nice if we got whatever this "mt. Olympus" machine is upstreamed. Is there any plan to do that?

rednewguy, 2017-07-05:
We do have plans to upstream but there is no definite timeline yet.

rednewguy, 2017-07-05:
We have replaced OpenBMC REST APIs with Redfish. However, we simply reused and extended the existing dbus objects. Your comment "The rest server is doing an introspection operation somewhere as part of its call, which is computationally intensive" makes me think I should rather refactor dbus objects to better align with Redfish URI and schema. Do you have any OpenBMC architecture/design document that you could share?

amboar, 2017-07-06:
@causten was looking at Redfish support for OpenBMC. Might be useful to get him in on the discussion.

williamspatrick, 2017-07-06:
@rednewguy - I'm not following what you're asking for as an OpenBMC architecture/design document.  Everything we have is on Github already.  Are you looking for the dbus schemas?

https://github.com/openbmc/phosphor-dbus-interfaces

The DBus objects / schemas are primarily a firmware-to-firmware communication mechanism.  When those can be aligned with a systems management function like Redfish it isn't harmful for us to model the schema in that way.  We have attempted to do so in some cases.  In general, I do not want to see all of the DBus objects attempting to be constrained by the requirements of Redfish because I do not see that as a good long-term firmware design.  That is, effectively, how "the competition" has their proprietary BMC product designed except tied to IPMI.  We don't want to have to rewrite everything for each new management interface that comes along.

amboar, 2019-07-25:
The nature of the original post is open-ended and so was probably better served by a post to the mailing list. Closing this given there's no other obvious way of concluding it, and it seems no-one is interested in following up.

closed 2019-07-25 by amboar