openbmc/openbmc #1894: Requested transaction contradicts existing jobs: Transaction is destructive.

shenki, 2017-07-04:
v1.99.7-74-gec3c51e (ec3c51e) on witherspoon, when doing `obmcutil poweron`:

```
Jul 04 07:07:31 witherspoon phosphor-host-state-manager[1017]: Host State transaction request
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-vcs_workaround.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Starting Witherspoon Fan Watchdog Reset...
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-phosphor\x2dgpio\x2dmonitor.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-ucd_disable_vcs.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-vrm\x2dcontrol.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-ir35221\x2don\x2dunbind.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Starting power on mode unbind ir35221 device driver devs...
Jul 04 07:07:31 witherspoon systemd[1]: Starting Disable UCD VCS rails...
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-phosphor\x2dwatchdog.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Started Phosphor poweron watchdog.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-start_host.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-avsbus\x2ddisable.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Starting Wait for /org/openbmc/control/chassis0...
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-avsbus\x2denable.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-vcs_on.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-vcs_off.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Starting Wait for /xyz/openbmc_project/led/groups/power_on...
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-phosphor\x2dfan\x2dcontrol\x2dinit.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-op\x2dpower\x2dstart.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-phosphor\x2dfan\x2dpresence\x2dtach.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Started Phosphor GPIO checkstop monitor.
Jul 04 07:07:31 witherspoon systemd[1]: Reached target Start Host checkstop monitor.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-avsbus\x2dworkaround.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-ir35221\x2don\x2dbind.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-fsi\x2dscan.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-mapper\x2dsubtree\x2dremove.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Starting mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition...
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-cfam_override.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-pcie\x2dslot\x2ddetect.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-fsi\x2dbind.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Created slice system-op\x2dwait\x2dpower\x2don.slice.
Jul 04 07:07:31 witherspoon systemd[1]: Starting Wait for Power0 to turn on...
Jul 04 07:07:33 witherspoon systemd[1]: Started Witherspoon Fan Watchdog Reset.
Jul 04 07:07:33 witherspoon systemd[1]: Started power on mode unbind ir35221 device driver devs.
Jul 04 07:07:33 witherspoon systemd[1]: Started Wait for /org/openbmc/control/chassis0.
Jul 04 07:07:33 witherspoon kernel: ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 04 07:07:33 witherspoon kernel[908]: [  184.060000] ucd9000 11-0064: Device ID UCD90160|2.3.4.0000|110603
Jul 04 07:07:33 witherspoon systemd[1]: Started Wait for /xyz/openbmc_project/led/groups/power_on.
Jul 04 07:07:33 witherspoon systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationBlocksTransition.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Service hold-off time over, scheduling restart.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Service hold-off time over, scheduling restart.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Service hold-off time over, scheduling restart.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Service hold-off time over, scheduling restart.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Service hold-off time over, scheduling restart.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4070.service: Stop job pending for unit, delaying automatic restart.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 04 07:07:33 witherspoon systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 04 07:07:33 witherspoon systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 04 07:07:33 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 04 07:07:34 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 04 07:07:34 witherspoon systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 04 07:07:34 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 04 07:07:34 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit not needed anymore. Stopping.
Jul 04 07:07:34 witherspoon systemd[1]: Requested transaction contradicts existing jobs: Transaction is destructive.
Jul 04 07:07:34 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Failed to enqueue stop job, ignoring: Transaction is destructive.
Jul 04 07:07:34 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40400-ucd90160\x4064.service: Unit not needed anymore. Stopping.
Jul 04 07:07:34 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jul 04 07:07:34 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 04 07:07:34 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jul 04 07:07:34 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40140-ir35221\x4070.service: Unit not needed anymore. Stopping.
Jul 04 07:07:34 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jul 04 07:07:34 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: Started Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@ahb-apb-i2c\x401e78a000-i2c\x2dbus\x40180-ir35221\x4071.service: Unit not needed anymore. Stopping.
Jul 04 07:07:34 witherspoon systemd[1]: Stopping Phosphor Hwmon Poller...
Jul 04 07:07:34 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: Starting Assert power_on LED...
Jul 04 07:07:34 witherspoon systemd[1]: Started Disable UCD VCS rails.
Jul 04 07:07:34 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
Jul 04 07:07:34 witherspoon systemd[1]: Stopped Phosphor Hwmon Poller.
```

williamspatrick, 2017-07-04:
Looks kind of like the bind / unbind calls are all happening so close together that systemd can't react fast enough.

mine260309, 2017-07-05:
Yup, I see this kind of log on Witherspoon, it happens on IR35221 unbind (`ir35221-on-unbind@.service`), but eventually the IR35221 driver will bind (`ir35221-on-bind@.service`) and hwmon will work fine, so I assume this is OK.

closed 2019-04-27 by stale[bot]