openbmc/openbmc #1895: mapper: Error finding '/xyz/openbmc_project/led/groups/power_on'

shenki, 2017-07-04:
Host fails to boot with `v1.99.7-74-gec3c51e` (ec3c51e) on witherspoon:

```
Jul 04 07:20:21 witherspoon systemd[1]: Started Disable VCS before Witherspoon power on.
Jul 04 07:20:21 witherspoon systemd[1]: Reached target Power0 On (Pre).
Jul 04 07:20:21 witherspoon systemd[1]: Starting Start Power0...
Jul 04 07:20:42 witherspoon hwmon.py[983]: Removing: /sys/class/hwmon/hwmon5/
Jul 04 07:20:42 witherspoon hwmon.py[983]: Removing: /sys/class/hwmon/hwmon8/
Jul 04 07:20:42 witherspoon hwmon.py[983]: Removing: /sys/class/hwmon/hwmon6/
Jul 04 07:20:42 witherspoon hwmon.py[983]: Removing: /sys/class/hwmon/hwmon7/
Jul 04 07:20:44 witherspoon mapper[1325]: Error finding '/xyz/openbmc_project/led/groups/power_on' service: Connection timed out
Jul 04 07:20:44 witherspoon systemd[1]: op-wait-power-on@0.service: Main process exited, code=exited, status=1/FAILURE
Jul 04 07:20:44 witherspoon pgood_wait[1324]: Error obtaining host service: Connection timed out
Jul 04 07:20:44 witherspoon systemd[1]: Failed to start Wait for Power0 to turn on.
Jul 04 07:20:44 witherspoon systemd[1]: Dependency failed for Chassis0 (Power On).
Jul 04 07:20:44 witherspoon systemd[1]: Dependency failed for Start Host0.
Jul 04 07:20:44 witherspoon mapper[1328]: Error starting event loop: Connection timed out
Jul 04 07:20:44 witherspoon systemd[1]: obmc-host-start@0.target: Job obmc-host-start@0.target/start failed with result 'dependency'.
Jul 04 07:20:44 witherspoon systemd[1]: obmc-host-start@0.target: Triggering OnFailure= dependencies.
Jul 04 07:20:44 witherspoon systemd[1]: obmc-chassis-poweron@0.target: Job obmc-chassis-poweron@0.target/start failed with result 'dependency'.
```

anoo1, 2017-07-04:
Do you see a mapper error regarding "No buffer space available"? ..  I've intermittently seen python oom in the latest driver during power on ( #1887 )

```
Jun 27 14:03:02 witherspoon mapper[1400]: Error starting event loop: No buffer space available
Jun 27 14:03:02 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-led-groups-power_on.service: Main process exited, code=exited, status=1/FAILURE
Jun 27 14:03:02 witherspoon systemd[1]: Failed to start Wait for /xyz/openbmc_project/led/groups/power_on.
Jun 27 14:03:02 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-led-groups-power_on.service: Unit entered failed state.
Jun 27 14:03:02 witherspoon systemd[1]: mapper-wait@-xyz-openbmc_project-led-groups-power_on.service: Failed with result 'exit-code'.
Jun 27 14:03:02 witherspoon systemd[1]: Starting Assert power_on LED...
Jun 27 14:03:05 witherspoon mapper[1390]: Error finding '/xyz/openbmc_project/led/groups/power_on' service: Connection timed out
```

gtmills, 2018-05-23:
@shenki @anoo1 Still a problem? If not, can we close?

closed 2018-05-24 by shenki