openbmc/openbmc #1899: Require the ability to set VLAN tagging configuration in-band using ipmitool

whowutwut, 2017-07-05:
Looking for the ability to set VLAN tag on the OpenBMC `eth0` using inband ipmitool. 

The command that we are looking to be able to execute is: `ipmitool -d 0 lan set 1 vlan id <ID>`

@bradbishop Please add this to #1880 
Priority: High

whowutwut, 2017-07-06:
@bradbishop and @ratagupt  Just making sure that it's clear inband is the primary focus, then out of band later on, if that separation is able to be done in the code.

williamspatrick, 2017-07-06:
The "out-of-band" is free, because we always define the DBus objects first and the REST API comes through introspection.  We have to do that part first anyhow.  The IPMI provider is then done second and with that we get both in-band and out-of-band IPMI support.

ratagupt, 2017-07-11:
Can we mark this issue as enhancement?

ratagupt, 2017-07-11:
- [ ] Implement the Dbus Object Implementing VLAN.interface.

- [ ] Define the create interface for VLAN.

- [ ] Implement the create interface for VLAN.

- [ ] Write the VLAN configuration in the systemd.networkd.

- [ ] Create the VLAN interfaces as a child object of Network Manager.

- [ ] Write the test cases to verify that VLAN configuration have been written.

ratagupt, 2017-07-13:
Update:- Worked with victor to get the VLAN setup.

whowutwut, 2017-07-13:
@ratagupt  I tried to set up the USB to serial on another node in the frame...  give that a shot and let me know if you can access the BMC.

ratagupt, 2017-07-14:
@whowutwut : I needed for openbmc machine.

rfrandse, 2017-08-04:
<https://gerrit.openbmc-project.xyz/5635> Add support for vlan(set lan command)
Resolves  #1899 [xCAT] Require the ability to set VLAN tagging configuration in-band using ipmitool

closed 2017-08-14 by williamspatrick