openbmc/openbmc #1900: [xCAT] How do we ssh to the BMC using the ipv6 address?

whowutwut, 2017-07-05:
This is useful in development because we sometimes hose the BMC network configuration by accident.  For the AMI versions of the OpenPower BMC, I'm able to connect a cat5 ethernet cable directly from my laptop.  For example: `ssh -6 <user>@fe80::72e2:84ff:fe14:0252`

Then we are able to issue ipmitools in the BMC and re-configure the network settings. 

Even though ipv6 seems to be enabled on OpenBMC, it seems like it doesn't allow me to ssh into the box.    Can someone help with providing instructions on how to do this?  Without this working,  we have to open up the P9 and connect a 8 pin serial cable to the boxelder board, it's a much more difficult process.

ratagupt, 2017-07-06:
You can not use link local address for ssh,assign some routable ipv6 address then try.

whowutwut, 2017-07-06:
@ratagupt Thanks, I'm not that well versed in ipv6, can you help with this by giving an example? 
If I have two P9 nodes connected to the same routable ipv4 subnet, how can I ssh to one of them using the ipv6 interface?  

The use case I'm trying to address is to be able to recover the BMC when the ipv4 network is hosed and i can't ssh or ping to it.  How do we handle this easily instead of connecting in the 8 pin serial to the boxelder board?   Is there a solution here?

williamspatrick, 2017-07-06:
We are not supporting ipv6 for the initial delivery.  @whowutwut is there a reason you are attempting to test this?

whowutwut, 2017-07-06:
We manage our own DEV cluster and we currently do not own one of these 8 pin serial cable.  Maybe the solution is to get a few sent to us from Austin.   The BMCs are also not on the 9. network... they are behind a login node.  When we hose up the BMC by accident since we need to do testing against configuration of the BMC...   there's no pinhole reset button to turn them back to DHCP mode. 

Then we have to... 
* find someone with the 8 pin serial cable 
* pull the server from the frame, open it up
* connect the 8 pin, dangle it from an open PCI slot out the back
* restore power and hook it up to a serial -> usb cable 
* restore the settings back
* power off , open the box and remove the 8 pin cable, 

So trying to find a way to restore the BMC properties without the need to pull out the server and open the box...

ratagupt, 2017-07-06:
@whowutwut : you can do the ssh of the machine with using the link local address if they are on the **same link** but you need to be sure..
did you try ping -6 / ping6 to the ip?.

whowutwut, 2017-07-21:
Sorry for the delay, but good news... with the 1730A firmware, I'm able to ssh directly plugged into the BMC... using IPv6....

```
++ ssh -6 root@fe80::72E2:84ff:fe14:290F
The authenticity of host 'fe80::72e2:84ff:fe14:290f (fe80::72e2:84ff:fe14:290f)' can't be established.
RSA key fingerprint is SHA256:04avwRzDhYrhFjOTSQofuoevt2H05BGSxPSvj7TnAqA.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'fe80::72e2:84ff:fe14:290f' (RSA) to the list of known hosts.
root@fe80::72e2:84ff:fe14:290f's password:
root@witherspoon:~# ifconfig
eth0      Link encap:Ethernet  HWaddr 70:E2:84:14:29:0F
          inet addr:172.20.140.111  Bcast:172.20.255.255  Mask:255.255.0.0
          inet6 addr: fe80::72e2:84ff:fe14:290f%1217803152/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:140877 errors:0 dropped:0 overruns:0 frame:0
          TX packets:127353 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:9133342 (8.7 MiB)  TX bytes:9075905 (8.6 MiB)
          Interrupt:20

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1%1217803152/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:679684 errors:0 dropped:0 overruns:0 frame:0
          TX packets:679684 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:51656208 (49.2 MiB)  TX bytes:51656208 (49.2 MiB)
```

closed 2017-07-21 by whowutwut