openbmc/openbmc #1901: openpower-proc-control: Failed to set clock mux on CPU2

mine260309, 2017-07-06:
In P9's `startHost()`, it is expected that it shall set the clock mux on both CPUs if the systems has 2 CPUs attached.

However, the log shows only one CPU is set clock mux from journal log:
```
"NUM_PROCS" : "1",
"MESSAGE" : "Running P9 procedure startHost",
```
This causes some XBUS error during host boot.

mine260309, 2017-07-06:
The root cause is in `targeting.cpp`, it does not correctly search the fsi path to get CPU2:
```
        // fsiSlaveBasePath  is "/sys/devices/platform/gpio-fsi/fsi0/slave@00:00/00:00:00:0a/"
        std::regex exp{"fsi1/slave@([0-9]{2}):00", std::regex::extended};
        for (auto& file : fs::directory_iterator(fsiSlaveBasePath))
        {
            std::smatch match;
            std::string path = file.path();
            if (std::regex_search(path, match, exp))
            {
                ...
                targets.push_back(std::make_unique<Target>(pos, path, swapper));
            }
```
The `directory_iterator` only iterates files in `.../00:00:00:0a/`, where it reaches `.../00:00:00:0a/fsi1`, but does not iterates into `.../00:00:00:0a/fsi1/`.
So it will never get a regex match and it only creates the default target.

One possible fix is to let it iterate into `.../00:00:00:0a/fsi1`, e.g.
```
diff --git a/targeting.hpp b/targeting.hpp
index d5b99fd..0a946f6 100644
--- a/targeting.hpp
+++ b/targeting.hpp
@@ -15,7 +15,7 @@ constexpr auto fsiMasterDevPathOld =
     "/sys/devices/platform/fsi-master/slave@00:00/raw";

 constexpr auto fsiSlaveBaseDir =
-    "/sys/devices/platform/gpio-fsi/fsi0/slave@00:00/00:00:00:0a/";
+    "/sys/devices/platform/gpio-fsi/fsi0/slave@00:00/00:00:00:0a/fsi1/";
 constexpr auto fsiSlaveBaseDirOld =
     "/sys/devices/platform/fsi-master/slave@00:00/hub@00/";
```
@eddiejames Could you please have a check and comment on this issue?

geissonator, 2017-07-06:
@spinler - looks like you also had a change in this area, thoughts?

spinler, 2017-07-06:
I had thought @eddiejames had made this code 4.10 friendly?  I wonder if something else has changed since then.

closed 2017-07-06 by williamspatrick