openbmc/openbmc #1902: Host checkstop condition detected durin poweron[ Witherspoon]

gkeishin, 2017-07-06:
To recreate:  Host auto reboot is disabled and does boot to OS back and forth. 

Note:  with autoboot enabled the system gets recovered but the purpose here is bring the system in fail state and debug

```
$ ./rest_cmd -i xx.xx.xx.xx -o GET -u /org/openbmc/settings/host0/attr/auto_reboot
Executing: GET  /org/openbmc/settings/host0/attr/auto_reboot

/org/openbmc/settings/host0/attr/auto_reboot
"no"

```

`Driver`
```
root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x02820000..0x02821000 !
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.17_1.46
        op-build-v1.17-101-g1c57f18-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-158-ga1e0a047b2a0
        hostboot-0ffbb55
        linux-4.11.6-openpower1-p57dd16e
        petitboot-v1.4.3-pfac92ff
        machine-xml-5b59a1d
        occ-1dc97a6
        hostboot-binaries-711147e
        capp-ucode-9c73e9f
        sbe-0df81aa
[==================================================] 100%
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-990f06c35b7e119f00967e90f0514dbbd2e1b51b #1 Wed Jul 5 16:22:21 UTC 2017 armv6l GNU/Linux
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-78"
VERSION_ID="v1.99.7-78-gb5e5644"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-78"
BUILD_ID="v1.99.7"
root@witherspoon:~#
```

`Journald snippet of the checkstop`

```
Apr 20 20:04:32 witherspoon phosphor-gpio-monitor[2197]: GPIO line altered
Apr 20 20:04:32 witherspoon systemd[1]: Created slice system-openpower\x2ddebug\x2dcollector\x2dcheckstop.slice.
Apr 20 20:04:32 witherspoon systemd[1]: Starting OpenPOWER debug data collector for host checkstop...
Apr 20 20:04:32 witherspoon checkstop_app[2493]: Host checkstop condition detected
Apr 20 20:04:32 witherspoon systemd[1]: Started OpenPOWER debug data collector for host checkstop.
Apr 20 20:04:32 witherspoon systemd[1]: Reached target Quiesce Target.
Apr 20 20:04:32 witherspoon systemd[1]: Reached target Host instance 0 crashed.
Apr 20 20:04:33 witherspoon phosphor-host-state-manager[1008]: Auto reboot disabled.
Apr 20 20:04:33 witherspoon phosphor-host-state-manager[1008]: Maintaining quiesce
Apr 20 20:04:33 witherspoon phosphor-host-state-manager[1008]: Change to Host State
```

`Error log`
```
$ ./rest_cmd -i xx.xx.xx.xx -o GET -u /xyz/openbmc_project/logging/enumerate
Executing: GET  /xyz/openbmc_project/logging/enumerate

/xyz/openbmc_project/logging/enumerate
{
    "/xyz/openbmc_project/logging/entry/1": {
        "AdditionalData": [],
        "Id": 1,
        "Message": "org.open_power.Host.Error.Checkstop",
        "Resolved": 0,
        "Severity": "xyz.openbmc_project.Logging.Entry.Level.Informational",
        "Timestamp": 1492718672835,
        "associations": []
    },
    "/xyz/openbmc_project/logging/internal/manager": {}
}
```

`FFDC`
[checkstop_ffdc.zip](https://github.com/openbmc/openbmc/files/1127151/checkstop_ffdc.zip)

gkeishin, 2017-07-06:
Looks like there is a new PNOR coming to fix this.. Will need to check.

geissonator, 2017-07-18:
Host checkstop issues should go to host.

closed 2017-07-18 by geissonator

sivassrr, 2017-07-18:
Can we move this to repository https://github.com/open-power/skiboot/issues

geissonator, 2017-07-18:
You can move issues around repos within openbmc but I don't think you can move issues across repos.

sivassrr, 2017-07-18:
Thank you @geissonator , @gkeishin we can open issue in CQ then.