openbmc/openbmc #1904: FSI OCC kernel driver causes WARN when probing FSI

shenki, 2017-07-06:
Using latest master on Witherspoon. Can be reproduced with `obmcutil poweron`:

```
[ 1645.220000] ------------[ cut here ]------------
[ 1645.220000] WARNING: CPU: 0 PID: 988 at kernel/softirq.c:161 __local_bh_enable_ip+0xa0/0xe4
[ 1645.220000] CPU: 0 PID: 988 Comm: openpower-occ-c Not tainted 4.10.17-00308-g5653d5db8db9 #467
[ 1645.220000] Hardware name: ASpeed SoC
[ 1645.220000] [<80010060>] (unwind_backtrace) from [<8000d924>] (show_stack+0x20/0x24)
[ 1645.220000] [<8000d924>] (show_stack) from [<801daefc>] (dump_stack+0x20/0x28)
[ 1645.220000] [<801daefc>] (dump_stack) from [<80018d04>] (__warn+0xe0/0x108)
[ 1645.220000] [<80018d04>] (__warn) from [<80018df8>] (warn_slowpath_null+0x30/0x38)
[ 1645.220000] [<80018df8>] (warn_slowpath_null) from [<8001c350>] (__local_bh_enable_ip+0xa0/0xe4)
[ 1645.220000] [<8001c350>] (__local_bh_enable_ip) from [<803afe90>] (unix_release_sock+0x68/0x250)
[ 1645.220000] [<803afe90>] (unix_release_sock) from [<803b00a4>] (unix_release+0x2c/0x3c)
[ 1645.220000] [<803b00a4>] (unix_release) from [<80315468>] (sock_release+0x2c/0xa0)
[ 1645.220000] [<80315468>] (sock_release) from [<803154f8>] (sock_close+0x1c/0x24)
[ 1645.220000] [<803154f8>] (sock_close) from [<800e3e68>] (__fput+0x94/0x1c4)
[ 1645.220000] [<800e3e68>] (__fput) from [<800e4000>] (____fput+0x18/0x1c)
[ 1645.220000] [<800e4000>] (____fput) from [<80031d3c>] (task_work_run+0x78/0xa0)
[ 1645.220000] [<80031d3c>] (task_work_run) from [<8001b2b4>] (do_exit+0x274/0x8e0)
[ 1645.220000] [<8001b2b4>] (do_exit) from [<8001b9b8>] (do_group_exit+0x4c/0xb4)
[ 1645.220000] [<8001b9b8>] (do_group_exit) from [<800253ac>] (get_signal+0x290/0x5b4)
[ 1645.220000] [<800253ac>] (get_signal) from [<8000ccf0>] (do_signal+0xd4/0x3dc)
[ 1645.220000] [<8000ccf0>] (do_signal) from [<8000d1d4>] (do_work_pending+0xac/0xbc)
[ 1645.220000] [<8000d1d4>] (do_work_pending) from [<8000a58c>] (slow_work_pending+0xc/0x20)
[ 1645.220000] ---[ end trace f6297087cb50289f ]---
```

eddiejames, 2017-07-06:
This is @vishwabmc application I believe

gkeishin, 2017-07-08:
I too see this during poweron and reboot
```
[43618.830000] ------------[ cut here ]------------
[43618.830000] WARNING: CPU: 0 PID: 968 at /tmp/openbmc/build/tmp/work-shared/witherspoon/kernel-source/kernel/softirq.c:161 __local_bh_enable_ip+0xa0/0xe4
[43618.830000] CPU: 0 PID: 968 Comm: openpower-occ-c Not tainted 4.10.17-990f06c35b7e119f00967e90f0514dbbd2e1b51b #1
[43618.830000] Hardware name: ASpeed SoC
[43618.830000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[43618.830000] [<8000d934>] (show_stack) from [<801dbecc>] (dump_stack+0x20/0x28)
[43618.830000] [<801dbecc>] (dump_stack) from [<80018d54>] (__warn+0xe0/0x108)
[43618.830000] [<80018d54>] (__warn) from [<80018e48>] (warn_slowpath_null+0x30/0x38)
[43618.830000] [<80018e48>] (warn_slowpath_null) from [<8001c398>] (__local_bh_enable_ip+0xa0/0xe4)
[43618.830000] [<8001c398>] (__local_bh_enable_ip) from [<803abd04>] (unix_release_sock+0x68/0x248)
[43618.830000] [<803abd04>] (unix_release_sock) from [<803abf10>] (unix_release+0x2c/0x3c)
[43618.830000] [<803abf10>] (unix_release) from [<80310eb0>] (sock_release+0x2c/0xa0)
[43618.830000] [<80310eb0>] (sock_release) from [<80310f40>] (sock_close+0x1c/0x24)
[43618.830000] [<80310f40>] (sock_close) from [<800e447c>] (__fput+0x94/0x1c4)
[43618.830000] [<800e447c>] (__fput) from [<800e4614>] (____fput+0x18/0x1c)
[43618.830000] [<800e4614>] (____fput) from [<80031e08>] (task_work_run+0x7c/0xa0)
[43618.830000] [<80031e08>] (task_work_run) from [<8001b2f8>] (do_exit+0x280/0x8e8)
[43618.830000] [<8001b2f8>] (do_exit) from [<8001b9fc>] (do_group_exit+0x4c/0xb4)
[43618.830000] [<8001b9fc>] (do_group_exit) from [<800253cc>] (get_signal+0x214/0x580)
[43618.830000] [<800253cc>] (get_signal) from [<8000cc90>] (do_signal+0x74/0x3dc)
[43618.830000] [<8000cc90>] (do_signal) from [<8000d1d4>] (do_work_pending+0xac/0xbc)
[43618.830000] [<8000d1d4>] (do_work_pending) from [<8000a58c>] (slow_work_pending+0xc/0x20)
[43618.830000] ---[ end trace 88b84ea62fe0a67b ]---
```

closed 2017-07-10 by williamspatrick