openbmc/openbmc #1907: Sprint 17.07.12 - Committed

rfrandse, 2017-07-06:


dkodihal, 2017-07-07:
Squad 1 sprint plan (WIP)

Deepak - carry overs from previous.
Ratan - VLAN.
Vishwa - OCC epic or power/thermal.
Tom - DCMI, #1920 
Devender - IPMI/error log cleanup
Jayanth - Dump
Dhruv - settings validation or host inventory persistence

geissonator, 2017-07-12:
**Andrew:geissonator -**

**17.06.28**

- #948 Customer Power Cap
  * 7/11 - Pushed up for code review

 **17.07.12:**

- **#1951 (occ sensors on zaius)**

- **#1933 (hard power off not working)**

- RAS error inject defects, Jordan voltage bias change requests defects

-------------------------------------------------------------------------------------------------

**Gunnar:gtmills: 80%**
 **17.06.28:**

  - **#1329 (power supply presence)**
     - 7/11 - pushed recipes, addressed some comments

 **17.07.12:**

  - #1860 (PDM elogs)
      - 7/12 - 

vacation planned

-------------------------------------------------------------------------------------------------
**Chris:cbostic**: 75%

**17.06.14:**
- **#1636 (watchdog driver DT)**
     - 7/10 - Guenter still looking for tweaks

**17.06.28:**

- **#1783 (witherspoon gpio-keys)**
      - 7/11 - will reach out to Joel

 **17.07.12:**

- #1727 (ucd driver support)
      - 7/11 - dug in yesterday, more straightforward than expected

------------------------------------------------------------------------------------------------
**Brandon:bjwyman**: 50%

**17.06.14:**
 - #1738 (fan fault led device)
     - 7/11 - still no feedback.

**17.06.28:**
 - #1731 (power supply monitor for faults)
     - 7/10 - WIP 
     - 7/11 - WIP

 **17.07.12:**
- #1835 (fan control final piece)

-------------------------------------------------------------------------------------------------
**Dinesh:chinaridinesh**: 65%

**17.06.14:**
- **#1315 (Ensure uboot fw_env update script uses new inventory object)**
    - 6/26 - need a review
    - 7/6 - working on uuid impl.

 **17.07.12:**

-------------------------------------------------------------------------------------------------
**Eddie:eddiejames**: 80%

**17.06.28:**

- #651 (power supply driver)
    - 7/10 - promoted to epic

- #1883 (hwmon skeleton for power supply driver)
   - 7/11 - working it.  got power register for hwmon.

 **17.07.12:**

- Code Update, move 651/1883 into this sprint?

upstreaming:
    - 7/10 - haven't checked email for feedback on occ/sbefifo

-------------------------------------------------------------------------------------------------

**Matt S:spinler: 80%** 
**17.06.28:**

- #1724 (chip fault analysis)
       - 6/29 - WIP, waiting on driver.
       - 7/6 - interrupt isn't going to work.  will probably need to poll.
       - 7/10 - all done, except driver work

 **17.07.12:**

- Code update, what's after that?

design:
      
vacation - out thurs/friday 13/14th.

-------------------------------------------------------------------------------------------------
**Matt B:msbarth: 80%**
**17.06.28:**

- #1835 (delay on events parameter)
     - 7/6 - need occ active dbus object
     - 7/7 - Critical for some one to work on while Matt B is out.  
     - 7/11 - working on email

- **#1839 (handle scale property)**
     - 7/6 - getting started
     - 7/7 - design issue snag
     - 7/11 - almost done

- #1792 (yaml updates)

vacation

design topics:

-------------------------------------------------------------------------------------------------
**Brad:bradbishop**: 70%

**17.05.31:**
- **#1318 (gpio presence)**
     - 7/6 - 

**17.06.14:**
 - **#1188 (both fan rpms)**
     - 7/6 - ran into problems
     - 7/11 - mostly done

**17.06.28:**
 - #1767 (mapper org.open_power)
      - 7/10 - up for review

 **17.07.12:**

anoo1, 2017-07-13:
# Squad 3 -

**Adriana:anoo1 -**
**17.06.14**
- #1551 mboxd: Use tmp location for pnor patches

**17.07.12**
- #1940 BMC: Ability for u-boot to execute the kernel that's in a ubi volume
- #1943 BMC: Tell u-boot where squashfs image is
---

**Brad:bradbishop -**
**17.06.28**
- #1318 gpio presence

**17.07.12**
- #1941 BMC: Build: Create flash-able image with a static layout
- #1842 BMC: Create Manifest file and add it to the <system>.tar image
---

**Charlie: -**
**17.06.28**
- openbmc/openbmc-test-automation#746 Code update: Remaining upload image of BMC/ PNOR test cases
 
**17.07.12**
- #1938 BMC: Reduce shutdown watchdog value back to default
---

**Eddie:eddiejames -**
**17.07.12**
#1937 BMC: Split u-boot env into 2 copies
#1942 BMC: Change primary and alt BMC layout to just have u-boot, boot env, and a ubi mtd
---

**Leo:lgon -**
**17.06.28**
- #1550 Implement Software Object.Delete for pnor updater and image manager
---

**Matt S:spinler - **
- #1939 BMC: Determine if there's still an issue with erase of boot env
---

**Michael:mtritz -**
**17.06.28**
- #1576 Implement BMC factory reset
  * 7/11 - Need to write service files next.

**17.07.12**
- #1756 BMC: Implement the Activation flow (write/update files)
---

**Saqib:saqibkh -**
**17.06.28**
- #1650 Move contents from /var/lib/obmc to /var/persist
  * 7/07 - Continuing, fstab my be overriding it.

**17.07.12**
- #1944 BMC: Block BMC reboot during u-boot update
---

closed 2017-08-03 by rfrandse