openbmc/openbmc #1909: BMC Kernel doesn't reset on kill PID 1

gkeishin, 2017-07-06:
```
    Login to BMC
    kill -ABRT 1
    Wait for BMC to come back online
    Check if the BMC rebooted
```
```
root@witherspoon:~# kill -ABRT 1
root@witherspoon:~# [ 1109.950000] systemd: 36 output lines suppressed due to ratelimiting
[ 1109.950000] systemd-journald[644]: Failed to send WATCHDOG=1 notification message: Connection refused

root@witherspoon:~# 

Apr 20 17:47:44 witherspoon kernel: systemd: 36 output lines suppressed due to ratelimiting
Apr 20 17:47:44 witherspoon systemd[1]: Caught <ABRT>, dumped core as pid 1082.
Apr 20 17:47:44 witherspoon kernel[903]: [ 1109.950000] systemd: 36 output lines suppressed due to ratelimiting
Apr 20 17:47:44 witherspoon systemd[1]: Freezing execution.
Apr 20 17:47:44 witherspoon kernel[903]: [ 1109.950000] systemd-journald[644]: Failed to send WATCHDOG=1 notification message: Connection refused
root@witherspoon:~# uptime
 18:12:14 up 43 min,  load average: 0.52, 0.39, 0.36
root@witherspoon:~#
```

After discussion,  looks like there's a config file the systemd watchdog needs to be enabled in, and it isn't.

@spinler @geissonator

spinler, 2017-07-06:
From http://0pointer.de/blog/projects/watchdog.html:
```To make use of the hardware watchdog it is sufficient to set the RuntimeWatchdogSec= option in /etc/systemd/system.conf. It defaults to 0 (i.e. no hardware watchdog use). Set it to a value like 20s and the watchdog is enabled. After 20s of no keep-alive pings the hardware will reset itself. Note that systemd will send a ping to the hardware at half the specified interval, i.e. every 10s. And that's already all there is to it. By enabling this single, simple option you have turned on supervision by the hardware of systemd and the kernel beneath it.```

@gkeishin Maybe try uncommenting the RuntimeWatchdogSec line in that file and setting it to maybe 20s (and reboot to pick up the changes) and then see if that helps?

spinler, 2017-07-06:
...Though, the fan watchdog will be at /dev/watchdog0.  Hopefully that is different than the /dev/watchdog that systemd will be using.

gkeishin, 2017-07-06:
@spinler Sure I ll try this one out..

gkeishin, 2017-07-06:
```
root@witherspoon:~# cat /etc/systemd/system.conf | grep RuntimeWatchdogSec
RuntimeWatchdogSec=20s
root@witherspoon:~# uptime
 17:31:38 up 2 min,  load average: 1.87, 1.34, 0.54
root@witherspoon:~# kill -ABRT 1
root@witherspoon:~# [  154.750000] systemd: 42 output lines suppressed due to ratelimiting
[  154.750000] watchdog: watchdog0: watchdog did not stop!

root@witherspoon:~# journalctl -f
-- Logs begin at Thu 2017-04-20 17:29:26 UTC. --
Apr 20 17:31:13 witherspoon login[944]: pam_lastlog(login:session): file /var/log/lastlog created
Apr 20 17:31:13 witherspoon login[944]: pam_unix(login:session): session opened for user root by LOGIN(uid=0)
Apr 20 17:31:13 witherspoon login[944]: pam_systemd(login:session): Failed to create session: The name org.freedesktop.login1 was not provided by any .service files
Apr 20 17:31:13 witherspoon login[1052]: ROOT LOGIN  on '/dev/ttyS4'
Apr 20 17:31:49 witherspoon systemd[1]: Caught <ABRT>, dumped core as pid 1059.
Apr 20 17:31:49 witherspoon systemd[1]: Freezing execution.
Apr 20 17:31:49 witherspoon kernel: systemd: 42 output lines suppressed due to ratelimiting
Apr 20 17:31:49 witherspoon kernel: watchdog: watchdog0: watchdog did not stop!
Apr 20 17:31:49 witherspoon kernel[922]: [  154.750000] systemd: 42 output lines suppressed due to ratelimiting
Apr 20 17:31:49 witherspoon kernel[922]: [  154.750000] watchdog: watchdog0: watchdog did not stop!

U-Boot 2016.07 (Jun 26 2017 - 22:20:10 +0000)

       Watchdog enabled
DRAM:  496 MiB
Flash: 32 MiB
In:    serial
Out:   serial
Err:   serial
Net:
```
It worked..

spinler, 2017-07-06:
@bradbishop @williamspatrick Any concern with enabling this?  What timeout should we use?

spinler, 2017-07-06:
So I checked the state of WDT1 with this enabled:

```# devmem 0x1e78500c
0x00000013
```
that means the WDT1 is running and will reset the system on a timeout.

So it is definitely trying to use the same hardware that we need to use for the fan watchdog.

I guess that means if we want to use this, we'd need to do something like use a udev rule or DTS update to link /dev/watchdog to WDT2, as systemd is hardcoded to use /dev/watchdog.

(Full disclosure, the above test was on a Palmetto, but no reason to suspect a witherspoon would be different.  Will try on a witherspoon when I can get one).

gkeishin, 2017-07-07:
```
KILL       9   exit      cannot be blocked
```
With also KILL Pid 1 doesnt seems to be killed
```
root@witherspoon:~# ps -aef  | grep systemd | grep init
    1 root       0:08 {systemd} /sbin/init
root@witherspoon:~#
```

mdmillerii, 2017-07-07:
@gkeishin pid 1 is special.   Quoting a comment in `kernel/signal.c`:
```
                 * Global init gets no signals it doesn't want.
                 * Container-init gets no signals it doesn't want from same
                 * container.
```

mdmillerii, 2017-07-07:
@spinler I just verified that the kernel is hard coded that /dev/watchdog is an aliased device to /dev/watchdog0.  However, when setting the id N in /dev/watchdogN the /aliases are searched so we can use that to change which hardware instance is which linux watchdog device.

mdmillerii, 2017-07-07:
Also, I believe the watchdog is set to ~45 minutes as a workaround for the allowing the flash to be updated.  Now that we have watchdogd in busybox we could change that, but it will not be needed when we get to the new bmc code update.

spinler, 2017-07-07:
@mdmillerii Thanks.  So I would do something like
```
aliases {
    watchdog0 = &wdt2;
    watchdog1 = &wdt1;
};
```
?

gkeishin, 2017-08-02:
@spinler anything on this one...

spinler, 2017-08-02:
I believe @shenki is submitting a systemd patch so that it will no longer be hardcoded to use /dev/watchdog0.  And then after that goes through the config file can be updated to turn on this watchdog.

gkeishin, 2017-08-02:
Perfect.. Thanks If you know which one is @shenki patch so that we can track..

rfrandse, 2017-10-10:
dup of #2407

closed 2017-10-10 by rfrandse

mdmillerii, 2017-10-11:
@rfrandse  you should have said #2427 not 2407, and when you look at it, 2427 was opened because this issue was reverse mirrored from this defect, so that one is the dup :-)