openbmc/openbmc #1912: Romulus VRMs fail to come up

amboar, 2017-07-07:
Stewart observed the following on a Romulus system:

```
Jul 07 00:51:52 romulus power_control.exe[901]: PowerControl: setting power up BMC_POWER_UP to 1
Jul 07 00:51:52 romulus system_manager.py[856]: Running System State: HOST_POWERING_ON
Jul 07 00:51:52 romulus timemanager[931]:  PGOOD has changed..
Jul 07 00:51:52 romulus systemd[1]: Started Start Power0.
Jul 07 00:51:52 romulus systemd[1]: Starting Perform AVS bus workaround on VRMs...
Jul 07 00:51:52 romulus systemd[1]: Started mapper subtree-remove /xyz/openbmc_project/software:xyz.openbmc_project.Software.ActivationB
locksTransition.
Jul 07 00:51:53 romulus systemd[1]: Started Trigger CPLD to give pgood.
Jul 07 00:52:07 romulus avsbus-workaround.sh[1927]: Error: Write failed
Jul 07 00:52:21 romulus systemd[1]: [[0;1;39mop-wait-power-on@0.service: Start operation timed out. Terminating.[[0m
Jul 07 00:52:21 romulus systemd[1]: [[0;1;39mop-wait-power-on@0.service: Main process exited, code=killed, status=15/TERM[[0m
Jul 07 00:52:21 romulus systemd[1]: [[0;1;31mFailed to start Wait for Power0 to turn on.[[0m
Jul 07 00:52:21 romulus systemd[1]: [[0;1;39mDependency failed for Chassis0 (Power On).[[0m
Jul 07 00:52:21 romulus systemd[1]: [[0;1;39mDependency failed for Start Host0.[[0m
Jul 07 00:52:21 romulus systemd[1]: [[0;1;39mobmc-host-start@0.target: Job obmc-host-start@0.target/start failed with result 'dependency
'.[[0m
Jul 07 00:52:21 romulus systemd[1]: obmc-host-start@0.target: Triggering OnFailure= dependencies.
Jul 07 00:52:21 romulus systemd[1]: [[0;1;39mobmc-chassis-poweron@0.target: Job obmc-chassis-poweron@0.target/start failed with result '
dependency'.[[0m
Jul 07 00:52:21 romulus systemd[1]: obmc-chassis-poweron@0.target: Triggering OnFailure= dependencies.
Jul 07 00:52:21 romulus systemd[1]: [[0;1;39mop-wait-power-on@0.service: Unit entered failed state.[[0m
Jul 07 00:52:21 romulus systemd[1]: [[0;1;39mop-wait-power-on@0.service: Failed with result 'timeout'.[[0m
Jul 07 00:52:21 romulus systemd[1]: Reached target Stop Host0 (Pre).
Jul 07 00:52:21 romulus systemd[1]: Stopped Trigger CPLD to give pgood.
Jul 07 00:52:21 romulus systemd[1]: Reached target Host0 (Stopping).
Jul 07 00:52:21 romulus systemd[1]: Reached target Host0 (Stopped).
Jul 07 00:52:21 romulus systemd[1]: Reached target Power0 Off (Pre).
Jul 07 00:52:21 romulus systemd[1]: Started Stop Power0.
Jul 07 00:52:21 romulus systemd[1]: Starting Wait for Power0 to turn off...
Jul 07 00:52:21 romulus systemd[1]: [[0;1;39mavsbus-workaround@0.service: Main process exited, code=killed, status=15/TERM[[0m
Jul 07 00:52:22 romulus power_control.exe[901]: PowerControl: setting power up SOFTWARE_PGOOD to 0
Jul 07 00:52:22 romulus power_control.exe[901]: PowerControl: setting power up BMC_POWER_UP to 0
Jul 07 00:52:22 romulus timemanager[931]:  PGOOD has changed..
Jul 07 00:52:22 romulus system_manager.py[856]: Running System State: HOST_POWERING_OFF
Jul 07 00:52:22 romulus systemd[1]: Started Wait for Power0 to turn off.
Jul 07 00:52:22 romulus systemd[1]: Reached target Power0 Off.
Jul 07 00:52:22 romulus systemd[1]: Starting Reset CPLD to wait after VDD is powered up...
Jul 07 00:52:22 romulus systemd[1]: Starting Reload mboxd after power off...
Jul 07 00:52:22 romulus systemd[1]: Stopped Perform AVS bus workaround on VRMs.
Jul 07 00:52:22 romulus systemd[1]: [[0;1;39mavsbus-workaround@0.service: Unit entered failed state.[[0m
Jul 07 00:52:22 romulus systemd[1]: [[0;1;39mavsbus-workaround@0.service: Failed with result 'signal'.[[0m
Jul 07 00:52:22 romulus systemd[1]: Stopped Start Power0.
Jul 07 00:52:23 romulus mboxctl[1956]: Reset: Success
Jul 07 00:52:23 romulus systemd[1]: Started Reload mboxd after power off.
Jul 07 00:52:24 romulus systemd[1]: Started Reset CPLD to wait after VDD is powered up.
Jul 07 00:52:24 romulus systemd[1]: Reached target Chassis0 (Power Off).
Jul 07 00:52:24 romulus phosphor-chassis-state-manager[904]: Received signal that power OFF is complete
Jul 07 00:52:24 romulus phosphor-chassis-state-manager[904]: Change to Chassis Power State
```
I extracted the full log:

[rom1-vrm-fail.txt](https://github.com/openbmc/openbmc/files/1129771/rom1-vrm-fail.txt)

Running `vrm-control.sh` gives similar errors and ultimately times out.

stewart-ibm, 2017-07-07:
With op-build of:
```open-power-romulus-v1.17-130-gd975690
        buildroot-2017.05-15-gcd06d25
        skiboot-v5.7-rc1
        hostboot-f0e3f6d
        linux-4.11.6-openpower1-pb76742d
        petitboot-v1.4.3-p43a7b94
        machine-xml-4a9a504
        occ-6a00709
        hostboot-binaries-0e2ad15
        capp-ucode-9c73e9f
        sbe-02021c6
```

When we get to this part of IPL:
```
15.90443|SBE|>>getHwKeyHashFromSbeImage: i_target=0x50000, i_seeprom=2, i_image_ptr=0xc0000000: reading from memory
 15.90445|SBE|<<getHwKeyHashFromSbeImage: err rc=0x0, plid=0x0, o_hash=0x00000000
 15.91203|SBE|getSbeInfoState() - procCustomizeSbeImg(): maxSize=0x38D20, actSize=0x37E50, crc=0x54D05C7C
 15.91207|PNOR|PnorDD::_readFlash(i_addr=0x007FEC00)> 
 15.91208|PNOR|astMboxDD::adjustMboxWindow using protocol version: 2
 15.91212|PNOR|astMboxDD::adjustMboxWindow opening read window at 0x007fe000 for addr 0x007fec00 req_size 0x00002000
 15.91213|PNOR|>>astMboxDD::doMessage(0x04)
 15.91218|PNOR|Command sent, waiting for response...
 15.91301|PNOR|Reading response data...
 15.91307|PNOR|Message: cmd:04 seq:51 a:00 cb 00 01 00..resp:01
 15.91309|PNOR|<<astMboxDD::doMessage() resp=0x01
 15.91310|PNOR| curWindowOffset    = 00700000
 15.91311|PNOR| curWindowSize      = 00100000
 15.91312|PNOR| curWindowLpcOffset = 0cb00000
 15.91428|SBE|>>getSbeBootSeeprom()
 15.91430|SBE|<<getSbeBootSeeprom(): o_bootSide=0x0 (reg=0x0, tgt=0x50000, master)
 15.91434|SBE|I>SBE Update tgt=0x50000: Seeprom0 flagged as clean: pnor=0, crc=0 (custom=0x54D05C7C/s0=0x54D05C7C), isSimics=0
 15.91438|SBE|I>SBE Update tgt=0x50000: Seeprom1 flagged as clean: pnor=0, crc=0 (custom=0x54D05C7C/s1=0x54D05C7C), isSimics=0
 15.91442|SBE|I>SBE Update tgt=0x50000: Both sides clean-no updates. cur side=0. Continue IPL. (sit=0x00, act=0x00000000)
 16.15402|IPMI|dd: I>read b2h ok 14:30 seq 18 len 0 cc 0
 16.15409|IPMI|dd: I>write ok 10:30 seq 19 len a
 16.16558|ERRL|I>Got an error log Msg - Type: 0x00000039
 16.16559|ERRL|I>Flush message received
 16.16561|INITSVC|<<doIstep: step 10, substep 2
 16.16563|INITSVC|>>doIstep: step 10, substep 3, task host_set_voltages
 16.16565|ISTEP 10. 3 - host_set_voltages
 16.17042|IPMI|rp: queuing async 18:22
 16.17048|ISTEPS_TRACE|call_host_set_voltages enter
 16.17137|ISTEPS_TRACE|Running p9_setup_evid HWP on processor target 0
```

Everything stops and then the host gets powered off.

You are now in the failed state.

stewart-ibm, 2017-07-07:
https://github.com/open-power/op-build/pull/1220 reverts the host firmware commit that triggered this issue.

amboar, 2017-07-07:
Given that it seems unrelated to the BMC, I'll close this issue.

closed 2017-07-07 by amboar

williamspatrick, 2017-07-07:
@geissonator - This seems like something we should send to the lab team for investigation.  If the host can do something to the UCD or the VRMs and then we can never recover without an AC pull, that is going to be a potential stability problem for Witherspoon.

ngruend, 2017-07-07:
Regulator faults can be checked with this on Romulus. 
"i2cget -y 12 0x31 0x05 w"     It should return 0xff7f
Anything else can be decoded with the register definition in the CPLD section of the wiki.  This is assuming there was a fault, which needing to clear AC sounds like.