openbmc/openbmc #1913: netipmid issues

yongli3, 2017-07-07:
I am testing the netipmid using ipmitool. But it does not work on my system. below is the test command and systemd error log
ipmitool -H 192.168.0.11 -P 0penBmc -I lanplus  chassis status
Error sending Chassis Status command: Unspecified error

netipmid[1738]: << openSession
netipmid[1738]: >> RAKP12
netipmid[1738]: << RAKP12
netipmid[1738]: >> RAKP34
netipmid[1738]: << RAKP34
netipmid[1738]: >> setSessionPrivilegeLevel
netipmid[1738]: << setSessionPrivilegeLevel
netipmid[1738]: E> Table:: Command Not found: 0xb03e
netipmid[1738]: Failed to get /org/openbmc/inventory/system/chassis/motherboard/bmc bus name: No such file or directory
netipmid[1738]: Failed to get bus name, return value: No such file or directory.

If I add the "-U root", the command just returns with below message, and there is no any log on the openbmc system.
ipmitool -H 192.168.0.11 -U root -P 0penBmc -I lanplus  chassis status
Error: Unable to establish IPMI v2 / RMCP+ session

tomjoseph83, 2017-07-07:
This is the output of chassis status on a Witherspoon. Please check if the chassis information is available via REST.

-bash-4.1$ ./ipmitool -I lanplus -H x.x.x.x -P 0penBmc -C 1 chassis status
System Power         : on
Power Overload       : false
Power Interlock      : inactive
Main Power Fault     : false
Power Control Fault  : false
Power Restore Policy : previous
Last Power Event     : 
Chassis Intrusion    : inactive
Front-Panel Lockout  : inactive
Drive Fault          : false
Cooling/Fan Fault    : false
Front Panel Control  : none

root@witherspoon:~# cat /etc/os-release 
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-29"
VERSION_ID="v1.99.7-29-ga20607b"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-29"
BUILD_ID="v1.99.7"

williamspatrick, 2017-07-07:
@yongli3 - What version of code are you running?  Code that is attempting to look in /org/openbmc/inventory sounds old.  Can you upgrade to something more recent, like the v1.99.7 tag?  As Tom mentioned we have this working on Witherspoon for certain and I think Barreleye is also supported.

yongli3, 2017-07-09:
Thanks @williamspatrick!  Let me update the code and test it again

yongli3, 2017-07-11:
@williamspatrick  I am upgraded the code and still receive the problem. It seems to me that the latest source code in host-ipmid uses the "/org/openbmc/inventory" name
https://github.com/openbmc/phosphor-host-ipmid/blob/master/apphandler.cpp#L111

Which component provides this dbus service?

yongli3, 2017-07-11:
After checking the code, I found the problem: inventory system has problems. I restarted it and fixed the problem

closed 2017-07-17 by yongli3

henbin, 2017-08-18:
@yongli3 Sorry to bother you. I met the same problem about "Failed to get /org/openbmc/inventory/system/chassis/motherboard/bmc bus name: No such file or directory"

Could you give more description how to resolve it?

Thanks.

sdssly, 2017-08-18:
@henbin could you share more information about your system ? ps to check which services are running ?