openbmc/openbmc #1914: phosphor-fan-control: Error in mapper call to get service name

amboar, 2017-07-07:
Saw the following when trying to power-on the host:

```
Jul 07 02:54:52 witherspoon phosphor-fan-control[1873]: Error in mapper call to get service name
Jul 07 02:54:52 witherspoon phosphor-fan-control[1873]: The operation failed internally.   
Jul 07 02:54:52 witherspoon phosphor-fan-control[1873]: terminate called after throwingan instance of 'sdbusplus::xyz::openbmc_project::Common::Error::InternalFailure'
Jul 07 02:54:52 witherspoon phosphor-fan-control[1873]:   what():  xyz.openbmc_project.Common.Error.InternalFailure: The operation failed internally.
```
BMC was running:

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-82"
VERSION_ID="v1.99.7-82-g7317c97"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-82"
BUILD_ID="v1.99.7"
```

amboar, 2017-07-07:
Note that this was with a slightly modified 4.10 kernel, however I think it's unlikely that the kernel was at fault:

```
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-00298-g990f06c35b7e-dirty #1048 Fri Jul 7 12:01:59 ACST 2017 armv6l GNU/Linux
```
Specifically, that is openbmc/linux@990f06c35b7e119f00967e90f0514dbbd2e1b51b plus a small change to `arch/arm/mach-aspeed/aspeed.c`.

amboar, 2017-07-07:
Ah, I have discovered the issue, and the kernel was at fault. I'd built it with an out of date defconfig (missing various hwmon drivers). Having regenerated it everything works again, but maybe a less vague error should be reported?

msbarth, 2017-07-07:
Not sure what more is being looked for here when I believe changing the verbosity of journalctl displays what (likely a sensor in this case) failed to be returned by the mapper. This is truly an internal failure where the driver was missing for fan control init to run successfully hence terminating.

We'll possibly consider having the path and/or interface displayed in the standard journal output.

williamspatrick, 2017-07-07:
We want the fan control program to terminate when any fan or thermal sensor is unloaded?  That doesn't sound right.

msbarth, 2017-07-07:
When fan control init mode is run, putting the fans to full speed at poweron, we do if it can not access the necessary sensors it requires to complete setting the fans to full speed successfully.

After init completes, the fan control application is restarted in control mode and does not terminate when any configured control mode sensors fail to be accessed.

My guess is the max31785 driver was not included.

williamspatrick, 2017-07-07:
> After init completes, the fan control application is restarted in control mode and does not terminate when any configured control mode sensors fail to be accessed.

The current behavior doesn't help the situation where your fan control application crashed and is now being restarted, but you cannot find one of your desired sensors.  In that case your application will still fail because it is considering it "init mode" still.

msbarth, 2017-07-07:
Right, and after its restarted by systemd and fails init mode a total of 3 times, it terminates the poweron.

init mode and control mode are separate instances of the fan control application (phosphor-fan-control --init vs phosphor-fan-control --control). init mode is only started at poweron and notifies systemd that it completes so that control mode can start. When control mode starts and throws an exception for some reason, systemd only restarts the phosphor-fan-control application back into control mode. If it fails a total of 3 times here, the watchdog should trip and run fans at fullspeed.

williamspatrick, 2017-07-07:
Got it.

@amboar is there still any concern?

amboar, 2017-07-07:
No real concern with the behaviour outlined, it sounds fine. Boot terminating after three failures was what I observed.

> We'll possibly consider having the path and/or interface displayed in the standard journal output

I think this is worthwhile and is why I left the issue open. If the error message had given me paths and interfaces it would have twigged a bit earlier that my defconfig was broken rather than userspace.

bradbishop, 2017-07-14:
The path and interface are already logged as journal metadata:

journalctl -o verbose

closed 2017-07-14 by bradbishop

amboar, 2017-07-14:
@bradbishop oh! Thanks for the tip.