openbmc/openbmc #1915: POWER checkstop cleanup causes an mbox reset, interfering with OCC takeover

amboar, 2017-07-07:
The OCC begins communicating with mboxd as the BMC is trying to clean up from the checkstop. This leads to mboxd being reset after the OCC has issued a `GET_MBOX_INFO`, and the OCC should then restart its mboxd session. Given the OCC runs without configuring mbox interrupts this means one of either the following must take place:

1. The OCC must check the status byte before every command
2. The RESET command to mboxd should be issued after the poweron command but prior to actually powering on the chassis.

williamspatrick, 2017-07-07:
I heard @pragupta was working on the OCC code for P9 to support the mailbox protocol, but I don't think that matters here?  Don't we already wait 30 seconds after a checkstop?  How is the OCC still doing activity?  Or did we miss the 30 second pause in the new checkstop path?

pragupta, 2017-07-07:
OCC uses those 30 seconds to grab fir data and put it in pnor. When @amboar and I were looking, the BMC initiated a mboxd reset on the first command sent from OCC after a checkstop.

amboar, 2017-07-07:
It might be related to the fall that the OCC MBOX implementation was buggy at the time and operations were generally taking longer than expected, but for safety i think we should move the reset it off the shutdown path. I can't see a reason it shouldn't work fine in the boot Perth instead.

williamspatrick, 2017-07-07:
It was previously done in the shutdown path because we had to wait a while for the mbox daemon to reload and it was assumed we'd rather pay that penalty on the shutdown than the power on.  We could play around with moving it to the power on path instead now that it should be relatively fast.

amboar, 2017-07-07:
Indeed

geissonator, 2017-07-25:
Required for PNOR code update and checkstop analysis so moving into current sprint and putting up quick fix.  https://gerrit.openbmc-project.xyz/#/c/5489/

geissonator, 2017-07-25:
**STATUS**
7/25/16 - Quick code patch put up in gerrit, will let HW CI verify it

closed 2017-07-25 by williamspatrick