openbmc/openbmc #1916: OCC: Unable to handle kernel paging request at virtual address e1a0c00d

amboar, 2017-07-07:
During testing host resilience to BMC reboots, I encounted the following oops on the BMC. At this point the BMC had been brought up, the host had been powered on to a petitboot shell and left there. The BMC was then rebooted.

I observed that the host was still powered on and waiting at the petitboot shell (i.e. chassis power hadn't been cut by the BMC reboot). Moments later the oops appeared on the BMC console:

```
root@witherspoon:~# [  108.070000] Unable to handle kernel paging request at virtual address e1a0c00d
[  108.070000] pgd = 9d7a0000
[  108.070000] [e1a0c00d] *pgd=00000000
[  108.070000] Internal error: Oops: 5 [#1] ARM
[  108.070000] CPU: 0 PID: 1201 Comm: sh Not tainted 4.10.17-990f06c35b7e119f00967e90f0514dbbd2e1b51b #1
[  108.070000] Hardware name: ASpeed SoC
[  108.070000] task: 9d60e4a0 task.stack: 9dc2e000
[  108.070000] PC is at strlen+0xc/0x38
[  108.070000] LR is at kernfs_name_hash+0x20/0x78
[  108.070000] pc : [<801e3f78>]    lr : [<80142578>]    psr: 60000013
[  108.070000] sp : 9dc2f908  ip : 9dc2f918  fp : 9dc2f914
[  108.070000] r10: 9c94ba5c  r9 : 00000000  r8 : 8061c654
[  108.070000] r7 : e1a0c00d  r6 : 00000000  r5 : e1a0c00d  r4 : 00000000
[  108.070000] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : e1a0c00d
[  108.070000] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  108.070000] Control: 00c5387d  Table: 9d7a0008  DAC: 00000055
[  108.070000] Process sh (pid: 1201, stack limit = 0x9dc2e188)
[  108.070000] Stack: (0x9dc2f908 to 0x9dc30000)
[  108.070000] f900:                   9dc2f92c 9dc2f918 80142578 801e3f78 9c94d600 9c94d960
[  108.070000] f920: 9dc2f954 9dc2f930 80142688 80142564 9e4017c8 9c94d7d0 e1a0c00d 9c94d960
[  108.070000] f940: 00000000 9e6fb068 9dc2f974 9dc2f958 80143e08 80142644 9e5b3814 9c94d960
[  108.070000] f960: 00000000 9e6fb068 9dc2f994 9dc2f978 8014635c 80143dd8 9c94d960 9e6fb068
[  108.070000] f980: 9c94ba10 00000000 9dc2f9bc 9dc2f998 80146734 80146320 9dc2f9c4 9dc2f9a8
[  108.070000] f9a0: 80143e2c 9e6fb07c 9c94ba10 805eab3c 9dc2f9d4 9dc2f9c0 80146854 801466f0
[  108.070000] f9c0: 9c94ba08 9c94ba10 9dc2f9f4 9dc2f9d8 8023c4d8 8014682c 9e6a0354 9c94ba08
[  108.070000] f9e0: 9c94ba10 9c8f1c10 9dc2fa34 9dc2f9f8 8023f760 8023c498 80142578 9c94ba08
[  108.070000] fa00: 00000005 9c8f1c10 9dc2fa80 9c94ba08 00000005 9c8f1c10 9dc2fa80 9e5b3800
[  108.070000] fa20: 00000000 8024585c 9dc2fa4c 9dc2fa38 8023f8ac 8023f580 9c94ba08 00000005
[  108.070000] fa40: 9dc2fa6c 9dc2fa50 802de320 8023f894 9dc2fa7c 00000004 800de56c 9c94e640
[  108.070000] fa60: 9dc2fa7c 9dc2fa70 802de37c 802de2e8 9dc2fab4 9dc2fa80 802457cc 802de36c
[  108.070000] fa80: 9e6fa000 9c937480 9dc2fac4 9c8f1c10 9c8f1c44 00000000 805eb070 00000034
[  108.070000] faa0: 00000000 9c8f1c64 9dc2fac4 9dc2fab8 80245cf0 80245674 9dc2faec 9dc2fac8
[  108.070000] fac0: 80242778 80245cc8 9e425b8c 9c8f1c10 805e5570 9c8e8410 8061c654 00000000
[  108.070000] fae0: 9dc2fafc 9dc2faf0 8024281c 80242630 9dc2fb1c 9dc2fb00 80240a38 80242808
[  108.070000] fb00: 9c8f1c64 9c8f1c10 9c8f1c18 9c8e8410 9dc2fb5c 9dc2fb20 8023f768 8024095c
[  108.070000] fb20: 9c8e43e0 805e988c 9dc2fb44 9dc2fb38 800c17a0 9c8f1c10 00000000 802d0f30
[  108.070000] fb40: 805e988c 00000034 00000000 9c8e8464 9dc2fb74 9dc2fb60 8023f8ac 8023f580
[  108.070000] fb60: 9c8f1c10 00000000 9dc2fb84 9dc2fb78 802fdeb0 8023f894 9dc2fb9c 9dc2fb88
[  108.070000] fb80: 802d0f50 802fdea0 00000000 00000000 9dc2fbc4 9dc2fba0 8023d9cc 802d0f3c
[  108.070000] fba0: 9c8c71e0 9e5d3f90 9c89c910 9c89c910 9c8e8400 00000000 9dc2fbdc 9dc2fbc8
[  108.070000] fbc0: 802d0f14 8023d974 9c8e8410 9c8e8410 9dc2fbf4 9dc2fbe0 80243dc0 802d0ee4
[  108.070000] fbe0: 9c8e8410 9c8e8444 9dc2fc1c 9dc2fbf8 80242768 80243d98 9e425b8c 9c8e8410
[  108.070000] fc00: 805e5570 9c8c5600 8061c654 00000000 9dc2fc2c 9dc2fc20 8024281c 80242630
[  108.070000] fc20: 9dc2fc4c 9dc2fc30 80240a38 80242808 9c8e8464 9c8e8410 9c8e8418 9c8c5600
[  108.070000] fc40: 9dc2fc8c 9dc2fc50 8023f768 8024095c 80142ee0 800ddda4 9e4017c8 801431b4
[  108.070000] fc60: 805dd8f4 9c8e8410 00000000 802cf41c 9c8c5600 00000034 00000000 9c8c5654
[  108.070000] fc80: 9dc2fca4 9dc2fc90 8023f8ac 8023f580 9c8e8410 00000000 9dc2fcb4 9dc2fca8
[  108.070000] fca0: 802fdeb0 8023f894 9dc2fccc 9dc2fcb8 802cf43c 802fdea0 00000000 00000000
[  108.070000] fcc0: 9dc2fcf4 9dc2fcd0 8023d9cc 802cf428 9c8c7f00 9c8c71f0 9dc2fd0c 8061d484
[  108.070000] fce0: 9c89cb40 8061d4d0 9dc2fd24 9dc2fcf8 802cf36c 8023d974 802cf2fc 9c8c5600
[  108.070000] fd00: 9c8c5634 00000000 805e9830 00000034 00000000 9c8c5654 9dc2fd4c 9dc2fd28
[  108.070000] fd20: 80242768 802cf308 9c89cc2c 9c8c5600 805e9204 9c8b6000 8061c654 00000000
[  108.070000] fd40: 9dc2fd5c 9dc2fd50 8024281c 80242630 9dc2fd7c 9dc2fd60 80240a38 80242808
[  108.070000] fd60: 9c8c5654 9c8c5600 9c8c5608 9c8b6000 9dc2fdbc 9dc2fd80 8023f768 8024095c
[  108.070000] fd80: 801de040 800c177c 9dc2fe0c 9dc2fd98 9c89a140 9c8c5600 00000000 802cc168
[  108.070000] fda0: 9d673080 00000000 00000000 00000000 9dc2fdd4 9dc2fdc0 8023f8ac 8023f580
[  108.070000] fdc0: 00000000 00000000 9dc2fde4 9dc2fdd8 802cc180 8023f894 9dc2fe0c 9dc2fde8
[  108.070000] fde0: 8023d9cc 802cc174 9c89a140 9c8c7f10 0000000e 9c8b6000 00000000 802cc130
[  108.070000] fe00: 9dc2fe24 9dc2fe10 802cc154 8023d974 00000000 00000000 9dc2fe4c 9dc2fe28
[  108.070000] fe20: 8023d9cc 802cc13c 9c89a1a0 9c89a150 9e54f480 00000002 9c8b6210 9dc2ff78
[  108.070000] fe40: 9dc2fe64 9dc2fe50 802cce88 8023d974 802cce60 00000002 9dc2fe7c 9dc2fe68
[  108.070000] fe60: 8023c350 802cce6c 8023c328 00000002 9dc2fe94 9dc2fe80 801453d4 8023c334
[  108.070000] fe80: 9dc22b00 00000002 9dc2fed4 9dc2fe98 80144acc 80145390 00000000 00000000
[  108.070000] fea0: 9dc2fed4 9dc22b0c 80049e08 9caa8860 801449e8 9dc2ff78 00000002 00000000
[  108.070000] fec0: 00000002 00000000 9dc2ff44 9dc2fed8 800e246c 801449f4 800ff01c 60000013
[... 51 lines cut ...]
[  108.070000] [<802cce88>] (master_rescan_store) from [<8023c350>] (dev_attr_store+0x28/0x34)
[  108.070000] [<8023c350>] (dev_attr_store) from [<801453d4>] (sysfs_kf_write+0x50/0x54)
[  108.070000] [<801453d4>] (sysfs_kf_write) from [<80144acc>] (kernfs_fop_write+0xe4/0x1e4)
[  108.070000] [<80144acc>] (kernfs_fop_write) from [<800e246c>] (__vfs_write+0x3c/0x128)
[  108.070000] [<800e246c>] (__vfs_write) from [<800e26e8>] (vfs_write+0xb0/0x178)
[  108.070000] [<800e26e8>] (vfs_write) from [<800e289c>] (SyS_write+0x4c/0xa0)
[  108.070000] [<800e289c>] (SyS_write) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[  108.070000] Code: 80410dcc e1a0c00d e92dd800 e24cb004 (e5d03000)
[  108.070000] ---[ end trace 15554b242bd00776 ]---
```
OpenBMC version:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-74"
VERSION_ID="v1.99.7-74-gec3c51e"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-74"
BUILD_ID="v1.99.7"
```
Kernel:
```
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-990f06c35b7e119f00967e90f0514dbbd2e1b51b #1 Thu Jul 6 17:21:08 UTC 2017 armv6l GNU/Linux
```

eddiejames, 2017-07-07:
Having trouble reproducing this. Is it some delayed reboot you're doing? Was the occ hwmon driver bound at the time? If i do a BMC reboot the console goes away immediately. I don't see any oops. I also have tried a number of combinations of binding/unbinding occ hwmon and rescanning fsi

amboar, 2017-07-07:
If you haven't applied [this patch] (http://patchwork.ozlabs.org/patch/785386/) then you won't be able to reproduce. Further, after the BMC reboot it takes some time for the oops to occur; it happens after the BMC reaches a getty but before, I expect, it reaches the BMC_READY state. Once the oops occurs the BMC is mostly unresponsive in that the console fails and you can't ssh in, but some applications may continue to function (I don't know why).

eddiejames, 2017-07-10:
Hmm. Still can't reproduce. I power on the system, let it get up to the host login, then reboot the bmc. it comes back fine and system is still powered on. I have tried with occ-hwmon devices bound and unbound, no difference. I can also bind them once the bmc is rebooted.

Are you using some openbmc image that binds the occ-hwmon automatically? Didn't think Vishwa was done.

geissonator, 2017-07-13:
Test team took a build generated by Eddie and verified all was good on this path on their witherspoon.  Only issue found was https://github.com/openbmc/openbmc/issues/1948.

amboar, 2017-07-13:
@geissonator @eddiejames and whoever else: If you are satisfied, can you please reply on the openbmc list with Reviewed-by/Tested-by tags for the GPIO reset tolerance patch mentioned above?

gkeishin, 2017-07-17:
Doing the same operation to do reboot when Host is up.  Post that looks like the reboot doesnt reset the BMC anymore

```
Apr 20 17:44:18 witherspoon systemd[1]: fsi-scan@0.service: State 'stop-sigterm' timed out. Killing.
Apr 20 17:44:18 witherspoon systemd[1]: fsi-scan@0.service: Killing process 1888 (sh) with signal SIGKILL.
Apr 20 17:45:49 witherspoon systemd[1]: fsi-scan@0.service: Processes still around after SIGKILL. Ignoring.
Apr 20 17:47:19 witherspoon systemd[1]: fsi-scan@0.service: State 'stop-final-sigterm' timed out. Killing.
Apr 20 17:47:19 witherspoon systemd[1]: fsi-scan@0.service: Killing process 1888 (sh) with signal SIGKILL
```

```
[ 1372.270000] [<801e5394>] (string) from [<801e720c>] (vsnprintf+0x154/0x3a8)
[ 1372.270000] [<801e720c>] (vsnprintf) from [<801e79d0>] (vscnprintf+0x14/0x2c)
[ 1372.270000] [<801e79d0>] (vscnprintf) from [<80047fc0>] (vprintk_emit+0xc8/0x424)
[ 1372.270000] [<80047fc0>] (vprintk_emit) from [<800484c8>] (vprintk_default+0x30/0x38)
[ 1372.270000] [<800484c8>] (vprintk_default) from [<800484e8>] (vprintk+0x18/0x1c)
[ 1372.270000] [<800484e8>] (vprintk) from [<80018cf0>] (__warn+0x7c/0x108)
[ 1372.270000] [<80018cf0>] (__warn) from [<80018dc0>] (warn_slowpath_fmt+0x44/0x4c)
[ 1372.270000] [<80018dc0>] (warn_slowpath_fmt) from [<801ddf54>] (kobject_get+0x6c/0xac)
[ 1372.270000] [<801ddf54>] (kobject_get) from [<8023faa4>] (device_shutdown+0x80/0x19c)
[ 1372.270000] [<8023faa4>] (device_shutdown) from [<80036348>] (kernel_restart_prepare+0x3c/0x48)
[ 1372.270000] [<80036348>] (kernel_restart_prepare) from [<80036484>] (kernel_restart+0x1c/0x60)
[ 1372.270000] [<80036484>] (kernel_restart) from [<80036780>] (SyS_reboot+0x114/0x1e0)
[ 1372.270000] [<80036780>] (SyS_reboot) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[ 1372.270000] Code: 3a000019 e3530000 e243e001 0a00001b (e5d2c000) 
[ 1372.270000] ---[ end trace 34c11d376e51f5c9 ]---
Segmentation fault
Execute reboot -f if all unmounted ok, or exec /init
shutdown-sh#
```
- [x] dmesg log 
[Kernel_crashed.txt](https://github.com/openbmc/openbmc/files/1153210/Kernel_crashed.txt)

mdmillerii, 2017-07-17:
This is a device being called for shutdown but the kobject in the struct device was not initialized.

Unfornately the debug can be worse than the observation when the name is also uninitialized.

gkeishin, 2017-07-20:
We are seeing this on GHE build as well.
```
root@witherspoon:~# dmesg
[   17.200000] Hardware name: ASpeed SoC
[   17.200000] task: 9c9073e0 task.stack: 9d0d6000
[   17.200000] PC is at strlen+0xc/0x38
[   17.200000] LR is at kobject_get_path+0x28/0x100
[   17.200000] pc : [<801e3fd8>]    lr : [<801ddc9c>]    psr: 20000013
[   17.200000] sp : 9d0d7da0  ip : 9d0d7db0  fp : 9d0d7dac
[   17.200000] r10: 804f6580  r9 : 9e42ed40  r8 : 8042c1a0
[   17.200000] r7 : 014000c0  r6 : 9c98b458  r5 : 9c98b458  r4 : 00000001
[   17.200000] r3 : 61765f34  r2 : 00000000  r1 : 014000c0  r0 : 61765f34
[   17.200000] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   17.200000] Control: 00c5387d  Table: 9cac0008  DAC: 00000055
[   17.200000] Process udevadm (pid: 761, stack limit = 0x9d0d6188)
[   17.200000] Stack: (0x9d0d7da0 to 0x9d0d8000)
[   17.200000] 7da0: 9d0d7ddc 9d0d7db0 801ddc9c 801e3fd8 800f8904 9c98b458 9c98b458 00000001
[   17.200000] 7dc0: 00000000 9d0bc000 8042c1a0 9e42ed40 9d0d7e34 9d0d7de0 801df1e0 801ddc80
[   17.200000] 7de0: 800ebec8 80102138 9d0d7ea0 00000000 00000000 9d0d7e60 9d0d7f54 00020001
[   17.200000] 7e00: 80515334 00000000 800ef108 00000003 9c98b450 9d0d7f78 9d0b7e40 00000000
[   17.200000] 7e20: 00000000 00000000 9d0d7e44 9d0d7e38 801df708 801df0e0 9d0d7e64 9d0d7e48
[   17.200000] 7e40: 8023e86c 801df700 9d12a340 00000000 8023e810 00000003 9d0d7e7c 9d0d7e68
[   17.200000] 7e60: 8023c3f0 8023e81c 8023c3c8 00000003 9d0d7e94 9d0d7e80 8014541c 8023c3d4
[   17.200000] 7e80: 9d12a340 00000003 9d0d7ed4 9d0d7e98 80144b14 801453d8 00000000 00000000
[   17.200000] 7ea0: 9e7f43f0 9d12a34c db859a5a 9d174c80 80144a30 9d0d7f78 00000003 00000000
[   17.200000] 7ec0: 00000003 00000000 9d0d7f44 9d0d7ed8 800e24b4 80144a3c 00000000 9d0d7ee0
[   17.200000] 7ee0: 00000003 a0000013 9d0bc000 800ed1e4 9d174c80 fffff000 9d174c88 9e220290
[   17.200000] 7f00: 9d0d7f2c 9d0d7f10 800dde24 800de0a8 9d0bc000 800e26f4 9d0bc000 9d174c80
[   17.200000] 7f20: 00000003 9d174c80 7e998f64 9d0d7f78 00000000 00000003 9d0d7f74 9d0d7f48
[   17.200000] 7f40: 800e2730 800e2484 00000000 800ffa0c 9d174c80 9d174c80 00000000 00000000
[   17.200000] 7f60: 7e998f64 00000003 9d0d7fa4 9d0d7f78 800e28e4 800e268c 00000000 00000000
[   17.200000] 7f80: 554e4c98 7e998424 54b94110 00000004 8000a6e4 9d0d6000 00000000 9d0d7fa8
[   17.200000] 7fa0: 8000a540 800e28a4 554e4c98 7e998424 00000003 7e998f64 00000003 00000008
[   17.200000] 7fc0: 554e4c98 7e998424 54b94110 00000004 00000003 7e998f64 54b7e708 54b940e8
[   17.200000] 7fe0: 00000000 7e9983fc 54b3a60c 76e8e33c 60000010 00000003 00000000 00000000
[   17.200000] [<801e3fd8>] (strlen) from [<801ddc9c>] (kobject_get_path+0x28/0x100)
[   17.200000] [<801ddc9c>] (kobject_get_path) from [<801df1e0>] (kobject_uevent_env+0x10c/0x620)
[   17.200000] [<801df1e0>] (kobject_uevent_env) from [<801df708>] (kobject_uevent+0x14/0x18)
[   17.200000] [<801df708>] (kobject_uevent) from [<8023e86c>] (uevent_store+0x5c/0x6c)
[   17.200000] [<8023e86c>] (uevent_store) from [<8023c3f0>] (dev_attr_store+0x28/0x34)
[   17.200000] [<8023c3f0>] (dev_attr_store) from [<8014541c>] (sysfs_kf_write+0x50/0x54)
[   17.200000] [<8014541c>] (sysfs_kf_write) from [<80144b14>] (kernfs_fop_write+0xe4/0x1e4)
[   17.200000] [<80144b14>] (kernfs_fop_write) from [<800e24b4>] (__vfs_write+0x3c/0x128)
[   17.200000] [<800e24b4>] (__vfs_write) from [<800e2730>] (vfs_write+0xb0/0x178)
[   17.200000] [<800e2730>] (vfs_write) from [<800e28e4>] (SyS_write+0x4c/0xa0)
[   17.200000] [<800e28e4>] (SyS_write) from [<8000a540>] (ret_fast_syscall+0x0/0x34)
[   17.200000] Code: 80417f2c e1a0c00d e92dd800 e24cb004 (e5d03000)
[   17.650000] ---[ end trace 07355a026aad28a9 ]---
[   21.900000] Unable to handle kernel paging request at virtual address 61765f34
[   21.900000] pgd = 9cb80000
[   21.900000] [61765f34] *pgd=00000000
[   21.900000] Internal error: Oops: 5 [#2] ARM
[   21.900000] CPU: 0 PID: 1118 Comm: udevadm Tainted: G      D         4.10.17-b2cd8101eec3c200940405f12582473617a520f7 #1
[   21.900000] Hardware name: ASpeed SoC
[   21.900000] task: 9e680040 task.stack: 9d19c000
[   21.900000] PC is at strlen+0xc/0x38
[   21.900000] LR is at kobject_get_path+0x28/0x100
[   21.900000] pc : [<801e3fd8>]    lr : [<801ddc9c>]    psr: 20000013
[   21.900000] sp : 9d19dda0  ip : 9d19ddb0  fp : 9d19ddac
[   21.900000] r10: 804f6580  r9 : 9e42ed40  r8 : 8042c1a0
[   21.900000] r7 : 014000c0  r6 : 9c98b458  r5 : 9c98b458  r4 : 00000001
[   21.900000] r3 : 61765f34  r2 : 00000000  r1 : 014000c0  r0 : 61765f34
[   21.900000] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   21.900000] Control: 00c5387d  Table: 9cb80008  DAC: 00000055
[   21.900000] Process udevadm (pid: 1118, stack limit = 0x9d19c188)
[   21.900000] Stack: (0x9d19dda0 to 0x9d19e000)
[   21.900000] dda0: 9d19dddc 9d19ddb0 801ddc9c 801e3fd8 800f8904 9c98b458 9c98b458 00000001
[   21.900000] ddc0: 00000000 9d51a000 8042c1a0 9e42ed40 9d19de34 9d19dde0 801df1e0 801ddc80
[   21.900000] dde0: 800ebec8 80102138 9d19dea0 00000000 00000000 9d19de60 9d19df54 00020001
[   21.900000] de00: 80515334 00000000 800ef108 00000003 9c98b450 9d19df78 9ca6fc40 00000000
[... 112 lines cut ...]
[  100.890000] [<8014541c>] (sysfs_kf_write) from [<80144b14>] (kernfs_fop_write+0xe4/0x1e4)
[  100.890000] [<80144b14>] (kernfs_fop_write) from [<800e24b4>] (__vfs_write+0x3c/0x128)
[  100.890000] [<800e24b4>] (__vfs_write) from [<800e2730>] (vfs_write+0xb0/0x178)
[... cut ...]
[  185.400000] watchdog: watchdog0: watchdog did not stop!
root@witherspoon:~#
```

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.8-41"
VERSION_ID="v1.99.8-41-g86a4abc"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.8-41"
BUILD_ID="v1.99.8"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-b2cd8101eec3c200940405f12582473617a520f7 #1 Wed Jul 19 08:19:38 CDT 2017 armv6l GNU/Linux
root@witherspoon:~#
```

amboar, 2017-07-20:
@gkeishin I suspect these are independent bugs, one in the I2C driver and another in the OCC driver. Can you please open a separate issue for the I2C oops? If it's shown they are the same then we can mark them as duplicate.

gkeishin, 2017-07-21:
Sure.. I ll open another defect for it separately. Thanks

gkeishin, 2017-08-19:
In latest master, on reboot
```
[  OK  ] Stopped target Host0 (Reset Check).
         Stopping Phosphor Host State Manager...
[  OK  ] Stopped Wait for Power0 to turn on.
[  OK  ] Removed slice system-op\x2dwait\x2dpower\x2don.slice.
[  OK  ] Removed slice system-phosphor\x2dfan\x2dpresence\x2dtach.slice.
[ 1591.450000] Unable to handle kernel paging request at virtual address 00001114
[ 1591.450000] pgd = 80004000
[ 1591.450000] [00001114] *pgd=00000000
[ 1591.450000] Internal error: Oops: 5 [#1] ARM
[ 1591.450000] CPU: 0 PID: 5 Comm: kworker/u2:0 Not tainted 4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1
[ 1591.450000] Hardware name: ASpeed SoC
[ 1591.450000] Workqueue: occ occ_worker
[ 1591.450000] task: 9e42ab80 task.stack: 9e4a8000
[ 1591.450000] PC is at occ_worker+0x100/0x3a4
[ 1591.450000] LR is at occ_worker+0x5c/0x3a4
[ 1591.450000] pc : [<802d2674>]    lr : [<802d25d0>]    psr: 20000093
[ 1591.450000] sp : 9e4a9ea8  ip : 00000100  fp : 9e4a9efc
[ 1591.450000] r10: 9d14a200  r9 : 9d51f2b4  r8 : 00000000
[ 1591.450000] r7 : 9d51f2a0  r6 : 00000100  r5 : 00000100  r4 : 00001100
[ 1591.450000] r3 : 00000002  r2 : 20000093  r1 : 00000003  r0 : 00000200
[ 1591.450000] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 1591.450000] Control: 00c5387d  Table: 9d174008  DAC: 00000055
[ 1591.450000] Process kworker/u2:0 (pid: 5, stack limit = 0x9e4a8188)
[ 1591.450000] Stack: (0x9e4a9ea8 to 0x9e4aa000)
[ 1591.450000] 9ea0:                   9dd6c018 9d51f2ac 9dfbcc00 9d51f2a8 02a2dec0 00000000
[ 1591.450000] 9ec0: 00000000 35d00600 00000120 00000000 9e4a9efc 9d51f2b4 9e407e00 9e402200
[ 1591.450000] 9ee0: 00000000 9e406e00 00000000 9e407e00 9e4a9f34 9e4a9f00 8002d21c 802d2580
[ 1591.450000] 9f00: 9e4a8000 00000088 9e402214 9e402200 9e407e18 805de340 9e4a8000 00000088
[ 1591.450000] 9f20: 9e402214 9e407e00 9e4a9f74 9e4a9f38 8002d4e8 8002d048 00000000 00000000
[ 1591.450000] 9f40: 9e40235c 9e4a8000 9e401c18 9e401c00 00000000 9e403e40 9e4a8000 9e407e00
[ 1591.450000] 9f60: 9e401c18 9e4a1e94 9e4a9fac 9e4a9f78 800335ec 8002d498 9e403e40 8002d48c
[ 1591.450000] 9f80: 00000000 9e403e40 800334c8 00000000 00000000 00000000 00000000 00000000
[ 1591.450000] 9fa0: 00000000 9e4a9fb0 8000a5e8 800334d4 00000000 00000000 00000000 00000000
[ 1591.450000] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1591.450000] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000 e22ef1db 6b45edfe
[ 1591.450000] [<802d2674>] (occ_worker) from [<8002d21c>] (process_one_work+0x1e0/0x450)
[ 1591.450000] [<8002d21c>] (process_one_work) from [<8002d4e8>] (worker_thread+0x5c/0x570)
[ 1591.450000] [<8002d4e8>] (worker_thread) from [<800335ec>] (kthread+0x124/0x164)
[ 1591.450000] [<800335ec>] (kthread) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)
[ 1591.450000] Code: 0a00004c e10f2000 f10c0080 e2854a01 (e5943014)
[ 1591.450000] ---[ end trace 1dd6a7494e2e3f3e ]---
[  OK  ] Removed slice system-phosphor\x2dwatchdog.slice.
[  OK  ] Removed slice system-phosphor\x2dgpio\x2dmonitor.slice.
[ 1592.300000] Unable to handle kernel paging request at virtual address 00bd0960
[ 1592.300000] pgd = 9ca98000
[ 1592.300000] [00bd0960] *pgd=00000000
[ 1592.300000] Internal error: Oops: 5 [#2] ARM
[ 1592.300000] CPU: 0 PID: 973 Comm: dbus-daemon Tainted: G      D         4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1
[ 1592.300000] Hardware name: ASpeed SoC
[ 1592.300000] task: 9d1a47a0 task.stack: 9d416000
[ 1592.300000] PC is at skb_copy_datagram_iter+0xd0/0x228
[ 1592.300000] LR is at unix_stream_read_actor+0x30/0x3c
[ 1592.300000] pc : [<80322b64>]    lr : [<803ac5e8>]    psr: a0000013
[ 1592.300000] sp : 9d417d10  ip : 9d417d48  fp : 9d417d44
[ 1592.300000] r10: 9d51f300  r9 : 803ac5b8  r8 : fd5e02b5
[ 1592.300000] r7 : 00000024  r6 : 00bd0960  r5 : 000004e4  r4 : 000004e4
[ 1592.300000] r3 : 000004e4  r2 : 9d417f50  r1 : 02b70000  r0 : 00000024
[ 1592.300000] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[ 1592.300000] Control: 00c5387d  Table: 9ca98008  DAC: 00000055
[ 1592.300000] Process dbus-daemon (pid: 973, stack limit = 0x9d416188)
[ 1592.300000] Stack: (0x9d417d10 to 0x9d418000)
[ 1592.300000] 7d00:                                     00000024 9d417f50 9cb756c0 000004e4
[ 1592.300000] 7d20: 00000000 9d1fa090 000004e4 9d1fa000 803ac5b8 9d417dfc 9d417d5c 9d417d48
[ 1592.300000] 7d40: 803ac5e8 80322aa0 9d51f300 00000000 9d417df4 9d417d60 803ad754 803ac5c4
[ 1592.300000] 7d60: 9d417e48 7ed616dc 00000010 00000001 9d416000 00000000 00000001 9d1fa1cc
[ 1592.300000] 7d80: 00000001 9e3831e0 00000000 00000000 00000055 00000371 00000000 0000031c
[ 1592.300000] 7da0: 00000000 00000000 00000000 00000000 00000000 00000001 9d417e88 9e3831e0
[ 1592.300000] 7dc0: 9d417e04 9d417dd0 801f17ac 803adc84 00000000 40000040 7ed61884 7ed61810
[... 244 lines cut ...]
[ 1964.760000] UBIFS (ubi7:2): un-mount UBI device 7
[ 1964.760000] UBIFS (ubi7:2): background thread "ubifs_bgt7_2" stops
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Unmounted /var/volatile.
[FAILED] Failed unmounting /media/pnor-prsv.
[  OK  ] Unmounted /media/pnor-ro-fd2cbe15.
[  OK  ] Unmounted /media/pnor-rw-fd2cbe15.
[  OK  ] Unmounted Temporary Directory.
[  OK  ] Stopped target Swap.
[  OK  ] Unmounted /usr/local/share/pnor.
         Unmounting /usr/local...
[  OK  ] Unmounted /usr/local.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
[... cut ...]
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
[ 1966.220000] watchdog: watchdog0: watchdog did not stop!
[ 1966.220000] systemd-shutdow: 35 output lines suppressed due to ratelimiting
[ 1966.320000] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[ 1966.320000] systemd-journald[569]: Received SIGTERM from PID 1 (systemd-shutdow).
[ 2056.430000] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[ 2056.470000] systemd-shutdown[1]: Sending SIGKILL to PID 973 (dbus-daemon).

```

gkeishin, 2017-08-21:
We see this in 984 Master build, BMC r/r when OS is booted
```
root@witherspoon:~# reboot
[  OK  ] Stopped target Powerwitherspoon-fan-watchdog (On).
         Stopping Phosphor Hwmon Poller...
[  OK  ] Stopped target Start Host0.
[  OK  ] Stopped target Chassis0 power on after reset.
[  OK  ] Stopped target Chassis0 (Power On).
[  OK  ] Stopped target Host0 (Started).
[  OK  ] Stopped Assert bmc_booted LED.
         Stopping Phosphor Hwmon Poller...
[  OK  ] Stopped target Fan Control Ready.
         Stopping Phosphor Hwmon Poller...
[  OK  ] Stopped Witherspoon Fan Watchdog Reset.
         Stopping Phosphor sysfs LED controller...
[  OK  ] Removed slice system-obmc\x2denable\x2dhost\x2dwatchdog.slice.
[  OK  ] Removed slice system-getty.slice.
[  OK  ] Stopped Wait for /xyz/openbmc_project/led/groups/bmc_booted.
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor Hwmon Poller...
[  OK  ] Stopped target Timers.
[  OK  ] Stopped PCIE Slot Presence Detection For Chassis0.
[  OK  ] Removed slice system-pcie\x2dslot\x2ddetect.slice.
         Stopping Phosphor sysfs LED controller...
[  OK  ] Removed slice system-op\x2dwait\x2dpower\x2doff.slice.
[ 1119.960000] occ-hwmon occ1-dev0: occ bus op failed rc:-4
         Stopping Phosphor poweron watchdog...
[  OK  ] Stopped Daily Cleanup of Temporary Directories.
[  OK  ] Stopped Wait for /xyz/openbmc_project/state/chassis0.
[  OK  ] Removed slice system-dropbear.slice.
[  OK  ] Removed slice system-phosphor\x2dfan\x2dcontrol\x2dinit.slice.
[  OK  ] Removed slice system-fsi\x2dbind.slice.
         Stopping Phosphor Webserver...
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor Hwmon Poller...
[  OK  ] Stopped Wait for /org/openbmc/control/chassis0.
         Stopping Phosphor Host Console SSH Per-Connection (9.3.62.216:45400)...
[  OK  ] Stopped Wait for /xyz/openbmc_project/state/host0.
[ 1121.510000] occ-hwmon occ2-dev0: occ bus op failed rc:-4
         Stopping Phosphor Hwmon Poller...
         Stopping Phosphor Fan Presence Tach Daemon...
[  OK  ] Removed slice system-phosphor\x2dreset\x2dhost\x2drunning.slice.
[  OK  ] Removed slice system-cfam_override.slice.
[  OK  ] Removed slice system-vcs_off_poweroff.slice.
[  OK  ] Removed slice system-phosphor\x2dreboot\x2dhost.slice.
[  OK  ] Stopped Wait for /xyz/openbmc_project/control/host0.
[ 1121.510000] watchdog: watchdog0: watchdog did not stop!
         Stopping Witherspoon Fan Watchdog Daemon...
[  OK  ] Stopped target System Time Synchronized.
         Stopping Phosphor GPIO checkstop monitor...
         Stopping Phosphor sysfs LED controller...
         Stopping Phosphor Fan Control Daemon...
[  OK  ] Removed slice system-phosphor\x2dreset\x2dhost\x2dcheck.slice.
[  OK  ] Removed slice system-op\x2dreset\x2dchassis\x2don.slice.
[  OK  ] Stopped Wait for /xyz/openbmc_project/watchdog/host0.
[  OK  ] Stopped Wait for /org/openbmc/HostIpmi/1.
[  OK  ] Stopped target Host0 (Starting).
[ 1122.040000] ------------[ cut here ]------------
[ 1122.040000] WARNING: CPU: 0 PID: 11 at /tmp/openbmc/build/tmp/work-shared/witherspoon/kernel-source/mm/slab_common.c:890 kmalloc_slab+0x98/0xb4
[ 1122.040000] CPU: 0 PID: 11 Comm: kworker/u2:1 Not tainted 4.10.17-1a4df30c39cf5ee0e3d2528c409787ccbb4a672a #1
[ 1122.040000] Hardware name: ASpeed SoC
[ 1122.040000] Workqueue: occ occ_worker
[ 1122.040000] [<80010080>] (unwind_backtrace) from [<8000d934>] (show_stack+0x20/0x24)
[ 1122.040000] [<8000d934>] (show_stack) from [<801dbf2c>] (dump_stack+0x20/0x28)
[ 1122.040000] [<801dbf2c>] (dump_stack) from [<80018d54>] (__warn+0xe0/0x108)
[ 1122.040000] [<80018d54>] (__warn) from [<80018e48>] (warn_slowpath_null+0x30/0x38)
[ 1122.040000] [<80018e48>] (warn_slowpath_null) from [<800c6d2c>] (kmalloc_slab+0x98/0xb4)
[ 1122.040000] [<800c6d2c>] (kmalloc_slab) from [<800ddba0>] (__kmalloc+0x28/0x134)
[... 32 lines cut ...]
[ 1122.040000] 3f20: 9e402214 9e592c80 9e5f3f74 9e5f3f38 8002d4e8 8002d048 00000000 00000000
[ 1122.040000] 3f40: 9e40235c 9e5f2000 9e5c81d8 9e5c81c0 00000000 9e5df6a0 9e5f2000 9e592c80
[ 1122.040000] 3f60: 9e5c81d8 9e4a9e9c 9e5f3fac 9e5f3f78 800335ec 8002d498 9e5df6a0 8002d48c
[ 1122.040000] 3f80: 00000000 9e5df6a0 800334c8 00000000 00000000 00000000 00000000 00000000
[ 1122.040000] 3fa0: 00000000 9e5f3fb0 8000a5e8 800334d4 00000000 00000000 00000000 00000000
[ 1122.040000] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 1122.040000] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[ 1122.040000] [<80040b40>] (__wake_up_common) from [<80040e28>] (__wake_up+0x30/0x3c)
[ 1122.040000] [<80040e28>] (__wake_up) from [<802d2740>] (occ_worker+0x1cc/0x3a4)
[ 1122.040000] [<802d2740>] (occ_worker) from [<8002d21c>] (process_one_work+0x1e0/0x450)
[ 1122.040000] [<8002d21c>] (process_one_work) from [<8002d4e8>] (worker_thread+0x5c/0x570)
[ 1122.040000] [<8002d4e8>] (worker_thread) from [<800335ec>] (kthread+0x124/0x164)
[ 1122.040000] [<800335ec>] (kthread) from [<8000a5e8>] (ret_from_fork+0x14/0x2c)
[ 1122.040000] Code: e8bd4000 e590c000 e1a07000 e150000c (e59c4000)
[ 1122.040000] ---[ end trace fe938989aba92c8b ]---
[  OK  ] Stopped Start host0 SBE.
[  OK  ] Removed slice system-start_host.slice.
[  OK  ] Removed slice system-ir35221\x2doff\x2dunbind.slice.
         Stopping Phosphor Hwmon Poller...
[  OK  ] Removed slice system-ir35221\x2doff\x2dbind.slice.
[  OK  ] Removed slice system-op\x2docc\x2ddisable.slice.
[  OK  ] Stopped target Start Host0 (Pre).
[  OK  ] Stopped Power on bind ir35221 device driver devs.

```

mdmillerii, 2017-08-21:
@shenki @eddiejames `trace fe938989aba92c8a` is an oversize kmalloc request without `__GFP_NOWARN`.

Maybe from an inlined `occ_getsram` or `occ_putsram` call site (`occ_putsram` is used exactly once).
There does appear to be -ENOMEM handling in those functions.

Is there a limit that makes sense in general that could be performed on the request or do we need to add  `__GFP_NOWARN`?   Do the callers properly handle the `-ENOMEM` return?

mdmillerii, 2017-09-21:
@eddiejames this was fixed with your occ work list fix right?

amboar, 2018-05-23:
The OCC has been significantly reworked since, and is undergoing a rewrite. Closing this as no-longer relevant.

closed 2018-05-23 by amboar