openbmc/openbmc #1918: [Witherspoon] : Rear power LED not turning ON after second power on.

rahulmah, 2017-07-07:
Rear power LED not turning ON after second power on. Tried multiple power on but power LED is not turing on. Power LED will not start working until we do BMC reboot.

See step to recreate for more info.

Step to recreate.
1. Make sure the system is off state and rear power LED is off
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
```
```
bash-4.1$ curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/led/physical/rear_power
{
  "data": {
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.Off"
  },
  "message": "200 OK",
  "status": "ok"
  ```
2. Power on the system using 'obmcutil poweron' from BMC.

3. Once system powers on, rear power LED is also ON.

root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
```
bash-4.1$ curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/led/physical/rear_power
{
  "data": {
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.On"
  },
  "message": "200 OK",
  "status": "ok"
  ```
4. Now power off the system using 'obmcutil poweroff' from BMC.

5. Power on the system for the second time 'obmcutil poweron' from BMC. Here you will see that Rear power LED will not turn ON. Waited until OS comes up. 
```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Running
```
```
bash-4.1$ curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/led/physical/rear_power
{
  "data": {
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.Off"
  },
  "message": "200 OK",
  "status": "ok"
}bash-4.1$
```
BMC and PNOR info:
```
BMC : v1.99.7-86-g51af974
PNOR:
IBM-witherspoon-ibm-OP9_v1.17_1.46
        op-build-v1.17-101-g1c57f18-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-158-ga1e0a047b2a0
        hostboot-0ffbb55
        linux-4.11.6-openpower1-p57dd16e
        petitboot-v1.4.3-pfac92ff
        machine-xml-5b59a1d
        occ-1dc97a6
        hostboot-binaries-711147e
        capp-ucode-9c73e9f
        sbe-0df81aa
```

FFDC : 
[journal_rear_poweron_led.txt](https://github.com/openbmc/openbmc/files/1130627/journal_rear_poweron_led.txt)

vishwabmc, 2017-07-18:
# **STATUS**

* 7/18:
Debugged this seem to have found the cause. Will test

vishwabmc, 2017-07-18:
Looks like the issue is due to `RemainAfterExit=true`. It seem to work fine with not having this and also putting a Conflicts between start/stop services

vishwabmc, 2017-07-19:
https://gerrit.openbmc-project.xyz/#/c/5376/

vishwabmc, 2017-07-19:
@rahulmah tested the patch and confirmed that the fix works.

rfrandse, 2017-07-19:
<https://gerrit.openbmc-project.xyz/5376> Fix a bug in driving LED during PowerOn and PowerOff
Fixes  #1918 [Witherspoon] : Rear power LED not turning ON after second power on.

closed 2017-07-24 by williamspatrick