openbmc/openbmc #1919: [Witherspoon] Rear power LED not Blinking at standby

rahulmah, 2017-07-07:
With BMC at ready state and Host at off state, Rear power LED is not Blinking. It is remains at off state.

```
root@witherspoon:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
root@witherspoon:~#
```
```
bash-4.1$ curl -c cjar -b cjar -k https://wsbmc002/xyz/openbmc_project/led/physical/rear_power
{
  "data": {
    "Color": "xyz.openbmc_project.Led.Physical.Palette.Unknown",
    "DutyOn": 50,
    "Period": 1000,
    "State": "xyz.openbmc_project.Led.Physical.Action.Off"   <--- Not blinking.
  },
  "message": "200 OK",
  "status": "ok"
```

BMC and PNOR info:
```
BMC : v1.99.7-86-g51af974
PNOR:
IBM-witherspoon-ibm-OP9_v1.17_1.46
        op-build-v1.17-101-g1c57f18-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-158-ga1e0a047b2a0
        hostboot-0ffbb55
        linux-4.11.6-openpower1-p57dd16e
        petitboot-v1.4.3-pfac92ff
        machine-xml-5b59a1d
        occ-1dc97a6
        hostboot-binaries-711147e
        capp-ucode-9c73e9f
        sbe-0df81aa
```

FFDC : 
[journal_power_blink.txt](https://github.com/openbmc/openbmc/files/1130669/journal_power_blink.txt)

vishwabmc, 2017-07-18:
# **STATUS**

* 7/18:
Debugged this and the reason why it does not result in the physical LED action is because `bmc_booted` Group does not have any LED populated.  The reason why `bmc_booted` is not populated is because the XML that gets used for building is back dated. I talked to @spinler and he said he would work on getting the latest XML. Once that is in, I can verify this.

rfrandse, 2017-07-25:
https://gerrit.openbmc-project.xyz/#/c/5379/
If applied, this commit will resolve openbmc/openbmc#1919

closed 2017-08-01 by williamspatrick