openbmc/openbmc #1921: Question how to change timeout of phosphor-softpoweroff and request of new obmcutil hard-poweroff command

yadron, 2017-07-07:
I've noted that after "obmcutil poweroff" need waiting for about 6 mins before actually host powered off.
Part of log (journald) during poweroff:

```
Jul 07 11:59:17 custom phosphor-softpoweroff[1844]: Host control signal values
Jul 07 11:59:17 custom phosphor-softpoweroff[1844]: Timer started waiting for host to shutdown
Jul 07 12:04:17 custom phosphor-softpoweroff[1844]: Timer expired
Jul 07 12:04:17 custom phosphor-softpoweroff[1844]: Host did not shutdown within configured time.
```

Question:

1. How to change timeout? The 5 mins it too long.

Request:

2. Need "obmcutil hardpoweroff" command. Does it exists? If not, could you please schedule addition of it? The idea is that if host frozen, then I can send "GPIO poweroff right now" signal (the workaround that already works as expected for my P8 - probably will work at Barreleye too - is "gpioutil -n BMC_POWER_UP -v 1").

williamspatrick, 2017-07-07:
There are two timeouts involved with shutting down.
1. A short timeout on the IPMI alert to the Host for shutdown.
2. A long timeout waiting for them to perform the shutdown.

The first timeout is really short.  The second one can be configured here: https://github.com/openbmc/phosphor-host-ipmid/blob/master/configure.ac#L107 . You can change the variables passed to configure this package for a particular machine if you do not like the defaults of 5 minutes.

The purpose of this timeout is to give the OS time to flush disk caches.  If we were to just pull power, the OS does not get a chance to flush their memory to disk and you can end up with data loss.  Ideally, for each machine someone would calculate "how long does it take to flush a max-config of memory out to a reasonable disk configuration" and they would set the timeout to that.  Anything less than that and you are risking data loss to your end users.

Since, in this case, we did not get the first timeout and only the second one, that means the host OS is actively running.  If they were hung, the first timeout would fire quickly and we would have shut the machine down.

If you look at the code for obmcutil you will see two commands:
- https://github.com/openbmc/skeleton/blob/master/pytools/obmcutil#L91
- https://github.com/openbmc/skeleton/blob/master/pytools/obmcutil#L77

If you want a power off without communicating to the host, you go to the chassis object and not the host object.  The host object always does a "soft power off" and the chassis object is effectively a "hard power off".  There is no need for additional development here.

closed 2017-07-07 by williamspatrick

yadron, 2017-07-10:
@williamspatrick 
, thank you for the detailed answer.
I've checked commands

obmcutil chassison
obmcutil chassisoff

- yes, it works. So yes, this issue can be closed.

Actually the original problem was that these commands undocumented (as far as I know) and before June the "poweroff" worked fast (without waiting for the host), so when it become slow (5 mins timeout) we were thinking that it is broken. After I've seen "Timer started waiting for host to shutdown" journald log message of course we've guessed why.
So this issue even if it is not actual already still could be very useful for other users/developers of OpenBMC/p8.

As I see, after chassison command platform powered on, but host not booting yet. The command poweron required after chassison. Or user may run poweron without prior chassison command.

The chassisoff exactly does hard power off.

One more thing that needs to be checked (I'll check it and then if so, then create separate issue): if host not booted at all (e.g. timeout in petitboot or shell in petitboot instead of host OS), then poweroff command should not wait for the host OS timeout (and currently it does waiting).

williamspatrick, 2017-07-10:
If there is a timeout in petitboot, they should respond to the shutdown request quickly. I don't see a reason for us to write special code to try to determine where the OS is in the boot sequence to decide how long to give them on shutdown, especially because those decision points are likely to be different with different host firmware structures (e.g. An x86 BIOS).

yadron, 2017-07-10:
Ok. How to contact petitboot team? Do they know that they should respond to the shutdown quickly?

williamspatrick, 2017-07-10:
Are you seeing evidence that they are ignoring the shutdown request from the BMC?  @sammj and @stewart-ibm could help and open-power/petitboot is their repo.