openbmc/openbmc #1922: Fan Control: Create a dbus object to enable/disable automatic fan control

msbarth, 2017-07-07:
Each fan control zone should create a dbus object that can be used to enable/disable automatic fan control. The ability to set a zone as active/inactive already exists, so this issue needs to address exposing the dbus object that a user could then enable or disable automatic fan control on a per zone basis.

Interface design discussion reference --> https://gerrit.openbmc-project.xyz/#/c/4897/

msbarth, 2017-08-11:
- [ ] Determine if https://gerrit.openbmc-project.xyz/#/c/4897/ interface will be used
- [ ] Create enable/disable fan control (from per zone) property & interface yaml
- [ ] Update zone code to serve enable/disable property
- [ ] Subscribe zone to property change signal of enable/disable property
- [ ] When enabled, automatic fan control function is active on the zone
       Update internal target speed value from current target and continue auto control
- [ ] When disabled, automatic fan control function is non-active on the zone
       Leave target speed at last set value (or go to full speed?)

amboar, 2017-08-13:
Out of curiosity, did anyone ever look at the kernel's of_thermal capability and the thermal devicetree bindings?

msbarth, 2017-08-16:
@amboar I have not looked into that, however I'm not sure how it would help in what this issue is going to address. The fan control application in userspace requires a way to enable/disable its control over the fan speeds automatically (for mfg, thermal characterization, etc..). This issue will bind a dbus property that a user could set to enable/disable the running of the automatic fan control application so fan speeds could be set manually.

rfrandse, 2018-08-20:
moved to https://github.com/ibm-openbmc/dev/issues/122

closed 2018-08-20 by rfrandse