openbmc/openbmc #1923: REST Upload - phosphor-gevent: DBusException: org.freedesktop.DBus.Error.NoReply

anoo1, 2017-07-07:
With today's driver, testing the REST Upload interface, the curl command to enumerate such as

"curl b cjar -k -H "Content-Type: application/json" -X GET https://$BMCIP/xyz/openbmc_project/software/enumerate"
"curl b cjar -k -H "Content-Type: application/json" -X GET https://$BMCIP/xyz/openbmc_project/enumerate"

would fail with "DBusException: org.freedesktop.DBus.Error.NoReply".

Restarted the phosphor-gevent service and rebooted the BMC and failure remained.

```
# curl -c cjar -b cjar -k -H "Content-Type: application/octet-stream" -X PUT -T witherspoon.pnor.squashfs.tar https://$BMCIP/upload/image/
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}

#  curl b cjar -k -H "Content-Type: application/json" -X GET https://$BMCIP/xyz/openbmc_project/software/enumerate              {
  "data": {
    "description": "Internal Server Error",
    "exception": "DBusException('Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.',)",
    "traceback": [
      "Traceback (most recent call last):",
      "  File \"/usr/lib/python2.7/site-packages/bottle.py\", line 858, in _handle",
      "    route, args = self.router.match(environ)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 1038, in custom_router_match",
      "    route.callback._setup(**args)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 170, in _setup",
      "    return self.setup(**kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 284, in setup",
      "    request.route_data['map'] = self.find(path)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 281, in find",
      "    self.mapper.get_subtree, path=path)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py\", line 203, in try_mapper_call",
      "    return f(**kw)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 72, in get_subtree",
      "    retries, interval)",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 42, in retry",
      "    return func()",
      "  File \"/usr/lib/python2.7/site-packages/obmc/mapper/bindings.py\", line 71, in <lambda>",
      "    path, depth, interfaces, signature='sias'),",
      "  File \"/usr/lib/python2.7/site-packages/dbus/proxies.py\", line 145, in __call__",
      "    **keywords)",
      "  File \"/usr/lib/python2.7/site-packages/dbus/connection.py\", line 651, in call_blocking",
      "    message, timeout)",
      "DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken."
    ]
  },
  "message": "500 Internal Server Error",
  "status": "error"
}
```

Attaching journal log.
[journal.Jul7.txt](https://github.com/openbmc/openbmc/files/1132263/journal.Jul7.txt)

williamspatrick, 2017-07-07:
Which version?  Is this due to the REST changes from @ojayanth I merged today or something else?

anoo1, 2017-07-08:
It was with the jenkins build from friday morning (https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/869/artifact/deploy/images/witherspoon/) that had this commit: https://gerrit.openbmc-project.xyz/#/c/5135/

After the errors, xcat team tried updating the BMC to the "latest build" (https://openpower.xyz/job/openbmc-build/distro=ubuntu,target=witherspoon/lastBuild/) from jenkins that was available at around 2:40pm CST.

dkodihal, 2017-07-10:
I don't think this should be seen in a build that has https://gerrit.openbmc-project.xyz/#/c/5165/. The xcat team not seeing a JSON response after an image upload is still an issue that needs to be fixed though - @ojayanth.

bradbishop, 2017-07-10:
I have a fix for the rest server not emitting json responses here:

https://gerrit.openbmc-project.xyz/#/c/5194

bradbishop, 2017-07-10:
```Jul 10 21:07:01 palmetto phosphor-gevent[923]: ::60f9:b676:8e91:f402%1101964284 - - [2017-07-10 21:07:01] "GET /xyz/openbmc_project/software/enumerate HTTP/1.1
" 200 543 0.123463
Jul 10 21:07:08 palmetto phosphor-version-software-manager[805]: tar: short read
Jul 10 21:07:08 palmetto phosphor-version-software-manager[805]: Error No manifest file
Jul 10 21:07:08 palmetto phosphor-version-software-manager[805]: An error when reading the Manifest file.
Jul 10 21:07:08 palmetto phosphor-version-software-manager[805]: Error processing image
Jul 10 21:07:20 palmetto phosphor-gevent[923]: ::60f9:b676:8e91:f402%1101964284 - - [2017-07-10 21:07:20] "POST /upload/image HTTP/1.1" 200 169 12.110375
Jul 10 21:07:20 palmetto phosphor-version-software-manager[805]: Untaring
```

bradbishop, 2017-07-10:
offending process is:

```773 root       1:18 /usr/sbin/openpower-update-manager```

saqibkh, 2017-07-10:
Fix

closed 2017-07-11 by williamspatrick