openbmc/openbmc #1924: [xCAT] First call to REST service after firmware power applied results in 503 Service Unavailable

whowutwut, 2017-07-07:
Possibly related to #1881, but will let you decide... 

I updated 2 ITC witherspoon boxes to firmware 1724B and 1 to 1728B and seen some intermittent issues booting and IPLing the nodes.  So I came across a defect saying we should pull the plug after the update to get hostboot to stabilize....  I did this before going home yesterday.  

Just went back to take a look at these nodes, so it's been 24 hours, they have just been sitting there...  The first call for REST to get "power status" returned 503 Service Unavailable 

```
[root@fs3 ~]# rpower mid05tor12cn[02,04,11] stat
mid05tor12cn11: Error: 503 Service Unavailable
mid05tor12cn04: Error: 503 Service Unavailable
mid05tor12cn02: Error: 503 Service Unavailable
```

Immediately after, it returns quickly....
```
[root@fs3 ~]# rpower mid05tor12cn[02,04,11] stat
mid05tor12cn11: off
mid05tor12cn04: off
mid05tor12cn02: off  
```

here's the command time in our logs... 
```
====================================================
[Date]       2017-07-07 16:36:18
[ClientType] cli
[Request]    rpower mid05tor12cn[02,04,11] stat
[Response]
mid05tor12cn11: Error: 503 Service Unavailable
mid05tor12cn04: Error: 503 Service Unavailable
mid05tor12cn02: Error: 503 Service Unavailable
[ElapsedTime] 31 s
====================================================
[Date]       2017-07-07 16:36:55
[ClientType] cli
[Request]    rpower mid05tor12cn[02,04,11] stat
[Response]
mid05tor12cn11: off
mid05tor12cn04: off
mid05tor12cn02: off
[ElapsedTime] 2 s
```

This is quite difficult to re-create because after the first call, it works OK .  But this is very bad for usability.  @gurevichmark , can we re-create this on firmware updates all the time?

williamspatrick, 2017-07-07:
I don't think this is the same as #1881, but I think we have seen this before.

What I suspect is going on is that, since all processes are being started in parallel, the REST server is able to come up before all of the dbus interfaces you want to interact with are up and ready.  You're making a REST call that is is getting through before the process in the backend that owns the dbus interface is up.  In a way, this is the opposite problem as #1881.

williamspatrick, 2017-07-07:
@bradbishop @geissonator - Any thoughts on this?  We could move external interfaces, like REST, to a separate target that happens after most of the internal code is running.  But, in the event something doesn't start up we would lose the REST APIs to pull debug data.  I think we actually have a REST API that indicates the BMC is "ready" already, so maybe we just push people to use that after a reset?

williamspatrick, 2017-07-07:
https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/State/BMC.interface.yaml#L14

whowutwut, 2017-07-07:
@williamspatrick would we need this bmc ready even after the bmc has been powered on for a few hours?

williamspatrick, 2017-07-07:
No.  Maybe I misunderstood your report then.

>So I came across a defect saying we should pull the plug after the update to get hostboot to stabilize.... I did this before going home yesterday.  Just went back to take a look at these nodes, so it's been 24 hours, they have just been sitting there... 

I read this to mean they were sitting powered off for 24 hours not that they were sitting at standby for 24 hours.

whowutwut, 2017-07-07:
Yea. I pulled the power cords and plugged them back last night before leaving the office but the China team maybe was not in the office so they were not booted... just the BMC had power.  Today I was brining up other nodes and didn't get around to looking at them until 4 and noticed this 503 error on the first call to REST,

gurevichmark, 2017-07-10:
I am not seeing "503 Service Unavailable" after manual BMC firmware update and reboot anymore. The very first `curl` login takes maybe 5 seconds, but always succeeds now.

closed 2019-04-27 by stale[bot]