openbmc/openbmc #1925: [Witherspoon] REST response does not contain data, message and status fields with latest openbmc build.

rahulmah, 2017-07-08:
With latest OpenBMC build(v1.99.7-107-gf9707d1), REST response seem to be broken. REST response does not have data, message and status fields while doing GET operation.

Here are two example for the failure.
```
bash-4.1$ curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/state/host0
{                                       <--- data field not present
  "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
  "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
}                                         <--- message, status not present
```

```
bash-4.1$ curl -c cjar -b cjar -k https://<BMC_IP>/xyz/openbmc_project/software/4c8a581c
{
  "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
  "Path": "",
  "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
  "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
  "Version": "v1.99.7-107-gf9707d1"
}bash-4.1$
```

With older openBMC build (v1.99.7-94-g9994014) all these fields were visible.

```
bash-4.1$ curl -c cjar -b cjar -k https://<other_BMC_IP>/xyz/openbmc_project/state/host0
{
  "data": {                   <--- Present
    "CurrentHostState": "xyz.openbmc_project.State.Host.HostState.Off",
    "RequestedHostTransition": "xyz.openbmc_project.State.Host.Transition.Off"
  },
  "message": "200 OK",        <--- Present
  "status": "ok"                       <--- Present
}bash-4.1$
```
bash-4.1$ curl -c cjar -b cjar -k https://<other_BMC_IP>/xyz/openbmc_project/software/42376d25
{
  "data": {                               <--- Present
    "Activation": "xyz.openbmc_project.Software.Activation.Activations.Active",
    "Path": "",
    "Purpose": "xyz.openbmc_project.Software.Version.VersionPurpose.BMC",
    "RequestedActivation": "xyz.openbmc_project.Software.Activation.RequestedActivations.None",
    "Version": "v1.99.7-94-g9994014"
  },
  "message": "200 OK",         <--- Present
  "status": "ok"                        <--- Present
}bash-4.1$
```

Firmware info:
```
BMC build : v1.99.7-107-gf9707d1
PNOR build : 
IBM-witherspoon-ibm-OP9_v1.17_1.46
        op-build-v1.17-101-g1c57f18-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-158-ga1e0a047b2a0
        hostboot-0ffbb55
        linux-4.11.6-openpower1-p57dd16e
        petitboot-v1.4.3-pfac92ff
        machine-xml-5b59a1d
        occ-1dc97a6
        hostboot-binaries-711147e
        capp-ucode-9c73e9f
        sbe-0df81aa
```
FFDC : 
[journal_for_REST_issue.txt](https://github.com/openbmc/openbmc/files/1132655/journal_for_REST_issue.txt)

geissonator, 2017-07-08:
Dup of #1927 ?

rahulmah, 2017-07-08:
Yes both issues are same.

gkeishin, 2017-07-10:
The code is now merged in dev.

closed 2017-07-10 by gkeishin