openbmc/openbmc #1927: REST response message not JSON format complaint

gkeishin, 2017-07-08:
`Older driver`
```
12:43:40.593 	INFO 	Get Request using : alias=openbmc, uri=/xyz/openbmc_project/sensors/fan_tach/fan0, headers=None json=None 	
12:43:40.593 	INFO 	${ret} = <Response [200]>

Response code:200, Content:{
  "data": {
    "Scale": 0, 
    "Target": 0, 
    "Unit": "xyz.openbmc_project.Sensor.Value.Unit.RPMS", 
    "Value": 0
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

`Newer Driver`

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.7-107"
VERSION_ID="v1.99.7-107-gf9707d1"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.7-107"
BUILD_ID="v1.99.7"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.17-04c6ddff01cec5e54c6aeedcaa7691913f404796 #1 Fri Jul 7 23:20:55 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#
```
Missing `data` in the response message.
```

22:15:17.579 	INFO 	Get Request using : alias=openbmc, uri=/xyz/openbmc_project/sensors/fan_tach/fan0, headers=None 	
22:15:17.579 	INFO 	${ret} = <Response [200]>

Response code:200, Content:{
  "Scale": 0, 
  "Target": 0, 
  "Unit": "xyz.openbmc_project.Sensor.Value.Unit.RPMS", 
  "Value": 0
}
```

gkeishin, 2017-07-08:
@bradbishop @williamspatrick Anything changed recently.

bradbishop, 2017-07-08:
probably something to do with this commit here:  https://gerrit.openbmc-project.xyz/#/c/5022/

gkeishin, 2017-07-08:
Thanks @bradbishop

dkodihal, 2017-07-08:
Fix here : https://gerrit.openbmc-project.xyz/#/c/5165/1

closed 2017-07-09 by williamspatrick